ompl/ompl #987: How to implement control based RRT* algorithm?

muk465, 2023-02-08:
1)I have used Control based RRT to follow  my custom kinematic model constraints as provided by ompl control planners.
2)I want to do the same with RRT * ,is there a similar implementation available in ompl ?
3)If not how can i use similar setup to write RRT* for this?

zkingston, 2023-02-08:
`RRT*` (and broadly most of the `*` algorithms) require the ability to directly connect two configurations, thus requiring solutions to the two-point boundary value problem for your dynamics model. If you have this, you can implement your dynamics as a geometric space with a custom interpolation function. If you do not, you must rely on kinodynamic AO planners such as `SST`.

closed 2023-02-08 by zkingston

muk465, 2023-02-10:
@zkingston thank you for the answer so are you saying if i dont have dynamics i should use SST planner? SST planner is a near optimal version of rrt for non holonomic planning ,is it better than rrt*? i couldnt find a comparison between sst and rrt* for non holonomic planning.

zkingston, 2023-02-10:
If you have a controller, you can use a geometric planner where interpolation uses your controller to connect two points. This is usually more effective than kinodynamic planning. RRT* cannot be used if there is no way to connect two points.