ompl/ompl #988: How to get corresponding actions taken by control based rrt ?

muk465, 2023-02-10:
1)I get my path using **ss.getSolutionPath().asGeometric().printAsMatrix()** it has around 13 steps
2)I am saving actions in the propogate step and then i am getting an array of size 47*10^6 which means its saving all actions of the points explored.
3)My question is how to only get actions of the points given by   **ss.getSolutionPath().asGeometric().printAsMatrix()**?

closed 2023-02-10 by zkingston

muk465, 2023-02-11:
@zkingston according to documentation this is returning a pointer of vector control ,i want the values for all controls so how can i get them ,i tried the index attribute like **a=ss.getSolutionPath().getControls() and then a.index(0)** to get the control for 0th path point but this is not working its returning an error,

> Try `ss.getSolutionPath().getControls()`. See https://ompl.kavrakilab.org/classompl_1_1control_1_1PathControl.html for more.

zkingston, 2023-02-11:
Yes, it returns a vector of the controls. You can also try the `.getControl(index)` function to access a specific control.

muk465, 2023-02-12:
> Yes, it returns a vector of the controls. You can also try the `.getControl(index)` function to access a specific control.

@zkingston  it is returning me this **<ompl.control._control.vectorControlPtr at 0x7fbab7ebe820>** how do i get values out of this ?for both **ss.getSolutionPath().getControls()** i am getting pointer refrencing at a location how to get value out of this in python?

zkingston, 2023-02-13:
To be clear, 
```
c0 = ss.getSolutionPath().getControls()[0]
print(c0[0])
```
does not work?