ompl/ompl #999: No module named 'ompl'

joonsub2000, 2023-02-26:
Hi, I installed the ompl and python bindings. However when I try to import ompl, following error occurs. "ModuleNotFoundError: No module named 'ompl'"

To import ompl, I add the directory as follows.

---------------------------------------------------------------------------------------------------------------
sys.path.append('/home/anaconda3/envs/simul/ompl-1.6.0/build/Release/lib')
sys.path.append('/home/anaconda3/envs/simul/ompl-1.6.0/py-bindings')
sys.path.append('/home/anaconda3/envs/simul/ompl-1.6.0')

 >>> import ompl

>>>ModuleNotFoundError                       Traceback (most recent call last)
Cell In[2], line 1
----> 1 from ompl.util._util import *
ModuleNotFoundError: No module named 'ompl'
---------------------------------------------------------------------------------------------------------------

I referred to the following Issues, but could not solve it (https://github.com/ompl/ompl/issues/977)

I will attach CMakeCache.txt and the log when making update_bindings, so please help.

joonsub2000, 2023-02-26:
# This is the CMakeCache file.
# For build in directory: /home/rnbmarch/anaconda3/envs/joonsub_simul/ompl-1.6.0/build/Release
# It was generated by CMake: /home/rnbmarch/anaconda3/envs/joonsub_simul/lib/python3.9/site-packages/cmake/data/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy38.so

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy38.so

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py38 library (debug)
Boost_PYTHON-PY38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY38_LIBRARY_DEBUG-NOTFOUND

//Boost python-py38 library (release)
Boost_PYTHON-PY38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY38_LIBRARY_RELEASE-NOTFOUND

//Boost python38 library (debug)
Boost_PYTHON38_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python38.so

//Boost python38 library (release)
Boost_PYTHON38_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python38.so

//Boost serialization library (debug)
Boost_SERIALIZATION_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost serialization library (release)
Boost_SERIALIZATION_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost system library (debug)
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost system library (release)
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost unit_test_framework library (debug)
Boost_UNIT_TEST_FRAMEWORK_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so

[... 763 lines cut ...]
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//linker supports push/pop state
_CMAKE_LINKER_PUSHPOP_STATE_SUPPORTED:INTERNAL=TRUE
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_FLANN:INTERNAL=flann
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann
pkgcfg_lib_FLANN_flann-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann_cpp
pkgcfg_lib_FLANN_flann_cpp-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_lz4
pkgcfg_lib_FLANN_lz4-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr//usr/lib/x86_64-linux-gnu

joonsub2000, 2023-02-26:
$ make update_bindings
[  0%] Preparing C++ header file for Python binding generation for module control
[  0%] Built target control.h
[  0%] Preparing C++ header file for Python binding generation for module base
[  0%] Built target base.h
[100%] Preparing C++ header file for Python binding generation for module util
[100%] Built target util.h
[100%] Creating C++ code for Python module util (see pyplusplus_util.log)
/usr/lib/pypy3/lib-python/3/importlib/_bootstrap.py:228: DeprecationWarning: invalid escape sequence 'm'
  return f(*args, **kwds)
std::vector<double>::reference std::vector<double>::at(std::size_t
__n) [member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<double>::reference std::vector<double>::front() [member
function] > warning W1008: The function returns non-const reference to
"Python immutable" > type. The value cannot be modified from Python.
std::vector<double>::reference std::vector<double>::back() [member
function] > warning W1008: The function returns non-const reference to
"Python immutable" > type. The value cannot be modified from Python.
std::vector<double>::pointer
std::vector<double>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const double *, std::vector<double>>
__first, __gnu_cxx::__normal_iterator<const double *,
std::vector<double>> __last) [member function] > compilation error
W1050: The function returns "std::vector<double>::pointer" > type. You
have to specify a call policies.Be sure to take a look on `Py++` >
defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer
__first, std::vector<double>::pointer __last,
std::vector<double>::pointer __result,
std::vector<double>::_Tp_alloc_type & __alloc, std::true_type arg4)
[member function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer arg0,
std::vector<double>::pointer arg1, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & arg3, std::false_type
arg4) [member function] > compilation error W1050: The function
returns "std::vector<double>::pointer" > type. You have to specify a
call policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_relocate(std::vector<double>::pointer __first,
std::vector<double>::pointer __last, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & __alloc) [member
function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies.Be sure to take a look on `Py++` > defined call policies
std::vector<unsigned int>::reference std::vector<unsigned
int>::at(std::size_t __n) [member function] > warning W1008: The
function returns non-const reference to "Python immutable" > type. The
value cannot be modified from Python.
std::vector<unsigned int>::reference std::vector<unsigned
int>::front() [member function] > warning W1008: The function returns
non-const reference to "Python immutable" > type. The value cannot be
modified from Python.
std::vector<unsigned int>::reference std::vector<unsigned int>::back()
[member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<unsigned int>::pointer std::vector<unsigned
int>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const unsigned int *,
std::vector<unsigned int>> __first, __gnu_cxx::__normal_iterator<const
unsigned int *, std::vector<unsigned int>> __last) [member function] >
compilation error W1050: The function returns "std::vector<unsigned >
int>::pointer" type. You have to specify a call policies.Be sure to
take a > look on `Py++` defined call policies
[... 2872 lines cut ...]
-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/rnbmarch/anaconda3/envs/joonsub_simul/ompl-1.6.0/build/Release
[100%] Built target update_control_bindings
[100%] Built target update_bindings

joonsub2000, 2023-02-26:
build/Release$ make
[ 38%] Built target ompl
[ 39%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/AbstractState.pypp.cpp.o
[ 39%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/AllValidStateValidityChecker.pypp.cpp.o
[ 39%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/AtlasChart.pypp.cpp.o
[ 39%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/AtlasState.pypp.cpp.o
[ 39%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/AtlasStateSampler.pypp.cpp.o
[ 39%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/AtlasStateSpace.pypp.cpp.o
[ 40%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/BridgeTestValidStateSampler.pypp.cpp.o
[ 40%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/CompoundState.pypp.cpp.o
[ 40%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/CompoundStateInternal.pypp.cpp.o
[ 40%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/CompoundStateSampler.pypp.cpp.o
[ 40%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/CompoundStateSpace.pypp.cpp.o
[ 40%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ConstrainedMotionValidator.pypp.cpp.o
[ 41%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ConstrainedSpaceInformation.pypp.cpp.o
[ 41%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ConstrainedState.pypp.cpp.o
[ 41%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ConstrainedStateSpace.pypp.cpp.o
[ 41%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ConstrainedValidStateSampler.pypp.cpp.o
[ 41%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/Constraint.pypp.cpp.o
[ 41%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ConstraintObjective.pypp.cpp.o
[ 42%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/Cost.pypp.cpp.o
[ 42%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/CostConvergenceTerminationCondition.pypp.cpp.o
[ 42%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DiscreteMotionValidator.pypp.cpp.o
[ 42%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DiscreteState.pypp.cpp.o
[ 42%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DiscreteStateSampler.pypp.cpp.o
[ 42%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DiscreteStateSpace.pypp.cpp.o
[ 43%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DubinsMotionValidator.pypp.cpp.o
[ 43%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DubinsState.pypp.cpp.o
[ 43%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/DubinsStateSpace.pypp.cpp.o
[ 43%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GaussianValidStateSampler.pypp.cpp.o
[ 43%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GenericParam.pypp.cpp.o
[ 43%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/Goal.pypp.cpp.o
[ 44%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GoalLazySamples.pypp.cpp.o
[ 44%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GoalRegion.pypp.cpp.o
[ 44%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GoalSampleableRegion.pypp.cpp.o
[ 44%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GoalSpace.pypp.cpp.o
[ 44%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GoalState.pypp.cpp.o
[ 44%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/GoalStates.pypp.cpp.o
[ 45%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/IterationTerminationCondition.pypp.cpp.o
[ 45%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MaximizeClearanceValidStateSampler.pypp.cpp.o
[ 45%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MaximizeMinClearanceObjective.pypp.cpp.o
[ 45%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MechanicalWorkOptimizationObjective.pypp.cpp.o
[ 45%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MinimaxObjective.pypp.cpp.o
[ 45%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MinimumClearanceValidStateSampler.pypp.cpp.o
[ 46%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MotionValidator.pypp.cpp.o
[ 46%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/MultiOptimizationObjective.pypp.cpp.o
[ 46%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ObstacleBasedValidStateSampler.pypp.cpp.o
[ 46%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/OptimizationObjective.pypp.cpp.o
[ 46%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/OrderedInfSampler.pypp.cpp.o
[ 46%] Building CXX object py-bindings/CMakeFiles/py_ompl_base.dir/bindings/base/ParamSet.pypp.cpp.o
[... cut ...]
[ 94%] Built target demo_ConstrainedPlanningKinematicChain
[ 94%] Built target demo_MultiLevelPlanningRigidBody2D
[ 94%] Built target demo_MultiLevelPlanningRigidBody3D
[ 95%] Built target demo_MultiLevelPlanningKinematicChain
[ 95%] Built target demo_MultiLevelPlanningHyperCube
[ 95%] Built target demo_MultiLevelPlanningHyperCubeBenchmark
[ 96%] Built target demo_SpaceTimePlanning
[ 97%] Built target demo_PlanarManipulator
[ 98%] Built target demo_MultiLevelPlanarManipulator
[100%] Built target demo_OpenDERigidBodyPlanning

zkingston, 2023-02-26:
Which version of Python are you importing ompl into? It appears that you successfully built the bindings for Python 3.8.

joonsub2000, 2023-02-26:
Oh, I currently use python 3.9 in conda environment. I think I have to downgrade my python version 3.9 to 3.8. Am I right?

zkingston, 2023-02-26:
Correct, or build the Python bindings with reference to the correct Python executable (through `-DPYTHON_EXEC`) and with proper boost support.

zkingston, 2023-03-21:
Closing given age. Reply if you want to follow up on this issue.

closed 2023-03-21 by zkingston

zkingston, 2023-07-09:
See previous discussion in #1059 and #984