ompl/ompl #1000: Issues compiling OMPL Demo with new package versions since upgrading to Ubuntu 22.04

IEI245MU5, 2023-03-06:
Hello, I am trying to compile the SE2RigidBodyPlanning.cpp Demo, but since upgrading to Ubuntu 22.04 I am unable to.

I get the following error when compiling:
```
Consolidate` compiler generated dependencies of target Demo
[ 37%] Linking CXX executable Demo
/usr/bin/ld: warning: libfcl.so.0.5, needed by /usr/local/lib/libompl_app_base.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_filesystem.so.1.71.0, needed by /usr/local/lib/libompl_app_base.so, may conflict with libboost_filesystem.so.1.74.0
/usr/bin/ld: warning: libboost_serialization.so.1.71.0, needed by /usr/local/lib/libompl.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::continuousCollide(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::Transform3f const&, fcl::ContinuousCollisionRequest const&, fcl::ContinuousCollisionResult&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::computeRule_median(fcl::OBBRSS const&, unsigned int*, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::computeRule_mean(fcl::OBBRSS const&, unsigned int*, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::Quaternion3f::fromAxisAngle(fcl::Vec3fX<fcl::details::Vec3Data<double> > const&, double)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::computeRule_bvcenter(fcl::OBBRSS const&, unsigned int*, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::endModel()'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::apply(fcl::Vec3fX<fcl::details::Vec3Data<double> > const&) const'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::collide(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionRequest const&, fcl::CollisionResult&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::addSubModel(std::vector<fcl::Vec3fX<fcl::details::Vec3Data<double> >, std::allocator<fcl::Vec3fX<fcl::details::Vec3Data<double> > > > const&, std::vector<fcl::Triangle, std::allocator<fcl::Triangle> > const&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::beginModel(int, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `vtable for fcl::BVFitter<fcl::OBBRSS>'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::AABB::AABB()'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::getNodeType() const'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::distance(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::DistanceRequest const&, fcl::DistanceResult&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::computeLocalAABB()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/Demo.dir/build.make:103: Demo] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:115: CMakeFiles/Demo.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```

Currently I have packages installed: libfcl0.7, libboost_filesystem1.71.0, libboost_filesystem1.74.0, libboost_filesystem1.74-dev, libboost_serialization-dev,  libboost_serialization1.74-dev, libboost_serialization1.74.0,   

When trying to install the lower versions libfcl.so.0.5 and libboost_serialization.so.1.71.0 through apt-get install, I get the errors: Unable to locate package / Couldn't find any package by glob / Couldn't find any package by regex . Seemingly Ubuntu 22.04 ships with the above versions, so only those exist for it.

Strangely libboost_filesystem1.71.0 is installed, but it still shows up in the error message.

Before upgrading to Ubuntu 22.04 everything worked fine.

My CMakeLists.txt looks like following:
```
cmake_minimum_required(VERSION 3.10)
project(OMPL_Demo)

find_package(
    ompl REQUIRED
    fcl REQUIRED
)

include_directories(
    ${EIGEN_INCLUDE_DIR}
    ${OMPL_INCLUDE_DIRS}
    ${fcl_INCLUDE_DIRS}
)

add_executable(Demo src/SE2RigidBodyPlanning.cpp)
target_link_libraries(Demo
    PUBLIC ${OMPLAPP_LIBRARIES}
    PUBLIC ${OMPL_LIBRARIES}
    PUBLIC ${fcl_LIBRARIES}
    PUBLIC fcl ${CCD_LIBRARIES}
)
```

Has anyone encountered this issue before?
Do I have to downgrade my Ubuntu version?

Thank you for your answers

zkingston, 2023-03-09:
I would guess that given you have conflicting versions of boost that the build is breaking in some interesting way. I would verify that all old versions of boost are removed and only one version remains, and then clean and build from scratch. You could also post your CMakeCache.txt from your build directory so we can see if multiple boost versions are being linked against.

IEI245MU5, 2023-03-13:
Thank you for your response!

I have uninstalled all libboost and livfcl packages and then reinstalled using

```
sudo apt-get install libboost-all-dev
sudo apt-get install libfcl-dev
```

after rebuilding from scratch I get the (almost same) error message:

```
cmake --build .
[ 50%] Building CXX object CMakeFiles/Demo.dir/src/SE2RigidBodyPlanning.cpp.o
[100%] Linking CXX executable Demo
/usr/bin/ld: warning: libfcl.so.0.5, needed by /usr/local/lib/libompl_app_base.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_filesystem.so.1.71.0, needed by /usr/local/lib/libompl_app_base.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_serialization.so.1.71.0, needed by /usr/local/lib/libompl.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::continuousCollide(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::Transform3f const&, fcl::ContinuousCollisionRequest const&, fcl::ContinuousCollisionResult&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::computeRule_median(fcl::OBBRSS const&, unsigned int*, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::computeRule_mean(fcl::OBBRSS const&, unsigned int*, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::Quaternion3f::fromAxisAngle(fcl::Vec3fX<fcl::details::Vec3Data<double> > const&, double)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `boost::filesystem::absolute(boost::filesystem::path const&, boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::computeRule_bvcenter(fcl::OBBRSS const&, unsigned int*, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::endModel()'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVSplitter<fcl::OBBRSS>::apply(fcl::Vec3fX<fcl::details::Vec3Data<double> > const&) const'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::collide(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionRequest const&, fcl::CollisionResult&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::addSubModel(std::vector<fcl::Vec3fX<fcl::details::Vec3Data<double> >, std::allocator<fcl::Vec3fX<fcl::details::Vec3Data<double> > > > const&, std::vector<fcl::Triangle, std::allocator<fcl::Triangle> > const&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::beginModel(int, int)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `vtable for fcl::BVFitter<fcl::OBBRSS>'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::AABB::AABB()'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::getNodeType() const'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::distance(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::DistanceRequest const&, fcl::DistanceResult&)'
/usr/bin/ld: /usr/local/lib/libompl_app_base.so: undefined reference to `fcl::BVHModel<fcl::OBBRSS>::computeLocalAABB()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/Demo.dir/build.make:103: Demo] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/Demo.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```

The only difference I can make out to the prior error message is the 5th line
`/usr/bin/ld: warning: libboost_filesystem.so.1.71.0, needed by /usr/local/lib/libompl_app_base.so, not found (try using -rpath or -rpath-link)`

This is how my CMakeCache.txt looks like:

```
# This is the CMakeCache file.
# For build in directory: /home/kaiwen/MA_Kaiwen/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a program.
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None Debug Release RelWithDebInfo
// MinSizeRel ...
[... 342 lines cut ...]
//ADVANCED property for variable: CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_SKIP_INSTALL_RPATH
CMAKE_SKIP_INSTALL_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_SKIP_RPATH
CMAKE_SKIP_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS
CMAKE_STATIC_LINKER_FLAGS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_DEBUG
[... cut ...]
//uname command
CMAKE_UNAME:INTERNAL=/usr/bin/uname
//ADVANCED property for variable: CMAKE_VERBOSE_MAKEFILE
CMAKE_VERBOSE_MAKEFILE-ADVANCED:INTERNAL=1
//Details about finding ompl
FIND_PACKAGE_MESSAGE_DETAILS_ompl:INTERNAL=[/usr/local/include/ompl-1.5;/usr/include;/usr/include/eigen3][/usr/local/lib;/usr/lib/x86_64-linux-gnu][/usr/local/lib/libompl.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;ode][v()]
```

zkingston, 2023-03-21:
It looks like there is an issue with your previously installed version of OMPL - you have a build of 1.5 in `/usr/local` that is messing with the build process. See what happens when you remove those files in `/usr/local/include/ompl-1.5`, `/usr/local/lib/libompl_*`, `/usr/local/share/ompl`, and `/usr/local/lib/pkgconfig/ompl.pc` (there may be more...)

IEI245MU5, 2023-03-23:
After removing those files cmake was unable to find ompl anymore, so I decided to remove all ompl files and reinstall from source using this option  `./install-ompl-ubuntu.sh --app` .

During installation I got the following error output:

```
-- Build type: Release
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
CMake Error at CMakeLists.txt:177 (include_directories):
  include_directories given empty-string as include directory.

<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * OpenGL, <http://opengl.org>
   Used to render graphics in ompl_app.
 * pypy, <http://pypy.org>
   Used to speed up the generation of python bindings.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., http://ompl.kavrakilab.org).
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.
 * Python, <http://python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.

-- The following REQUIRED packages have been found:

 * draco
 * assimp, <http://assimp.org>
   Used in ompl_app for reading meshes representing robots and environments.
 * ccd, <https://github.com/danfis/libccd>
   Collision detection library used by fcl.
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * octomap
 * fcl, <https://github.com/flexible-collision-library/fcl>
   The default collision checking library.
 * Boost, <http://boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Drawstuff, <http://ode.org>
   Part of the ODE source distribution, used in one demo program.
 * PQP, <http://gamma.cs.unc.edu/SSV/>
   Used as an alternative, additional collision checking library (the default is FCL).

-- The following Python modules were found:

 * PyQt5 OpenGL numpy pyplusplus pygccxml flask celery

-- Configuring incomplete, errors occurred!
See also "/home/kaiwen/Downloads/omplapp-1.6.0-Source/build/Release/CMakeFiles/CMakeOutput.log".

```
I have attached the log file here:
[CMakeOutput.log](https://github.com/ompl/ompl/files/11049142/CMakeOutput.log)

and the CMakeCache.txt:
[CMakeCache.txt](https://github.com/ompl/ompl/files/11049200/CMakeCache.txt)

There was a similar issue here: #737  and it was also suggested that a previously installed version is messing with the installation. I tried removing all files from previous installations that I could find, but it did not make any difference.

Sorry, I am very inexperienced in reading such log files, so I could not make out what the issue is.
Thank you a lot for your help.

zkingston, 2023-03-27:
Oh, sorry, I wasn't looking at the `ompl.app` CMake. It's failing at this line:
```
include_directories(
    "${OMPLAPP_INCLUDE_DIRS}"
    "${OMPL_INCLUDE_DIRS}"
    "${EIGEN3_INCLUDE_DIR}"
    "${FCL_INCLUDE_DIRS}"
    "${ASSIMP_INCLUDE_DIRS}")
```
Could you add some lines to the CMake so I can see which of these is empty? Something like:
```
message(WARNING ${EIGEN3_INCLUDE_DIR})
```
for each of the dependencies.

IEI245MU5, 2023-03-29:
I added the following lines to the CMake at the preceding line numbers:
```
176 message(WARNING ${OMPLAPP_INCLUDE_DIRS})
177 message(WARNING ${OMPL_INCLUDE_DIRS})
178 message(WARNING ${EIGEN3_INCLUDE_DIR})
179 message(WARNING ${FCL_INCLUDE_DIRS})
180 message(WARNING ${ASSIMP_INCLUDE_DIRS})
```
Running cmake gets me the following output:
```
kaiwen@kaiwen-MS-7A68:~/Downloads/omplapp-1.6.0-Source/build$ cmake ..
-- Build type: Release
-- Could NOT find Python (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) 
CMake Warning at CMakeLists.txt:99 (message):
  Either PyQt5, PySide2, PyQt4 or PySide needs to be installed to use the
  GUI.

-- Could NOT find PY_OpenGL (missing: PY_OPENGL) 
CMake Warning at CMakeLists.txt:106 (message):
  Both OpenGL and the Python OpenGL module need to be installed to use the
  GUI

-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
CMake Warning at CMakeLists.txt:176 (message):
  
  /home/kaiwen/Downloads/omplapp-1.6.0-Source/src/home/kaiwen/Downloads/omplapp-1.6.0-Source/build/src

CMake Warning at CMakeLists.txt:177 (message):
  
  /home/kaiwen/Downloads/omplapp-1.6.0-Source/ompl/src/home/kaiwen/Downloads/omplapp-1.6.0-Source/build/ompl/src

CMake Warning at CMakeLists.txt:178 (message):

CMake Warning at CMakeLists.txt:179 (message):
  /usr/include/usr/include/usr/include

CMake Warning at CMakeLists.txt:180 (message):
  /usr/include/usr/include

CMake Error at CMakeLists.txt:183 (include_directories):
  include_directories given empty-string as include directory.

-- Could NOT find PY_pyplusplus (missing: PY_PYPLUSPLUS) 
-- Could NOT find PY_pygccxml (missing: PY_PYGCCXML) 
-- Could NOT find PY_pyplusplus (missing: PY_PYPLUSPLUS) 
-- Could NOT find PY_pygccxml (missing: PY_PYGCCXML) 
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * OpenGL, <http://opengl.org>
   Used to render graphics in ompl_app.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., http://ompl.kavrakilab.org).
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.

-- The following REQUIRED packages have been found:

 * draco
 * assimp, <http://assimp.org>
   Used in ompl_app for reading meshes representing robots and environments.
 * ccd, <https://github.com/danfis/libccd>
   Collision detection library used by fcl.
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * octomap
 * fcl, <https://github.com/flexible-collision-library/fcl>
   The default collision checking library.
 * Boost (required version >= 1.58), <http://boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following features have been disabled:

 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Drawstuff, <http://ode.org>
   Part of the ODE source distribution, used in one demo program.
 * PQP, <http://gamma.cs.unc.edu/SSV/>
   Used as an alternative, additional collision checking library (the default is FCL).
 * Python, <http://python.org>
   Used for python bindings.

-- The following Python modules were NOT found:

 * PyQt5 PySide2 PyQt4 PySide OpenGL pyplusplus pygccxml

-- Configuring incomplete, errors occurred!
See also "/home/kaiwen/Downloads/omplapp-1.6.0-Source/build/CMakeFiles/CMakeOutput.log".

```

The packages OpenGl, Python OpenGl, PyQt5 und such that CMake could not find are all installed.
The line corresponding to Eigen3 is empty. What is the significance of this?

zkingston, 2023-04-01:
Not quite sure, if you could comment out the `${EIGEN3_INCLUDE_DIR}` line and see if things compile?

IEI245MU5, 2023-04-27:
I have reinstalled Ubuntu 20.04 and now the cpp Demos work fine.

closed 2023-04-27 by IEI245MU5