ompl/ompl #1002: How to give a goal region for a control based RRT in a simple setup?(python)

muk465, 2023-03-16:
Currently i am using a single point as a goal and the following is the code

        ```
  space = ob.SE3StateSpace()
        
        space.setSubspaceWeight(1,0)
             
        bounds = ob.RealVectorBounds(3)
        bounds.setLow(0)
        bounds.setHigh(255)
        space.setBounds(bounds)
        
        cspace = oc.RealVectorControlSpace(space, 1)
          
      
        cbounds = ob.RealVectorBounds(1)
        cbounds.setLow(0)
        cbounds.setHigh(2*pi)
        cspace.setBounds(cbounds)
        
         ss = oc.SimpleSetup(cspace)
       
        ss.setStateValidityChecker(ob.StateValidityCheckerFn(self.isStateValid))
        
        ss.setStatePropagator(oc.StatePropagatorFn(self.propagate))
          
        # create a start state
        start = ob.State(space)
        start().setX(self.start_state[0])
        start().setY(self.start_state[1])
        start().setZ(self.start_state[2])
        start().rotation().setIdentity()
        
        goal = ob.State(space)
        goal().setX(self.goal_state[0])
        goal().setY(self.goal_state[1])
        goal().setZ(self.goal_state[2])
        goal().rotation().setIdentity()
        
      
       
        ss.setStartAndGoalStates(start, goal,0.05)
        ss.getSpaceInformation().setStateValidityCheckingResolution(0.1)
        
        si = ss.getSpaceInformation()
        planner = oc.RRT(si)
        ss.setPlanner(planner)
        si.setPropagationStepSize(pi/20)
```

1)What I want to ask is after I create a class of goa region how do i set start and goal states like in   **ss.setStartAndGoalStates(start, goal,0.05)** .
2)Is there a python tutorial for creating goal region available ?

zkingston, 2023-03-16:
1. Use `ss.setStartState(start)` and `ss.setGoal(goal)` (for abstract goals) or `ss.setGoalState(goal, epsilon)` for a goal state.

2. Quick example of [goal sampleable region](https://ompl.kavrakilab.org/classompl_1_1base_1_1GoalSampleableRegion.html) in Python:
```python
class GoalExample(ob.GoalSampleableRegion):

    # Initialization
    def __init__(self, si: ob.SpaceInformation, ...):
        super().__init__(si)
        ...

    # How far away is the state from the goal region?
    def distanceGoal(self, state: ob.State) -> float:
        return ...

    # Sample a state in the goal region
    def sampleGoal(self, state: ob.State):
        ...
    
    # How many different samples are available in the goal region?
    def maxSampleCount(self) -> int:
        return ...

    # Are there more samples that could be drawn from the goal region?
    def couldSample(self) -> bool:
        return ...
```

closed 2023-03-16 by zkingston

muk465, 2023-03-18:
@zkingston 1) i have a image where points of label =2 i.e **img[:,:,:]==2** make goal region how can i make it a goal region?is there a more explicit example explaining to make a goal region ?

2) Why have you used goal sampalable region rather than just goal ? like given here [https://ompl.kavrakilab.org/goalRepresentation.html] in myarbitrary goal ,here cant I just give return true if its in the region of label 
3)how do i define these functions  **sampleGoal**, **maxSampleCount**,**couldSample** for my case i can just write i can return img[:,:,:]==2 for sample goal ,but do i return all possible voxels for max sample count and i don't understand could sample?
4)for **distanceGoal** how do i calculate distance of a region with  current state which is a point? distance is between 2 points

zkingston, 2023-03-18:
1. https://ompl.kavrakilab.org/Point2DPlanning_8py_source.html is an example of planning in 2D with a PPM image.
2. Simply remove the functions specific to `GoalSampleableRegion` to implement a `Goal`
3. If you have a single goal sample, then return 1 for `maxSampleCount`. `sampleGoal` sets the input state to a goal state.
4. Whatever the distance in your state space is could be used.