ompl/ompl #1003: How to give a tolerance to control based rrt? that is to stop within this distance of goal?

muk465, 2023-03-18:
Currently i am using a single point as a goal and the following is the code

        ```
  space = ob.SE3StateSpace()
        
        space.setSubspaceWeight(1,0)
             
        bounds = ob.RealVectorBounds(3)
        bounds.setLow(0)
        bounds.setHigh(255)
        space.setBounds(bounds)
        
        cspace = oc.RealVectorControlSpace(space, 1)
          
      
        cbounds = ob.RealVectorBounds(1)
        cbounds.setLow(0)
        cbounds.setHigh(2*pi)
        cspace.setBounds(cbounds)
        
         ss = oc.SimpleSetup(cspace)
       
        ss.setStateValidityChecker(ob.StateValidityCheckerFn(self.isStateValid))
        
        ss.setStatePropagator(oc.StatePropagatorFn(self.propagate))
          
        # create a start state
        start = ob.State(space)
        start().setX(self.start_state[0])
        start().setY(self.start_state[1])
        start().setZ(self.start_state[2])
        start().rotation().setIdentity()
        
        goal = ob.State(space)
        goal().setX(self.goal_state[0])
        goal().setY(self.goal_state[1])
        goal().setZ(self.goal_state[2])
        goal().rotation().setIdentity()
        
      
       
        ss.setStartAndGoalStates(start, goal,0.05)
        ss.getSpaceInformation().setStateValidityCheckingResolution(0.1)
        
        si = ss.getSpaceInformation()
        planner = oc.RRT(si)
        ss.setPlanner(planner)
        si.setPropagationStepSize(pi/20)
```

1)My goal is a point and I want RRT to stop within reaching a certain distance of goal how can this be done?

zkingston, 2023-03-18:
Creating custom PTCs can be done with:
```python
from ompl import base as ob

# Put whatever function in here
ob.PlannerTerminationCondition(
    ob.PlannerTerminationConditionFn(lambda: True)
)
```

closed 2023-03-18 by zkingston

muk465, 2023-03-19:
I created a function like this 
```
def terminate_condition(self,state):
        x=int(state.getX())
        y=int(state.getY())
        z=int(state.getZ())
        
        if self.img[x,y,z]==2:
            return True
        
  
```

and then in the space definition function I gave 

```
term_condition=ob.PlannerTerminationCondition(ob.PlannerTerminationConditionFn(self.terminate_condition))
 # attempt to solve the problem
  solved = ss.solve(term_condition(state))
```
But I cant access the variable state i.e current state so how do I access this variable ?
@zkingston

muk465, 2023-03-19:
@zkingston i also tried doing this 
```
state_c=ob.SE3StateSpace().getSubspace(0)
solved = ss.solve(term_condition(state_c))
```
but this gives this error 
**Python argument types in
    PlannerTerminationCondition.__call__(PlannerTerminationCondition, RealVectorStateSpace)
did not match C++ signature:
    __call__(ompl::base::PlannerTerminationCondition {lvalue})**

zkingston, 2023-03-19:
Termination conditions do not accept states as input. They control when the planner should stop searching for a plan - planners will terminate on their own if they find a goal. That condition should be in your goal.