ompl/ompl #1004: update_bindings does not do any work

MarionLepert, 2023-03-18:
I'm on macOS 13.0.1 on an M1 machine, trying to install ompl from source. 

I've successfully installed all the dependencies using brew. However, when I run `cmake ../.. -DPYTHON_EXEC=/opt/homebrew/bin/python3` (pointing to homebrew python, since I installed boost-python3 using brew), the command terminates after a couple seconds. stderr and stdout from running the `cmake` command are attached here: [debug.txt](https://github.com/ompl/ompl/files/11007748/debug.txt) . 

Not sure if relevant but lines 325-329 are suspicious to me:
```
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- Code for module util not found; type "make update_bindings"
```

`make update_bindings` results in the following stdout/stderr: [make_debug.txt](https://github.com/ompl/ompl/files/11007755/make_debug.txt)

Thanks for your help!

zkingston, 2023-03-18:
I think there might be an issue with you having both Python 3.8 and 3.11 on the same machine - I see Boost Python is found for 3.11, but the PYTHONPATH is for 3.8. Maybe try explicitly setting `PYTHON_EXEC` in the CMake args?

MarionLepert, 2023-03-19:
@zkingston thanks - that's interesting. I have been setting `PYTHON_EXEC` - the command I'm running is `cmake ../.. -DPYTHON_EXEC=/opt/homebrew/bin/python3`, where that python3 is the same homebrew python3.11 for which boost was installed:
```
(clutter_ompl) marion@Marions-MBP Release % /opt/homebrew/bin/python3 --version
Python 3.11.2
```

is there any other flag needed to override the bad pointer to 3.8?

MarionLepert, 2023-03-19:
I also tried setting the `PYTHONPATH` explicitly, to point to homebrew's site-packages for python3.11 (where I confirm I see pyplusplus and pygccxml):

`cmake ../.. -DPYTHON_EXEC=/opt/homebrew/bin/python3 -DPYTHONPATH=/opt/homebrew/lib/python3.11/site-packages`

This still results in `make update_bindings` finishing in a couple seconds. 

here's the stdout/stderr for cmake:  [cmake_debug.txt](https://github.com/ompl/ompl/files/11010370/cmake_debug.txt)
stdout/stderr for `make update_bindings`: [make_update_bindings_debug.txt](https://github.com/ompl/ompl/files/11010372/make_update_bindings_debug.txt)
and CMakeCache: 
[CMakeCache.txt](https://github.com/ompl/ompl/files/11010373/CMakeCache.txt)

zkingston, 2023-03-19:
Hmm, could be something strange with the include paths that `castxml` is using - If you explicitly modify in https://github.com/ompl/ompl/blob/main/CMakeModules/Findcastxml.cmake the `CASTXMLCFLAGS` option to include `-I/usr/include` (or whatever it is on your system), do you get the same missing `stdlib.h` error?

siilats, 2023-04-15:
I followed this to make a pyenv local system  in the ompl folder with the homebrew python11 with boost-python3 https://thecesrom.dev/2021/06/28/how-to-add-python-installed-via-homebrew-to-pyenv-versions/

zkingston, 2023-05-05:
Unfortunately there seems to be an issue with the binding generation in Python 3.11 (due to upstream issues with Py++). I don't have the bandwidth to address this issue at the time, please downgrade to Python 3.10 to generate the bindings.

mamoll, 2023-05-05:
> Unfortunately there seems to be an issue with the binding generation in Python 3.11 (due to upstream issues with Py++)

@zkingston Can you elaborate? I am using python3.11, castxml 0.6.1, py++ 1.8.5, and Boost 1.81 (all installed through MacPorts on an Intel-based Mac running macOS 13.3) and that seems to work just fine.

mamoll, 2023-05-05:
@siilats Try generating the bindings without using `pypy3`, just to test if that is the source of the problem.

zkingston, 2023-05-05:
Running on archlinux with Python 3.11.3, Py++ 1.8.5, Boost 1.81, castxml 0.6.1, and pygccxml 2.2.1 gives this output for binding generation (this is without PyPy3). Seems to actually be something with how pygccxml/castxml is handling things (potentially same cause as https://github.com/microsoft/vscode-cpptools/issues/9091):
```
In file included from /tmp/tmpdt8j5csf.h:1:
In file included from ./bindings/util.h:46:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/stdexcept:39:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/string:42:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/bits/char_traits.h:42:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/bits/postypes.h:40:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/cwchar:44:
/usr/include/wchar.h:397:8: error: unknown type name '_Float32'
extern _Float32 wcstof32 (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:402:8: error: unknown type name '_Float64'
extern _Float64 wcstof64 (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:407:8: error: unknown type name '_Float128'
extern _Float128 wcstof128 (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:412:8: error: unknown type name '_Float32x'
extern _Float32x wcstof32x (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:417:8: error: unknown type name '_Float64x'
extern _Float64x wcstof64x (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:512:8: error: unknown type name '_Float32'
extern _Float32 wcstof32_l (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:518:8: error: unknown type name '_Float64'
extern _Float64 wcstof64_l (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:524:8: error: unknown type name '_Float128'
extern _Float128 wcstof128_l (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:530:8: error: unknown type name '_Float32x'
extern _Float32x wcstof32x_l (const wchar_t *__restrict __nptr,
       ^
/usr/include/wchar.h:536:8: error: unknown type name '_Float64x'
extern _Float64x wcstof64x_l (const wchar_t *__restrict __nptr,
       ^
In file included from /tmp/tmpdt8j5csf.h:1:
In file included from ./bindings/util.h:46:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/stdexcept:39:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/string:54:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/bits/basic_string.h:4110:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/ext/string_conversions.h:43:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/../../../../include/c++/13.1.1/cstdlib:79:
/usr/include/stdlib.h:141:8: error: unknown type name '_Float32'
extern _Float32 strtof32 (const char *__restrict __nptr,
       ^
/usr/include/stdlib.h:147:8: error: unknown type name '_Float64'
extern _Float64 strtof64 (const char *__restrict __nptr,
       ^
/usr/include/stdlib.h:153:8: error: unknown type name '_Float128'
extern _Float128 strtof128 (const char *__restrict __nptr,
       ^
/usr/include/stdlib.h:159:8: error: unknown type name '_Float32x'
extern _Float32x strtof32x (const char *__restrict __nptr,
       ^
/usr/include/stdlib.h:165:8: error: unknown type name '_Float64x'
extern _Float64x strtof64x (const char *__restrict __nptr,
       ^
/usr/include/stdlib.h:234:10: error: unknown type name '_Float32'
                       _Float32 __f)
                       ^
/usr/include/stdlib.h:240:10: error: unknown type name '_Float64'
                       _Float64 __f)
                       ^
/usr/include/stdlib.h:246:4: error: unknown type name '_Float128'
                        _Float128 __f)
                        ^
[... 16 lines cut ...]
  File "/home/zak/.local/lib/python3.11/site-packages/pyplusplus/module_builder/boost_python_builder.py", line 154, in __parse_declarations
    decls = reader.read_files( files, compilation_mode )
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/project_reader.py", line 255, in read_files
    return self.__parse_all_at_once(files)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/project_reader.py", line 356, in __parse_all_at_once
    return self.read_string(''.join(header_content))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/project_reader.py", line 369, in read_string
    decls = reader.read_string(content)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/source_reader.py", line 377, in read_string
    decls = self.read_file(header_file)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/source_reader.py", line 303, in read_file
    return self.read_cpp_source_file(source_file)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/source_reader.py", line 322, in read_cpp_source_file
    xml_file = self.create_xml_file(ffname)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zak/.local/lib/python3.11/site-packages/pygccxml/parser/source_reader.py", line 263, in create_xml_file
    raise RuntimeError(
RuntimeError: Error occurred while running CASTXML xml file does not exist
```

zkingston, 2023-05-05:
Confirmed that the binding generation works with `clang` 15.0.7 in the same environment. Seems like there is some issue with `gcc`.

zkingston, 2023-09-22:
Please see #1100 and let me know if it now works on your M1 machine.

zkingston, 2023-10-09:
Closing due to no activity and other successes.

closed 2023-10-09 by zkingston