ompl/ompl #1005: Getting error in using GoalRegion in a simple setup for control based RRT planner

muk465, 2023-03-20:
```
class MyGoalRegion(ob.GoalRegion):
    def __init__(self,si:ob.SpaceInformation,goal_point):
        super().__init__(si)
        #img_path=os.path.join(image_dir,label_name)
        
        #self.img=img_arr
        self.goal_point=goal_point
    def isSatisfied(self,state:ob.State,distance):
        #x=int(state.getX())
        #y=int(state.getY())
        #z=int(state.getZ())
        
        x=np.round(state.getX()).astype(int)
        y=np.round(state.getY()).astype(int)
        z=np.round(state.getZ()).astype(int)
        
        if distance<=5:
            return True
        
    def distanceGoal(self,state:ob.State):
        #x=int(state.getX())
        #y=int(state.getY())
        #z=int(state.getZ())
        
        x=np.round(state.getX()).astype(int)
        y=np.round(state.getY()).astype(int)
        z=np.round(state.getZ()).astype(int)
        
        
        xg=self.goal_point[0]
        yg=self.goal_point[1]
        zg=self.goal_point[2]
        
        dist_curr=np.sqrt((x-xg)**2+(y-yg)**2+(z-zg)**2)
        
        return dist_curr
        

```

1)Above is the Goal region class I have defined distanceGoal function ,but  I am having this error **No to_python (by-value) converter found for C++ type: double**

2) Further is code for reference

```
         space = ob.SE3StateSpace()
        
        space.setSubspaceWeight(1,0)
             
        bounds = ob.RealVectorBounds(3)
        bounds.setLow(0)
        bounds.setHigh(255)
        space.setBounds(bounds)
        
        cspace = oc.RealVectorControlSpace(space, 1)
          
        # set the bounds for the control space
        cbounds = ob.RealVectorBounds(1)
        cbounds.setLow(0)
        cbounds.setHigh(2*pi)
        cspace.setBounds(cbounds)
        
        
        ss = oc.SimpleSetup(cspace)
        
        
        ss.setStateValidityChecker(ob.StateValidityCheckerFn(self.isStateValid))
        
        ss.setStatePropagator(oc.StatePropagatorFn(self.propagate))
          
        # create a start state
        start = ob.State(space)
        start().setX(self.start_state[0])
        start().setY(self.start_state[1])
        start().setZ(self.start_state[2])
        start().rotation().setIdentity()
        
        #create a random goal state
        #goal = ob.State(space)
        #goal.random()
        #goal().setX(self.goal_state[0])
        #goal().setY(self.goal_state[1])
        #goal().setZ(self.goal_state[2])
        #goal().rotation().setIdentity()
        
        goal= MyGoalRegion(ss.getSpaceInformation(),self.goal_state)
      
        # set the start and goal states
        #ss.setStartAndGoalStates(start, goal,0.05)
        ss.setStartState(start)
        ss.setGoal(goal)
        ss.getSpaceInformation().setStateValidityCheckingResolution(0.1)
        
        
        si = ss.getSpaceInformation()
        
        planner = oc.RRT(si)
        #planner.setGoalBias(0.05)
        #gb=planner.getGoalBias()
        ss.setPlanner(planner)
        
        # (optionally) set propagation step size
        si.setPropagationStepSize(pi/20)
        #state_c=ob.SE3StateSpace()
        state_c=ss.getStateSpace
        #term_condition=ob.PlannerTerminationCondition(ob.PlannerTerminationConditionFn(self.terminate_condition(state_c)))
             # attempt to solve the problem
        
        solved = ss.solve(60)
        
```

zkingston, 2023-03-20:
Is the conversion error in the `distanceGoal` function? Try to change `return dist_curr` to `return float(dist_curr)`. There might be an issue converting `numpy` types.

muk465, 2023-03-20:
1)yes error was there
2) so i removed isSatisfied function and just put setThreshold to the required  distance its working,is it correct method?
3)Since my goal is a region is it okay to use  **ss.setGoal(goal)** or some other function needs to be used?

zkingston, 2023-03-20:
`setGoal` is all you need, and yes, `setThreshold` should be sufficient.