ompl/ompl #1006: OMPL Python Bindings Missing tools

cohnt, 2023-03-23:
When I build the python bindings for ompl, all of the submodules are created except for tools. I've confirmed this occurs both with the install script, and when building from source (on master). I'm using Ubuntu 22.04 and Python 3.10. Any ideas why the tools folder is breaking in particular, or where I could start getting some more debug info?

```
/usr/lib/python3/dist-packages/ompl$ tree
.
├── base
│   ├── _base.so
│   └── __init__.py
├── bindings_generator.py
├── control
│   ├── _control.so
│   └── __init__.py
├── geometric
│   ├── _geometric.so
│   └── __init__.py
├── __init__.py
├── tools
│   └── __init__.py
└── util
    ├── __init__.py
    ├── libompl.so
    └── _util.so

```

```
Python 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ompl
>>> import ompl.base
>>> import ompl.control
>>> import ompl.geometric
>>> import ompl.util
>>> import ompl.tools
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/ompl/tools/__init__.py", line 2, in <module>
    from ompl.tools._tools import *
ModuleNotFoundError: No module named 'ompl.tools._tools'
```

zkingston, 2023-03-23:
Currently bindings are not made for the `tools` namespace - this is expected.

closed 2023-03-23 by zkingston

cohnt, 2023-03-23:
These bindings are necessary for all of the python [constrained motion planning demos](https://ompl.kavrakilab.org/constrainedPlanning.html). How can I run these examples without bindings for the tools namespace?

zkingston, 2023-03-23:
Good catch, you can comment out the block at https://github.com/ompl/ompl/blob/main/demos/constraint/ConstrainedPlanningCommon.py#L256 and remove the import at the top. `tools` is only used for benchmarking in these examples.

zkingston, 2023-03-23:
Please do.

zkingston, 2023-03-28:
@cohnt Check #1011.