ompl/ompl #1012: Dockerfile build failed

botoph4, 2023-03-30:
Hi,

I am trying to creat a docker image for OMPL. In `scripts` folder there are two docker files namly `omplap.Dockerfile` and `omplwebapp.yml`. As soon as ran `docker build -f ompl.Dockerfile -t ompl-env:1.0 .`, I got this error message:

```
 [+] Building 2.0s (4/4) FINISHED                                                                     
 => [internal] load build definition from ompl.Dockerfile                                       0.0s
 => => transferring dockerfile: 2.29kB                                                          0.0s
 => [internal] load .dockerignore                                                               0.0s
 => => transferring context: 2B                                                                 0.0s
 => ERROR [internal] load metadata for docker.io/library/ubuntu:jammy                           1.8s
 => [auth] library/ubuntu:pull token for registry-1.docker.io                                   0.0s
------
 > [internal] load metadata for docker.io/library/ubuntu:jammy:
------
ERROR: failed to solve: rpc error: code = Unknown desc = failed to solve with frontend dockerfile.v0: failed to create LLB definition: failed to authorize: rpc error: code = Unknown desc = failed to fetch oauth token: Post "https://auth.docker.io/token": proxyconnect tcp: dial tcp: lookup https on 127.0.0.53:53: server misbehaving
```
I just start get familiar with docker, please correct me if I am doing something wrong.

System: Ubuntu 22.04.2 LTS
Docker version: 23.0.2

zkingston, 2023-03-30:
This seems to be some issue with your local configuration. A stack overflow solution that might have answers: https://stackoverflow.com/questions/64221861/an-error-failed-to-solve-with-frontend-dockerfile-v0

botoph4, 2023-03-31:
Hey @zkingston,

thanks for your reply. Yes it helps when add ` DOCKER_BUILDKIT=0 `. 
 However, after installing python libs, an error seems related to compiling process.
```
Removing intermediate container 0c1cc529acdf
 ---> 16966a38adea
Step 4/13 : COPY . /ompl
 ---> fc7035decced
Step 5/13 : WORKDIR /build
 ---> Running in 694ca86489f2
Removing intermediate container 694ca86489f2
 ---> 9a959f0a6a60
Step 6/13 : RUN cmake         -DPYTHON_EXEC=/usr/bin/python3         -DOMPL_REGISTRATION=OFF         -DCMAKE_INSTALL_PREFIX=/usr         -G Ninja         /ompl &&     ninja update_bindings -j `nproc` &&     ninja -j `nproc` &&     ninja install
 ---> Running in cbc21aa0752c
CMake Error: The source directory "/ompl" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
The command '/bin/sh -c cmake         -DPYTHON_EXEC=/usr/bin/python3         -DOMPL_REGISTRATION=OFF         -DCMAKE_INSTALL_PREFIX=/usr         -G Ninja         /ompl &&     ninja update_bindings -j `nproc` &&     ninja -j `nproc` &&     ninja install' returned a non-zero code: 1
```

zkingston, 2023-03-31:
If you are in the `scripts/docker` directory you need to invoke build with the root directory at the top level of the OMPL repo, so for you: `docker build -f ompl.Dockerfile -t ompl-env:1.0 ../..` (note `../..`)

zkingston, 2023-04-11:
Closing this due to lack of activity - let me know if the problem still exists.

closed 2023-04-11 by zkingston