ompl/ompl #1014: DubinsState space implementation not using set classification

Jaeyoung-Lim, 2023-04-02:
**Problem Description**
While the DubinsStateSpace implementation mentions the Dubins set classification method, the implementation does not reflect the comment

https://github.com/ompl/ompl/blob/55588baaccbf4b48da926939e87f76d2f717ff70/src/ompl/base/spaces/DubinsStateSpace.h#L56-L62

@mamoll Was there a specific reason that the Dubins set classification scheme was not implemented?

mamoll, 2023-04-03:
No, there was no reason I didn't do it. Implementing this state space was a bit of side project without an immediate need, so I considered the current implementation "good enough" at the time. If you implement this, please do measure the speed up. I'm curious to see how much of a difference it makes.

Jaeyoung-Lim, 2023-04-03:
@mamoll Thanks! These are some primary results on my side constrained to the long path case

[image]

zkingston, 2023-04-03:
Wow, great speed up. I would definitely review a PR that implements these changes.

Jaeyoung-Lim, 2023-07-28:
Closing as addressed in #1080

closed 2023-07-28 by Jaeyoung-Lim