ompl/ompl #1016: No module named "ompl.base._base"

gunaa12, 2023-04-07:
I cloned the github manually and tried doing the steps mentioned in the README manually. It seemed to build without any errors, but when I import ompl in Python environement, I keep getting this issue. I went through the past issues similar to this, but they were of no help. I would appreciate any help on this issue.

For reference, here is my Cmakecache.txt from the build folder:

# This is the CMakeCache file.
# For build in directory: /home/corallab/ompl/build/Release
# It was generated by CMake: /home/corallab/.local/lib/python3.8/site-packages/cmake/data/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy38.so

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy38.so

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py38 library (debug)
Boost_PYTHON-PY38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY38_LIBRARY_DEBUG-NOTFOUND

//Boost python-py38 library (release)
Boost_PYTHON-PY38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY38_LIBRARY_RELEASE-NOTFOUND

//Boost python38 library (debug)
Boost_PYTHON38_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python38.so

//Boost python38 library (release)
Boost_PYTHON38_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python38.so

//Boost serialization library (debug)
Boost_SERIALIZATION_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost serialization library (release)
Boost_SERIALIZATION_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost system library (debug)
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost system library (release)
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so
[... 770 lines cut ...]
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//linker supports push/pop state
_CMAKE_LINKER_PUSHPOP_STATE_SUPPORTED:INTERNAL=TRUE
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_FLANN:INTERNAL=flann
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann
pkgcfg_lib_FLANN_flann-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann_cpp
pkgcfg_lib_FLANN_flann_cpp-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_lz4
pkgcfg_lib_FLANN_lz4-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr//usr/lib/x86_64-linux-gnu

zkingston, 2023-04-07:
Seems like everything was found, what's the output of `make update_bindings`?

gunaa12, 2023-04-07:
Here is the terminal output when I run the command "make -j 2 update_bindings":

[ 50%] Preparing C++ header file for Python binding generation for module tools
[ 50%] Preparing C++ header file for Python binding generation for module control
[ 50%] Built target tools.h
[ 50%] Built target control.h
[ 50%] Preparing C++ header file for Python binding generation for module base
[100%] Preparing C++ header file for Python binding generation for module util
[100%] Built target util.h
[100%] Built target base.h
[100%] Preparing C++ header file for Python binding generation for module geometric
[100%] Creating C++ code for Python module util (see pyplusplus_util.log)
[100%] Built target geometric.h
/usr/lib/pypy3/lib-python/3/importlib/_bootstrap.py:228: DeprecationWarning: invalid escape sequence 'm'
  return f(*args, **kwds)
std::vector<double, std::allocator<double> >::reference
std::vector<double, std::allocator<double> >::at(std::size_t __n)
[member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<double, std::allocator<double> >::reference
std::vector<double, std::allocator<double> >::front() [member
function] > warning W1008: The function returns non-const reference to
"Python immutable" > type. The value cannot be modified from Python.
std::vector<double, std::allocator<double> >::reference
std::vector<double, std::allocator<double> >::back() [member function]
> warning W1008: The function returns non-const reference to "Python
immutable" > type. The value cannot be modified from Python.
std::vector<double, std::allocator<double> >::pointer
std::vector<double, std::allocator<double>
>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const double *, std::vector<double,
std::allocator<double> > > __first, __gnu_cxx::__normal_iterator<const
double *, std::vector<double, std::allocator<double> > > __last)
[member function] > compilation error W1050: The function returns
"std::vector<double, > std::allocator<double> >::pointer" type. You
have to specify a call > policies.Be sure to take a look on `Py++`
defined call policies
static std::vector<double, std::allocator<double> >::pointer
std::vector<double, std::allocator<double>
>::_S_do_relocate(std::vector<double, std::allocator<double>
>::pointer __first, std::vector<double, std::allocator<double>
>::pointer __last, std::vector<double, std::allocator<double>
>::pointer __result, std::vector<double, std::allocator<double>
>::_Tp_alloc_type & __alloc, std::true_type arg4) [member function] >
compilation error W1050: The function returns "std::vector<double, >
std::allocator<double> >::pointer" type. You have to specify a call >
policies.Be sure to take a look on `Py++` defined call policies
static std::vector<double, std::allocator<double> >::pointer
std::vector<double, std::allocator<double>
>::_S_do_relocate(std::vector<double, std::allocator<double>
>::pointer arg0, std::vector<double, std::allocator<double> >::pointer
arg1, std::vector<double, std::allocator<double> >::pointer __result,
std::vector<double, std::allocator<double> >::_Tp_alloc_type & arg3,
std::false_type arg4) [member function] > compilation error W1050: The
function returns "std::vector<double, > std::allocator<double>
>::pointer" type. You have to specify a call > policies.Be sure to
take a look on `Py++` defined call policies
static std::vector<double, std::allocator<double> >::pointer
std::vector<double, std::allocator<double>
>::_S_relocate(std::vector<double, std::allocator<double> >::pointer
__first, std::vector<double, std::allocator<double> >::pointer __last,
std::vector<double, std::allocator<double> >::pointer __result,
std::vector<double, std::allocator<double> >::_Tp_alloc_type &
__alloc) [member function] > compilation error W1050: The function
returns "std::vector<double, > std::allocator<double> >::pointer"
type. You have to specify a call > policies.Be sure to take a look on
`Py++` defined call policies
std::vector<unsigned int, std::allocator<unsigned int> >::reference
std::vector<unsigned int, std::allocator<unsigned int>
[... 3419 lines cut ...]
ompl::control::EST::MotionInfo::operator[](unsigned int i) [member
operator] > compilation error W1005: `Py++` cannot expose function
that takes as > argument/returns instance of non-public class.
Generated code will not > compile.
ompl::control::EST::Motion const *
ompl::control::EST::MotionInfo::operator[](unsigned int i) const
[member operator] > compilation error W1005: `Py++` cannot expose
function that takes as > argument/returns instance of non-public
class. Generated code will not > compile.
void
ompl::control::EST::MotionInfo::push_back(ompl::control::EST::Motion *
m) [member function] > compilation error W1005: `Py++` cannot expose
function that takes as > argument/returns instance of non-public
class. Generated code will not > compile.
void ompl::control::EST::addMotion(ompl::control::EST::Motion *
motion) [member function] > compilation error W1005: `Py++` cannot
expose function that takes as > argument/returns instance of non-
public class. Generated code will not > compile.
ompl::control::EST::Motion * ompl::control::EST::selectMotion()
[member function] > compilation error W1005: `Py++` cannot expose
function that takes as > argument/returns instance of non-public
class. Generated code will not > compile.
ompl::control::PlannerDataStorage [class] > warning W1031: `Py++` will
generate class wrapper - user asked to expose non - > public member
function "

zkingston, 2023-04-08:
Seems like the bindings were generated fine, what about the output of `make`?

fishbotics, 2023-04-14:
I am getting the same import error after I build as well. I built and installed using the install script. This was in a clean Docker running Ubuntu 20.04.

gunaa12, 2023-04-15:
I was able to fix this issue. Although it built properly, I just had to import it from python correctly. By this I mean that add the build directory to path in python script using sys. Hope that helps.

closed 2023-04-15 by gunaa12

dbdxnuliba, 2023-05-11:
@gunaa12 ,could you please give the related code about how to do "add the build directory to path in python script using sys."

> I was able to fix this issue. Although it built properly, I just had to import it from python correctly. By this I mean that add the build directory to path in python script using sys. Hope that helps.