ompl/ompl #1019: how should I install the ompl in my anaconda?

rangfeng1, 2023-04-18:
Do you know how should I install the ompl in my anaconda.

zkingston, 2023-04-18:
What issue are you having installing OMPL inside of a conda environment? There should be `conda-forge` package available with `1.5.2`: https://anaconda.org/conda-forge/ompl

rangfeng1, 2023-04-19:
When I use the conda forge package ,I can not import ompl

zkingston, 2023-04-19:
If you want Python bindings, you will have to compile OMPL yourself. You may need to include the conda path in compilation, e.g., `cmake -Bbuild -DCMAKE_PREFIX_PATH=$CONDA_PREFIX .`

zkingston, 2023-05-05:
See https://github.com/ompl/conda-recipes for recipes to build OMPL in a conda environment.

rangfeng1, 2023-05-06:
It have a problem when I use it
(pytorch) wxj@wxj-vpc:~$ conda build -c conda-forge ompl
Glob /home/wxj/ompl did not match in root_dir /home/wxj
ERROR:conda_build.utils:Glob /home/wxj/ompl did not match in root_dir /home/wxj
Traceback (most recent call last):
  File "/home/wxj/anaconda3/bin/conda-build", line 11, in <module>
    sys.exit(main())
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/cli/main_build.py", line 496, in main
    execute(sys.argv[1:])
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/cli/main_build.py", line 485, in execute
    outputs = api.build(args.recipe, post=args.post, test_run_post=args.test_run_post,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/api.py", line 184, in build
    raise ValueError(f'No valid recipes found for input: {recipe_paths_or_metadata}')
ValueError: No valid recipes found for input: ['ompl']

rangfeng1, 2023-05-06:
when use conda build -c conda-forge ompl
Solving environment: ...working... failed
Traceback (most recent call last):
  File "/home/wxj/anaconda3/bin/conda-build", line 11, in <module>
    sys.exit(main())
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/cli/main_build.py", line 496, in main
    execute(sys.argv[1:])
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/cli/main_build.py", line 485, in execute
    outputs = api.build(args.recipe, post=args.post, test_run_post=args.test_run_post,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/api.py", line 186, in build
    return build_tree(
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/build.py", line 3091, in build_tree
    packages_from_this = build(metadata, stats,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/build.py", line 2054, in build
    output_metas = expand_outputs([(m, need_source_download, need_reparse_in_env)])
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/render.py", line 788, in expand_outputs
    for (output_dict, m) in deepcopy(_m).get_output_metadata_set(permit_unsatisfiable_variants=False):
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/metadata.py", line 2121, in get_output_metadata_set
    conda_packages = finalize_outputs_pass(ref_metadata, conda_packages, pass_no=0,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/metadata.py", line 782, in finalize_outputs_pass
    fm = finalize_metadata(om, parent_metadata=parent_metadata,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/render.py", line 546, in finalize_metadata
    build_unsat, host_unsat = add_upstream_pins(m,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/render.py", line 408, in add_upstream_pins
    host_deps, host_unsat, extra_run_specs_from_host = _read_upstream_pin_files(m, 'host',
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/render.py", line 373, in _read_upstream_pin_files
    deps, actions, unsat = get_env_dependencies(m, env, m.config.variant,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/render.py", line 138, in get_env_dependencies
    actions = environ.get_install_actions(tmpdir, tuple(dependencies), env,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda_build/environ.py", line 799, in get_install_actions
    actions = install_actions(prefix, index, specs, force=True)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/common/io.py", line 86, in decorated
    return f(*args, **kwds)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/plan.py", line 476, in install_actions
    txn = solver.solve_for_transaction(prune=prune, ignore_pinned=not pinned)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/core/solve.py", line 156, in solve_for_transaction
    unlink_precs, link_precs = self.solve_for_diff(update_modifier, deps_modifier,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/core/solve.py", line 199, in solve_for_diff
    final_precs = self.solve_final_state(update_modifier, deps_modifier, prune, ignore_pinned,
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/core/solve.py", line 317, in solve_final_state
    ssc = self._add_specs(ssc)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/core/solve.py", line 711, in _add_specs
    conflicts = ssc.r.get_conflicting_specs(tuple(MatchSpec(_)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/resolve.py", line 1110, in get_conflicting_specs
    C = r2.gen_clauses()
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/common/io.py", line 86, in decorated
    return f(*args, **kwds)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/resolve.py", line 913, in gen_clauses
    for ms in self.ms_depends(prec):
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/resolve.py", line 797, in ms_depends
    deps = [MatchSpec(d) for d in prec.combined_depends]
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/models/records.py", line 341, in combined_depends
    result = {ms.name: ms for ms in MatchSpec.merge(self.depends)}
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/models/match_spec.py", line 480, in merge
    reduce(lambda x, y: x._merge(y, union), group) if len(group) > 1 else group[0]
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/models/match_spec.py", line 480, in <lambda>
    reduce(lambda x, y: x._merge(y, union), group) if len(group) > 1 else group[0]
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/models/match_spec.py", line 511, in _merge
    final = this_component.merge(that_component)
  File "/home/wxj/anaconda3/lib/python3.9/site-packages/conda/models/match_spec.py", line 762, in merge
    raise ValueError("Incompatible component merge:\n  - %r\n  - %r"
ValueError: Incompatible component merge:
  - '*mpich*'
  - 'mpi_mpich_*'

meneguzzi, 2023-12-05:
This seems to still be an issue, right? What's the point of the conda package for ompl if it does not get it to be useable in the conda environment?

traversaro, 2025-06-02:
> This seems to still be an issue, right? What's the point of the conda package for ompl if it does not get it to be useable in the conda environment?

Replying after a bit of time, just in case users found this issue on search engines. conda-forge is a multi-language distribution (a bit like Linux distributions like Debian or Fedora), so the point of a C++ package being available (even without Python bindings) is to be usable as a C++ library, for example by C++ developers that install their dependencies via conda-forge.

Habing said that, if anyone is interested in contributing there is an open issue that tracks adding Python bindings to the ompl conda-forge package in https://github.com/conda-forge/ompl-feedstock/issues/3 .