ompl/ompl #1021: By test AIT* we got several log info such as  " Found a new exact solution of cost  ", but the  value of cost first decreases then suddenly increases and then decreases again

dbdxnuliba, 2023-04-20:
AIT* is an optimizing planner, and we test it in  moveit  https://github.com/alaflaquiere/moveit-fork which integrate ompl 1.6.0
We focus on the path length rather than the speed, in moveit  we set  group planning time  5s,in a p2p in joint space plan demo 
[image]

We got several log info such as  " Found a new exact solution of cost  ", but the  value of cost first decreases then suddenly increases and then decreases again,here is the log info  :
[image]
[image]
[image]

we can see from the log that, the cost value is:11.5320->10.0759->10.0755->7.2064-> 16.1385(here suddenly increases,and then decrease)->13.7333->13.1666->12.8491->10.3396->9.2996.
Question:
Why the  value of cost first decreases then suddenly increases and then decreases again in moveit when call AIT*

zkingston, 2023-04-20:
The logs contain two separate planning queries, the first of which goes from around 11 -> 7 cost, the second goes from 16 -> 9. AIT* is planning without knowledge of the prior solution, and finds different solutions each time.

closed 2023-04-20 by zkingston

dbdxnuliba, 2023-04-21:
@alaflaquiere,what your opinio