ompl/ompl #1026: PathGeometric::checkAndRepair problem with path composed of two states

krakjack5, 2023-04-27:
I am using PathGeometric::checkAndRepair method to fix a path that becomes invalid due to an obstacle. The path has only two states - start and goal. When I call checkAndRepair on this path, I always get (true, true) as output. 

Looking through the code, it looks like the error is originating due to the condition used in the for loop on line 234 (PathGeometric.cpp on main, last commit [a02c63f]). When a path has only two states, the for loop on line 234 never gets executed, and hence path does not get repaired with the method returning (true, true). If I add an intermediate state to the path, then checkAndRepair works as expected.

I would appreciate any input in this regard!

Thank you and regards!

ChamzasKonstantinos, 2023-04-27:
Hi @krakjack5 . The obstacle that you are inserting makes either the start and goal invalid or is an obstacle that is in between? 

If either your start, goal is invalid then this function would not work as these are always assumed to be valid and part of the problem definition. 

In the other case, where the object make the edge invalid, but not the endpoints. 
I am not sure the path can be repaired since it only has 2 states, and you cannot change start and goal since it would be a different motion planning problem. 

Would calling [interplolate()](https://ompl.kavrakilab.org/classompl_1_1geometric_1_1PathGeometric.html#a824b91d04ba886c82c1dda3e356fd1f0) before you call checkAndRepair address your usecase?

krakjack5, 2023-04-27:
Hi @ChamzasKonstantinos The way I have set up the test, the obstacle makes the edge invalid and both endpoints are valid. The checkAndRepair method computes a temp state to uniformly sample states around for repair. In the case of an invalid edge (path with only two states), if the for loop on line 234 used the stopping condition of "i <= n1", wouldn't it sample a new state around the goal with radius equal to the length of the edge? That will be able to repair the path as long as the obstacle is small enough.

Calling interpolate does work for my use case, but I was wondering why an edge is effectively treated as a special case.

Thanks!

ChamzasKonstantinos, 2023-04-27:
Yeah, I don't think it is possible to repair a single edge, as the endpoints would be start, and goal and you cannot change them since that would be a different motion planning problem. I think what you are alluding to would be adding an intermediate point that goes away from the obstacle (the output would have 2 edges), but I think an underlying assumption of this function is that the number of waypoints cannot change.

krakjack5, 2023-04-27:
@ChamzasKonstantinos  Ok, that makes sense; I missed the fact that no new additional states are added during repair. Thank you for that clarification!

I have another sort of related question if you don't mind. I used interpolate and it works fine. However, I think my understanding of interpolate is not clear. I had set up the validity check such that interpolate added 42 states, thus making the total number of states 44. Just to try, I called interpolate again, and that made the total number of states equal to 224. I thought interpolate worked such that total number of states will be approximately similar to that used for validity checking. I did not understand why after calling interpolate the second time, the number of states jumped from 44 to 224.

Thanks!

krakjack5, 2023-04-28:
Never mind my last comment about the interpolate method. The discrepancy was of my doing. 

Coming back to the checkAndRepair method, I think the behavior in the case of a path with only two states is not intuitive in terms of the output returned by the method. The comments in PathGeometric.h state 

"\brief Check if the path is valid. If it is not,  attempts are made to fix the path by sampling around invalid states." 

Clearly, for the case of a path with only two states, the path does not get checked for validity. I was wondering if it will be useful to update the comments accordingly. Another option may be to handle this as a separate case, similar to how the case of a path with only a single state is handled. In that case, the method will return a pair where pair.first = path.check() and pair.second = false.

I will appreciate any input!

zkingston, 2023-04-28:
We should add the special case to make sure the two state case gets handled appropriately. Feel free to submit a PR to address this.

PR #1048 (2023-05-26): Add a special case to checkAndRepair for the two state case
Fixes #1026. Special case for two state case to validate motion, as it won't be checked otherwise.
closed 2023-05-26 by zkingston