ompl/ompl #1027: Error:   RRTstar: There are no valid initial states!

YaboCheung, 2023-04-28:
I continue to encounter problems like "Error:   RRTstar: There are no valid initial states!" 
here are the results:

Info:    RRTstar: Space information setup was not yet called. Calling now.
Debug:   RRTstar: Planner range detected to be 8.808863
Info:    RRTstar: No optimization objective specified. Defaulting to optimizing path length for the allowed planning time.
Warning: RRTstar: Skipping invalid start state (invalid state)
         at line 249 in /home/nvidia/ompl/src/ompl/base/src/Planner.cpp
Debug:   RRTstar: Discarded start state Compound state [
RealVectorState [1.85227 0.940515 0.356055]
SO3State [0 0 0 1]
]
Error:   RRTstar: There are no valid initial states!
         at line 193 in /home/nvidia/ompl/src/ompl/geometric/planners/rrt/src/RRTstar.cpp

Can anyone help?

zkingston, 2023-04-28:
This implies that your starting configuration is not valid according to your state validity checker. Double check and inspect by explicitly calling the state validity checker's `isValid` on your starting state.

zkingston, 2023-05-05:
Closing for now unless there is follow up.

closed 2023-05-05 by zkingston

YaboCheung, 2023-05-06:
I just followed what u say to double check the isValid func and correct it!
tks a lot,