ompl/ompl #1031: Error while running ConstrainedPlanningTorus.py

jacobjj, 2023-05-06:
When I ran the python `ConstrainedPlanningTorus.py` example I get the following error

```
Info:    Using Projection-Based State Space!
Start =  [-0.88587213  0.94680376 -0.71080768]
Goal =  [2.06211537 1.7856852  0.68577026]
Debug:   RRT: Planner range detected to be 2.078461
Info:    RRT: Starting planning with 1 states already in datastructure
Traceback (most recent call last):
  File "ConstrainedPlanningTorus.py", line 210, in <module>
    torusPlanning(parser.parse_args())
  File "ConstrainedPlanningTorus.py", line 185, in torusPlanning
    torusPlanningOnce(cp, planners[0], options.output)
  File "ConstrainedPlanningTorus.py", line 140, in torusPlanningOnce
    stat = cp.solveOnce(output, "torus")
  File "/ompl/demos/constraint/ConstrainedPlanningCommon.py", line 215, in solveOnce
    stat = self.ss.solve(self.options.time)
TypeError: No Python class registered for C++ class Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> >

```

I believe this was reported earlier in #1007, and fixed in #1008, but it doesn't seem to work.

Undoing the changes in #1008 to `generate_bindings.py` resolved the issue.

zkingston, 2023-05-06:
Thanks for reporting. Probably both signatures (before and after #1008) need to be attempted in the binding generation.

zkingston, 2023-05-08:
See if #1031 works for you.

jacobjj, 2023-05-08:
This works; thanks.

PR #1033 (2023-05-08): Add back in alternative signature for Python bindings for constrained planning
Fixes #1031
closed 2023-05-08 by jacobjj