ompl/ompl #1032: /usr/bin/ld: ../lib/libompl.so.1.6.0: undefined reference to `dBodyGetQuaternion(dxBody*)'

dbdxnuliba, 2023-05-08:
dell@dell-NUC8i7BEH:~/refer/ompl/build/Release$ cmake ../.. -DPYTHON_EXEC=/usr/bin/python3.8
dell@dell-NUC8i7BEH:~/refer/ompl/build/Release$ make -j 4 update_bindings
dell@dell-NUC8i7BEH:~/refer/ompl/build/Release$ make -j 4

[image]

zkingston, 2023-05-08:
It appears again that `ode` is missing somehow from your installation. For ubuntu, you should be able to install `ode` with `apt-get install libode-dev`. Confirm that `ode` is installed with `pkg-config --libs ode` which should give you something like `-lode`. If that appears correct, look at your `CMakeCache.txt` in the `build/` directory and find the section with the `ODE_*` variables: these should be filled in.

dbdxnuliba, 2023-05-09:
@zkingston  yes ,dell@dell-NUC8i7BEH:~/refer$ pkg-config --libs ode
-lode

and in CMakeCache.txt in the build/Release directory 
[image]
[image]
 is 
so what should I fill in the ODE_*

zkingston, 2023-05-26:
Closing your current issues, please open another one if you still have troubles with installation.

closed 2023-05-26 by zkingston

dbdxnuliba, 2023-06-01:
ok