ompl/ompl #1035: [update_bindings](make: *** No rules to make target 'update_bindings'. stop.)

dbdxnuliba, 2023-05-09:
ubuntu18.04 melodic python3:python3.8
[update_bindings](make: *** No rules to make target 'update_bindings'. stop.)
~/ref/ompl$ bash install-ompl-ubuntu.sh.in --python

[image]
[image]

[update_bindings](make: *** No rules to make target 'update_bindings'. stop.)

zkingston, 2023-05-09:
It appears that `numpy` is missing from your system, please install and try again. Complete remove the build directory and create another.

dbdxnuliba, 2023-05-10:
after sudo apt-get install python3.8-dev ,it can continue compile,thanks

closed 2023-05-10 by dbdxnuliba

Shihao-Feng-98, 2023-10-24:
I have the same error in ubuntu20, and I also have a question about how to install pybindings in conda?