ompl/ompl #1036: FileNotFoundError: [Errno 2] No such file or directory: 'bindings/base

dbdxnuliba, 2023-05-10:
-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/dell/ref/ompl/ompl-db4e6bf581cd531462ce8a3fa1889a659bd1c4b0/build/Release
[100%] Built target update_control_bindings
[100%] Creating C++ code for Python module tools (see pyplusplus_tools.log)
Replacing member function ::ompl::base::OptimizationObjective::print
Replacing member function ::ompl::base::GoalRegion::print
Replacing member function ::ompl::base::Goal::print
Replacing member function ::ompl::base::ProblemDefinition::print
Replacing member function
::ompl::base::ScopedState<ompl::base::StateSpace>::print
Replacing member function ::ompl::base::Path::print
Replacing member function ::ompl::base::ProjectionMatrix::print
Replacing member function ::ompl::base::ParamSet::print
Replacing member function ::ompl::tools::SelfConfig::print
Replacing member function ::ompl::geometric::PathGeometric::print
Replacing member function ::ompl::geometric::SimpleSetup::print
Replacing member function ::ompl::control::SimpleSetup::print
Replacing member function ::ompl::control::PathControl::print
Replacing member function
::ompl::tools::Benchmark::saveResultsToStream
Replacing member function
::ompl::tools::Benchmark::setPlannerSwitchEvent
Replacing member function ::ompl::tools::Benchmark::setPreRunEvent
Replacing member function ::ompl::tools::Benchmark::setPostRunEvent
Traceback (most recent call last):
  File "/home/dell/ref/ompl/ompl-db4e6bf581cd531462ce8a3fa1889a659bd1c4b0/py-bindings/generate_bindings.py", line 983, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/dell/ref/ompl/ompl-db4e6bf581cd531462ce8a3fa1889a659bd1c4b0/py-bindings/generate_bindings.py", line 880, in __init__
    code_generator_t.__init__(self, 'tools', \
  File "/home/dell/ref/ompl/ompl-db4e6bf581cd531462ce8a3fa1889a659bd1c4b0/py-bindings/ompl/bindings_generator.py", line 154, in __init__
    self.mb.register_module_dependency(dep)
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/module_builder/boost_python_builder.py", line 142, in register_module_dependency
    db.load( other_module_generated_code_dir )
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/utils/__init__.py", line 129, in load
    f = open( fpath, 'r+b' )
**FileNotFoundError: [Errno 2] No such file or directory: 'bindings/base'**
-- Build type: Release
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

zkingston, 2023-05-10:
Can you post the entire output of `make update_bindings`? The error you are seeing is just a symptom, not the main cause of error.

dbdxnuliba, 2023-05-11:
```
dell@dell-NUC11PAHi7:~/ref/ompl-1.6.0/build/Release$ make update_bindings
[  0%] Preparing C++ header file for Python binding generation for module control
[  0%] Built target control.h
[  0%] Preparing C++ header file for Python binding generation for module base
[  0%] Built target base.h
[ 33%] Preparing C++ header file for Python binding generation for module util
[ 33%] Built target util.h
[ 33%] Creating C++ code for Python module util (see pyplusplus_util.log)
std::vector<double, std::allocator<double> >::reference
std::vector<double, std::allocator<double> >::at(std::size_t __n)
[member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<double, std::allocator<double> >::reference
std::vector<double, std::allocator<double> >::front() [member
function] > warning W1008: The function returns non-const reference to
"Python immutable" > type. The value cannot be modified from Python.
std::vector<double, std::allocator<double> >::reference
std::vector<double, std::allocator<double> >::back() [member function]
> warning W1008: The function returns non-const reference to "Python
immutable" > type. The value cannot be modified from Python.
std::vector<double, std::allocator<double> >::pointer
std::vector<double, std::allocator<double>
>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const double *, std::vector<double,
std::allocator<double> > > __first, __gnu_cxx::__normal_iterator<const
double *, std::vector<double, std::allocator<double> > > __last)
[member function] > compilation error W1050: The function returns
"std::vector<double, > std::allocator<double> >::pointer" type. You
have to specify a call > policies.Be sure to take a look on `Py++`
defined call policies
std::vector<unsigned int, std::allocator<unsigned int> >::reference
std::vector<unsigned int, std::allocator<unsigned int>
>::at(std::size_t __n) [member function] > warning W1008: The function
returns non-const reference to "Python immutable" > type. The value
cannot be modified from Python.
std::vector<unsigned int, std::allocator<unsigned int> >::reference
std::vector<unsigned int, std::allocator<unsigned int> >::front()
[member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<unsigned int, std::allocator<unsigned int> >::reference
std::vector<unsigned int, std::allocator<unsigned int> >::back()
[member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<unsigned int, std::allocator<unsigned int> >::pointer
std::vector<unsigned int, std::allocator<unsigned int>
>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const unsigned int *,
std::vector<unsigned int, std::allocator<unsigned int> > > __first,
__gnu_cxx::__normal_iterator<const unsigned int *,
std::vector<unsigned int, std::allocator<unsigned int> > > __last)
[member function] > compilation error W1050: The function returns
"std::vector<unsigned int, > std::allocator<unsigned int> >::pointer"
type. You have to specify a call > policies.Be sure to take a look on
`Py++` defined call policies
std::vector<ompl::PPM::Color, std::allocator<ompl::PPM::Color>
>::pointer std::vector<ompl::PPM::Color,
std::allocator<ompl::PPM::Color> >::_M_allocate_and_copy(std::size_t
__n, __gnu_cxx::__normal_iterator<const ompl::PPM::Color *,
std::vector<ompl::PPM::Color, std::allocator<ompl::PPM::Color> > >
__first, __gnu_cxx::__normal_iterator<const ompl::PPM::Color *,
std::vector<ompl::PPM::Color, std::allocator<ompl::PPM::Color> > >
__last) [member function] > compilation error W1050: The function
returns "std::vector<ompl::PPM::Color, >
std::allocator<ompl::PPM::Color> >::pointer" type. You have to specify
a call > policies.Be sure to take a look on `Py++` defined call
policies
[... 519 lines cut ...]
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/dell/ref/ompl-1.6.0/build/Release
[100%] Built target update_control_bindings
[100%] Built target update_bindings

```

zkingston, 2023-05-11:
It seems you are compiling against the 1.6.0 release of OMPL, please update to the most recent version on the main branch of this repository, the error you are seeing was fixed in #980.

dbdxnuliba, 2023-05-15:
ok thanks