ompl/ompl #1042: Obstacle checking resolution for RRT

MohiniMohiniMohini, 2023-05-19:
I am using SE2 state space with 0,240 and 0,240 as the x and y min max bounds respectively. statePropogator is overridden and set. Now i want the vehicle to move for 1 sec in the direction of the new random node and for that 1 sec motion, i want to check the obstacles every 0.002 sec. Also, i want the RRT to progress one step size at a time. is this possible? i've tried it this way but failed.

spaceInformation->setStatePropagator(bind(&propagateEgo, placeholders::_1, placeholders::_2, placeholders::_3, placeholders::_4));
spaceInformation->setPropagationStepSize(1.0);
spaceInformation->setMinMaxControlDuration(1.0, 1.0);
stateSpace->as<ob::SE2StateSpace>()->setLongestValidSegmentFraction(0.002);
[image]

zkingston, 2023-05-19:
If you want to check every 0.002s, you should set your propagation step size to this value. If you want to always propagate for 1 seconds, you should set the min/max control duration to the multiple of this value (500). Note the min/max control duration parameters are integers - these are the # of steps that propagation is applied, using the propagation step size. That is,
```cpp
spaceInformation->setPropagationStepSize(0.002);
spaceInformation->setMinMaxControlDuration(500, 500);
```

closed 2023-05-19 by zkingston

MohiniMohiniMohini, 2023-05-22:
@zkingston thanks for the reply. With these settings and not using the method: stateSpace->asob::SE2StateSpace()->setLongestValidSegmentFraction(0.002);, here is output:

Initialized: 
Info:    RRT: Starting planning with 1 states already in datastructure
Info:    ProblemDefinition: Adding approximate solution from planner RRT
Info:    RRT: Created 3455 states
Found solution!
Time required by RRT is: 1.00318
 number of vertices: 3455
state count: 2
Start yaw: 1.5708
Expected Goal yaw: 1.5708 Actual yaw: 0
*** stack smashing detected ***: terminated
[ros2run]: Aborted

[image]

MohiniMohiniMohini, 2023-05-22:
also, I want to match the goal orientation, which is currently ignored by RRT:

Start yaw: 1.5708
Expected Goal yaw: 1.5708, Actual Goal yaw: 0

Here is the code to print this:
og::PathGeometric* path = pdef->getSolutionPath()->as<og::PathGeometric>();
size_t stateCount = path->getStateCount();
cout << "state count: " << stateCount << endl;
cout << "Start yaw: " << start->getYaw() << endl;
// cast the abstract state type to the type we expect
const ob::SE2StateSpace::StateType *se2state = path->getState(stateCount-1)->as<ob::SE2StateSpace::StateType>();
// extract the first component of the state and cast it to what we expect
const ob::RealVectorStateSpace::StateType *pos = se2state->as<ob::RealVectorStateSpace::StateType>(0);
cout << "Expected Goal yaw: " << goal->getYaw() << " Actual Goal yaw: " << pos->values[2] << endl;

MohiniMohiniMohini, 2023-05-22:
@zkingston any insights on the last prints when the code is run?

*** stack smashing detected ***: terminated
[ros2run]: Aborted

MohiniMohiniMohini, 2023-05-23:
@zkingston any comment I'm unable to get the results as per our suggestions. is there any way i can re-open this issue?

zkingston, 2023-05-23:
`*** stack smashing detected ***` usually indicates a buffer overflow somewhere in your code.
Also, you should compare `goal->getYaw()` and `se2State->getYaw()`

MohiniMohiniMohini, 2023-05-24:
Hello @zkingston thanks for the reply to the 2 questions, will you please answer the 3rd question? I am getting start and goal state as solution. I've uploaded an image for the same

MohiniMohiniMohini, 2023-05-24:
> ` Also, you should compare `goal->getYaw()` and `se2State->getYaw()`
even with this correction, the orientation does not match. How can i match the goal orientation?

MohiniMohiniMohini, 2023-05-25:
@zkingston any thoughts, please?

MohiniMohiniMohini, 2023-05-26:
@zkingston i've attached 2 files, one is main.cpp zip and the other is a ros node zip to its minimum form. Its a blank scenario.
 
the issues is, the RRT returns only 2 state path i.e. start and goal state.

[ompl_pkg_issue.zip](https://github.com/ompl/ompl/files/11576206/ompl_pkg_issue.zip)
[main.zip](https://github.com/ompl/ompl/files/11576208/main.zip)

MohiniMohiniMohini, 2023-05-30:
Hello @zkingston do you need any more infomation?

MohiniMohiniMohini, 2023-06-05:
Hello @zkingston, is there a way to reopen this issue? Shall I create a new issue?

zkingston, 2023-06-05:
Taking a quick look at your code & the messages you posted, the issue seems to be you are only getting approximate solutions. Check if `solved == EXACT_SOLUTION` if you want to avoid this case. You should also add tolerance to the goal `pdef->setStartAndGoalStates(start, goal, <tolerance>)` as control planners will never exactly reach the goal.

MohiniMohiniMohini, 2023-06-08:
Hello @zkingston , thanks for the help, it is working now.