ompl/ompl #1043: Allow reusing PlannerData in EIRM*

simonschmeisser, 2023-05-22:
As noted in https://github.com/ros-planning/moveit/pull/3437  EIRM* currently does not have a constructor accepting and using `PlannerData`. MoveIt uses this to store roadmaps to disk and later reuse them.

Would that be hard to implement @marlinstrub ?

yurirocha15, 2023-05-22:
I was also taking a look at this and the EIT* base class already implements the [`getPlannerData`](https://github.com/ompl/ompl/blob/main/src/ompl/geometric/planners/informedtrees/src/EITstar.cpp#L449-L470) function.  I believe it can be used as a reference on how to load the data in the constructor.

vhartman, 2023-07-25:
I think this should not be too hard - however, as far as I know there is no way to store edge validity in the `PlannerData`, therefore making much of the reuse of information that EIRM* does impossible.

Do you have a specific usecase?