ompl/ompl #1044: Unexpected intepolation with DubinsStateSpace

krakjack5, 2023-05-22:
I am running into an expected behavior with interpolation of a path when using with DubinsStateSpace. I have an original path planned by a planner. If I interpolate that path, result is as expected. However, I am creating a new PathGeometric by translating that original path in x and y coordinates (I am creating a completely new Dubin's state space with its corresponding bounds for this translated path). If I interpolate the tranlated path with this newly created state space, it adds unexpected circular motions between certain states (please see the attached plots). My initial thought was that may be the yaw angles were slightly off between the two paths, but I have verified that yaw angles are the same for the two paths for all states. Could something be different between the two state spaces (they use the same turning radius) that could explain this result with unexpected circular motions?

I am using 1.6.0 release branch.

The first plot below shows the original and the translated path. The other plot shows results of interpolation.

[image]
[image]

zkingston, 2023-05-22:
Could you provide a minimum example script that demonstrates the problem?

zkingston, 2023-06-02:
Closing for now unless there is follow up.

closed 2023-06-02 by zkingston

krakjack5, 2023-06-29:
Sorry for such a late reply. I figured out what the problem was, and the issue was not caused by interpolation through OMPL code. I was computing a path in local coordinates and then converting it into lat/lon. Before translating the path, I converted it from lat/lon to local coordinates. This conversion resulted in very small numerical differences in coordinate position (less than 8th decimal place) but that indeed caused interpolation to take longer circular path between a few of those successive states.