ompl/ompl #1046: How to take into account max speed joint in motion planning ?

sancelot, 2023-05-24:
I am wondering how I could estimate the time of a planned trajectory, knowing the max allowed speed of each joint, and using a global speed  percentage value, using this code template : 

```
#include <ompl/base/ScopedState.h>
#include <ompl/base/spaces/RealVectorStateSpace.h>
#include <ompl/geometric/SimpleSetup.h>
#include <ompl/geometric/planners/rrt/RRT.h>

namespace ob = ompl::base;
namespace og = ompl::geometric;

// Define the robot's state validity checker
bool isStateValid(const ob::State* state)
{
    // Implement your state validity checking logic here
    // This function should return true if the state is valid, and false otherwise
}

// Compute a linear trajectory with a given speed percentage
void computeLinearTrajectory(double start[], double goal[], double maxSpeed, double speedPercentage)
{
    // Create a state space (for a 3D robot with 6 joints)
    ob::StateSpacePtr space(new ob::RealVectorStateSpace(6));
  
    // Set bounds for each joint
    ob::RealVectorBounds bounds(6);
    bounds.setLow(-3.14159);  // Example lower bound for joint angles
    bounds.setHigh(3.14159);  // Example upper bound for joint angles
    space->as<ob::RealVectorStateSpace>()->setBounds(bounds);
  
    // Create a simple setup
    og::SimpleSetup ss(space);
  
    // Set the state validity checker
    ss.setStateValidityChecker(isStateValid);
  
    // Create a start state
    ob::ScopedState<> startState(space);
    for (int i = 0; i < 6; ++i)
        startState[i] = start[i];
  
    // Create a goal state
    ob::ScopedState<> goalState(space);
    for (int i = 0; i < 6; ++i)
        goalState[i] = goal[i];
  
    // Set the start and goal states
    ss.setStartAndGoalStates(startState, goalState);
  

  
    // Set the planner
    ob::PlannerPtr planner(new og::RRT(ss.getSpaceInformation()));
    ss.setPlanner(planner);
  
    // Set the planner termination condition
    ob::PlannerTerminationCondition ptc = ob::timedPlannerTerminationCondition(1.0);  // Plan for 1 second
  
    // Attempt to solve the problem within the time limit
    ss.solve(ptc);
  
    // Get the computed path
    og::PathGeometric path = ss.getSolutionPath();
  
    // Print the waypoints along the path
    std::cout << "Computed path with " << path.getStateCount() << " waypoints:" << std::endl;
    for (std::size_t i = 0; i < path.getStateCount(); ++i)
    {
        const ob::State* state = path.getState(i);
        const double* values = state->as<ob::RealVectorStateSpace::StateType>()->values;
        std::cout << "Waypoint " << i << ": ";
        for (int j = 0; j < 6; ++j)
            std::cout << values[j] << " ";
        std::cout << std::endl;
    }
}
```

sancelot, 2023-05-30:
I found the pb , this is a call to parent class in overloaded function (the function does not exist in parent class...):
        return ob::MotionValidator::checkMotion(s1, s2, lastValid);
The following piece of code works, however, I don't understand how to avoid computing unuseful trajectory for joints that are already at goal position  (joints where start pos = goal pos...) ???

```
#include <ompl/base/ScopedState.h>
#include <ompl/base/spaces/RealVectorStateSpace.h>
#include <ompl/base/MotionValidator.h>

#include <ompl/geometric/SimpleSetup.h>
#include <ompl/geometric/planners/rrt/RRT.h>

namespace ob = ompl::base;
namespace og = ompl::geometric;
double speedPercentage = 100;
#define radians(x) (x*3.14159/180.)
/*
If you know the maximum speed of all the robot joints and would like to compute a Cartesian linear trajectory using a general speed percentage, 
you can use an inverse kinematics solver to compute the joint angles corresponding to the start and goal positions of the end-effector. 
Once you have computed the start and goal joint angles, you can use a motion planning library such as OMPL to compute a linear trajectory in joint space that satisfies the maximum speed constraints for each joint.
*/
double maxSpeeds[6] = {radians(160), radians(120), radians(120),radians(225), radians(225), radians(225)}; // Maximum speed for each joint deg/sec
 

/*
This custom MotionValidator takes as input an array of maximum speeds for each joint and a desired speed percentage. 
It overrides the checkMotion method to reject motions where the distance between two consecutive states is greater than maxSpeeds[i] * speedPercentage for any joint i.*/
class MyMotionValidator : public ob::MotionValidator
{
  public:
    explicit MyMotionValidator(const ob::SpaceInformationPtr &si, double maxSpeeds[], double speedPercentage)
        : ob::MotionValidator(si), maxSpeeds_(maxSpeeds), speedPercentage_(speedPercentage)
    {
    }

    //virtual bool checkMotion(const ob::State* s1, const ob::State* s2) const
    virtual bool checkMotion(const ob::State *s1, const ob::State *s2) const override
    {
        // Check if the distance between the two states is greater than the maximum allowed distance for each joint
        const double *values1 = s1->as<ob::RealVectorStateSpace::StateType>()->values;
        const double *values2 = s2->as<ob::RealVectorStateSpace::StateType>()->values;
        for (int i = 0; i < 6; ++i)
        {
            if (std::abs(values2[i] - values1[i]) > maxSpeeds_[i] * speedPercentage_)
                return false;
        }

        // Use the default motion validation logic
        // return ob::MotionValidator::checkMotion(s1, s2);
        return true;
    }
 
    virtual bool checkMotion(const ompl::base::State *s1, const ompl::base::State *s2, std::pair<ompl::base::State *, double> &lastValid) const override
    {
        
        // Check if the distance between the two states is greater than the maximum allowed distance for each joint
        const double *values1 = s1->as<ob::RealVectorStateSpace::StateType>()->values;
        const double *values2 = s2->as<ob::RealVectorStateSpace::StateType>()->values;
        for (int i = 0; i < 6; ++i)
        {
            if (std::abs(values2[i] - values1[i]) > maxSpeeds_[i] * speedPercentage_)
                return false;
        }
        return true;
        // Use the default motion validation logic
       // return ob::MotionValidator::checkMotion(s1, s2, lastValid);
    }

  private:
    double *maxSpeeds_;
[... 99 lines cut ...]

    // Print the waypoints along the path
    std::cout << "Computed path with " << path.getStateCount() << " waypoints:" << std::endl;
    for (std::size_t i = 0; i < path.getStateCount(); ++i)
    {
        const ob::State *state = path.getState(i);
        const double *values = state->as<ob::RealVectorStateSpace::StateType>()->values;
        std::cout << "Waypoint " << i << ": ";
        for (int j = 0; j < 6; ++j)
            std::cout << values[j] * 180. / 3.141549<< " ";
        std::cout << std::endl;
    }
}
#

int main()
{
    // joints 
    double start[] = {0., radians(20.), 0, 0, 0, 0};
    double goal[] = {0, radians(50.),radians(360), 0, 0, 0};

    plan(start, goal, 100, 100);

}
```

zkingston, 2023-06-02:
The algorithms in OMPL are sampling-based and thus will sometimes include spurious motion. I would use the simplification routines available (e.g., call `ss.simplifySolution()`) before extracting the solution trajectory and seeing if a better path can be found with local search.

closed 2023-06-02 by zkingston