ompl/ompl #1053: The PRMstar planner throws ompl::Exception on infeasible solutions when constructed from previous PlannerData

Hugal31, 2023-06-02:
When I create a PRMstar planner using PlannerData from a previous run and call `solve` with an infeasible plan, it throws an `ompl::Exception("Could not find solution path")` from [here](https://github.com/ompl/ompl/blob/48523a77397918a33a1fc811a3230124e0d18d08/src/ompl/geometric/planners/prm/src/PRM.cpp#LL710C64-L710C64).

Here the plan is infeasible due to a split in the valid state space.

I don't understand what's exactly is the issue here. Either the graph is wrongly stored/loaded, and this should be fixed, either the exception is normal, but this is not documented in `Planner::solve`, and I think the error should be caught in `maybeConstructSolution` or in `checkForSolution`. I do not fully understand the PRM so I cannot make this decision.

Here is a minimal example :

```cpp
#include <ompl/base/SpaceInformation.h>
#include <ompl/base/spaces/RealVectorStateSpace.h>
#include <ompl/geometric/planners/prm/PRMstar.h>

ompl::base::PlannerStatus tryToSolve(ompl::base::Planner &planner, ompl::base::SpaceInformationPtr const &si, double start, double goal)
{
  auto startState = si->allocState()->as<ompl::base::RealVectorStateSpace::StateType>(),
       goalState = si->allocState()->as<ompl::base::RealVectorStateSpace::StateType>();
  startState->values[0] = start;
  goalState->values[0] = goal;
  auto problem = std::make_shared<ompl::base::ProblemDefinition>(si);
  problem->setStartAndGoalStates(startState, goalState);
  planner.setProblemDefinition(problem);
  if (!planner.isSetup())
    planner.setup();
  return planner.solve(1.0);
}

int main()
{
  ompl::base::RealVectorBounds bounds {1};
  bounds.setLow(0, 0);
  bounds.setHigh(0, 10);

  // Create a 1D state space between 0 and 10 with an obstacle between 3 and 4.
  auto stateSpace = std::make_shared<ompl::base::RealVectorStateSpace>(1);
  stateSpace->setBounds(bounds);
  auto spaceInformation = std::make_shared<ompl::base::SpaceInformation>(stateSpace);
  spaceInformation->setStateValidityChecker([](ompl::base::State const *state)
                                            {
                                              auto const *state1D = state->as<ompl::base::RealVectorStateSpace::StateType>();
                                              return state1D->values[0] < 3 || state1D->values[0] > 4;
                                            });

  // Interestingly, if I change this to 0.5, the error goes away.
  stateSpace->setLongestValidSegmentFraction(0.1);
  spaceInformation->setup();

  // Create the first PRMstar and make it grow its graph.
  ompl::geometric::PRMstar prm1 {spaceInformation};
  tryToSolve(prm1, spaceInformation, 1, 6);
  ompl::base::PlannerData data1 {spaceInformation};
  prm1.getPlannerData(data1);
  
  // Use the graph from the previous PRMstar.
  ompl::geometric::PRMstar prm2 {data1};
  ompl::base::PlannerData data2 {spaceInformation};
  prm1.getPlannerData(data2);

  tryToSolve(prm2, spaceInformation, 1, 6);

  return 0;
}

```

I originally had the bug using MoveIt! 1.11.1 with OMPL 1.6.0.

zkingston, 2023-11-28:
Was this fixed with #1112?

Hugal31, 2023-11-29:
Yes, thank you!

closed 2023-11-29 by Hugal31