ompl/ompl #1057: Optimal paths between semi-constrained states

peik-etzell, 2023-06-12:
Hello!

I'm planning for a 6DOF arm on a 3DOF gantry, most interested in semi-constrained cartesian movements (moving the end-effector along a path, with an optimal orientation), and I am getting quite good results using the constrained planning pipeline through MoveIt2. 

MoveIt2 does not expose the multi-goal planning that seems to be available with PRM planners though. I'd also be interested in having multiple start states,  i.e. planning {any start configuration with EEF at start} => {any goal configuration with EEF at goal}. 

I noticed at least ProblemDefinition has support for multiple start and goal states, but does any planner support this though? Do you have any intuition how this would scale? and would it be a generally bad idea to add _many_ start and goal states to a request? Or is there some other mechanism which would enable semi-constrained starts and targets? 

Thank you in advance.

zkingston, 2023-06-12:
Yes, most planners support multiple starts and goals (e.g., PRM, RRT, RRTConnect), although some may only add one start state or one goal state, or simply sample goal states up to some heuristic amount. There is a trade-off to adding too many starts and goals as a) this adds extra complexity to evaluating if a query is complete, all pairs need to be checked for example in PRM and b) can bias search in possibly bad ways, e.g., RRT has multiple start trees when there are multiple start configurations, leading to possibly interesting tree expansion behavior. It's not necessarily a bad idea, but would require evaluation for your specific problem context. I think for planning constrained motions for a gantry like this multiple starts/goals would be helpful, especially given that the constraint on the system might cause disconnected components of the configuration space.

I believe MoveIt supports multiple configuration goals if you specify the goal as a position constraint, e.g., some desired end-effector pose. This results in the goal being a lazily sampled goal region that will sample many different satisfying IK configurations, i.e., multiple goal configurations.

closed 2023-06-12 by zkingston

peik-etzell, 2023-06-14:
Thank you for the detailed answer! This sounds good, I didn't realize even RRTConnect would be theoretically supporting this kind of planning. I'll have to experiment some more with this then.

I suppose the constrained goal region is  a continuous space, even if it is sampled lazily (is it sampled at the closest point maybe?), is there a reason the start positions could not also be such a region? I expect this is a lot harder though.