ompl/ompl #1059: Issue with install Python Bindings

sujaygarlanka, 2023-06-12:
Hello,

The cmake script is not able to find pyplusplus and pygccxml even though I installed them. I got the following message after installing them. I also have the PYTHON_EXEC path pointing to the same environment as is shown in my CMakeCache.txt below. Any help would be really appreciated. Thanks!

```Using pip 22.3.1 from /scr/garlanka/miniconda3/envs/omnigibson/lib/python3.7/site-packages/pip (python 3.7)
Requirement already satisfied: pygccxml in /scr/garlanka/miniconda3/envs/omnigibson/lib/python3.7/site-packages (2.2.1)
Requirement already satisfied: pyplusplus in /scr/garlanka/miniconda3/envs/omnigibson/lib/python3.7/site-packages (1.8.5)

CMakeCache.txt
```
# This is the CMakeCache file.
# For build in directory: /scr/garlanka/ompl-1.6.0/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy37 library (debug)
Boost_NUMPY37_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY37_LIBRARY_DEBUG-NOTFOUND

//Boost numpy37 library (release)
Boost_NUMPY37_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY37_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY3_LIBRARY_DEBUG-NOTFOUND

//Boost numpy3 library (release)
Boost_NUMPY3_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY3_LIBRARY_RELEASE-NOTFOUND

//Boost numpy library (debug)
Boost_NUMPY_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY_LIBRARY_DEBUG-NOTFOUND

//Boost numpy library (release)
Boost_NUMPY_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY_LIBRARY_RELEASE-NOTFOUND

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py37 library (debug)
Boost_PYTHON-PY37_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY37_LIBRARY_DEBUG-NOTFOUND

//Boost python-py37 library (release)
Boost_PYTHON-PY37_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY37_LIBRARY_RELEASE-NOTFOUND
[... 727 lines cut ...]
//Components requested for this build tree.
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;numpy;numpy3;numpy37;program_options;python;python-py37;python3;python37;serialization;system;unit_test_framework
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr//usr/lib/x86_64-linux-gnu

zkingston, 2023-06-12:
It appears that you are building in a conda environment. You will need to specify the conda prefix / boost root directory, e.g., `cmake -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" ...`.
Also, you do not have `boost-numpy` installed, which is required for the Python bindings. Make sure that all of boost is installed.

zkingston, 2023-06-12:
It does, but the issue is your are compiling with Python in a conda environment. The Python and boost versions need to match, and the boost found from a distribution's package manager provides matching boost-python and boost-python-numpy versions for the system installed Python, not whatever version you have in conda.

sujaygarlanka, 2023-06-14:
Ran the command below with boost in my conda environment and got the following error. Do know what may be the issue?

### Command
```
cmake ../.. -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" -DPYTHON_EXEC=/scr/garlanka/miniconda3/envs/omnigibson/bin/python3.7 -DPYTHONPATH=/scr/garlanka/miniconda3/envs/omnigibson/lib/python3.7/site-packages
```

### Conda environment
```
# packages in environment at /scr/garlanka/miniconda3/envs/omnigibson:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
addict                    2.4.0                    pypi_0    pypi
attrs                     23.1.0                   pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
bddl                      3.0.0b1                  pypi_0    pypi
boost                     1.73.0          py37h06a4308_12  
bzip2                     1.0.8                h7f98852_4    conda-forge
ca-certificates           2023.5.7             hbcca054_0    conda-forge
castxml                   0.4.5                    pypi_0    pypi
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1                   pypi_0    pypi
cloudpickle               2.2.1                    pypi_0    pypi
cryptography              41.0.1                   pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
exceptiongroup            1.1.1                    pypi_0    pypi
fastjsonschema            2.17.1                   pypi_0    pypi
future                    0.18.3                   pypi_0    pypi
gitdb                     4.0.10                   pypi_0    pypi
gitpython                 3.1.31                   pypi_0    pypi
gym                       0.26.2                   pypi_0    pypi
gym-notices               0.0.8                    pypi_0    pypi
h5py                      3.8.0                    pypi_0    pypi
icu                       58.2              hf484d3e_1000    conda-forge
imageio                   2.31.0                   pypi_0    pypi
imageio-ffmpeg            0.4.8                    pypi_0    pypi
importlib-metadata        6.6.0                    pypi_0    pypi
importlib-resources       5.12.0                   pypi_0    pypi
iniconfig                 2.0.0                    pypi_0    pypi
ipython                   7.34.0                   pypi_0    pypi
jedi                      0.18.2                   pypi_0    pypi
jsonschema                4.17.3                   pypi_0    pypi
jupyter-core              4.12.0                   pypi_0    pypi
jupytext                  1.14.6                   pypi_0    pypi
ld_impl_linux-64          2.38                 h1181459_1  
libboost                  1.73.0              h28710b8_12  
libffi                    3.4.4                h6a678d5_0  
libgcc-ng                 11.2.0               h1234567_1  
libgomp                   11.2.0               h1234567_1  
libstdcxx-ng              11.2.0               h1234567_1  
lz4-c                     1.9.4                h6a678d5_0  
markdown-it-py            2.2.0                    pypi_0    pypi
matplotlib-inline         0.1.6                    pypi_0    pypi
mdit-py-plugins           0.3.5                    pypi_0    pypi
mdurl                     0.1.2                    pypi_0    pypi
nbformat                  5.8.0                    pypi_0    pypi
ncurses                   6.4                  h6a678d5_0  
nest-asyncio              1.5.6                    pypi_0    pypi
networkx                  2.6.3                    pypi_0    pypi
numpy                     1.21.6                   pypi_0    pypi
omnigibson                0.1.0                    pypi_0    pypi
opencv-python             4.7.0.72                 pypi_0    pypi
openssl                   1.1.1t               h7f8727e_0  
packaging                 23.1                     pypi_0    pypi
parso                     0.8.3                    pypi_0    pypi
pexpect                   4.8.0                    pypi_0    pypi
pickleshare               0.7.5                    pypi_0    pypi
pillow                    9.5.0                    pypi_0    pypi
[... 96 lines cut ...]
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  CMakeModules/FindPython.cmake:212 (find_package)
  CMakeLists.txt:153 (find_boost_numpy)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  CMakeModules/FindPython.cmake:189 (find_package)
  CMakeModules/PythonBindingsUtils.cmake:5 (find_boost_python)
  py-bindings/CMakeLists.txt:1 (include)

-- Found castxml: /scr/garlanka/miniconda3/envs/omnigibson/bin/castxml  
CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  tests/CMakeLists.txt:1 (find_package)
```

sujaygarlanka, 2023-06-14:
Actually warnings are not issues. Ends up producing the correct makefile and eventually the bindings.

closed 2023-06-14 by sujaygarlanka