ompl/ompl #1065: Bug: Ubuntu 23.04 failure to build OMPL - Could NOT find Boost (missing: filesystem system)

Ryanf55, 2023-06-23:
OS: Ubuntu 23.04
Boost: `Version: 1.74.0.3ubuntu7` installed through `apt` as `libboost-dev`
g++ version: `Ubuntu 12.2.0-17ubuntu1`
CMake version: `3.27.0-rc3`

```
--- stderr: ompl
CMake Error at /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: filesystem system) (found suitable version
  "1.74.0", minimum required is "1.58")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.27/Modules/FindBoost.cmake:2378 (find_package_handle_standard_args)
  CMakeLists.txt:52 (find_package)
```

Reference: https://stackoverflow.com/questions/71610458/could-not-find-boost-missing-system-found-suitable-versihow-on-1-78-0-min

zkingston, 2023-06-23:
I believe you need to install `libboost-all-dev`. `libboost-dev` does not pull in all components of boost.

Ryanf55, 2023-06-23:
Yea, perhaps they changed what's installed by default... Here's what's on my system with `libboost-dev`. 

```
ryan@ryan-B650-970:~/Development/nav2_ws/src/ompl/ompl$ find /usr  -name "*.cmake" | grep -v 3.27 | grep -v 3.25 | grep boost
/usr/lib/x86_64-linux-gnu/cmake/boost_wserialization-1.74.0/boost_wserialization-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_wserialization-1.74.0/boost_wserialization-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_wserialization-1.74.0/libboost_wserialization-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_wserialization-1.74.0/libboost_wserialization-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_regex-1.74.0/boost_regex-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_regex-1.74.0/libboost_regex-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_regex-1.74.0/boost_regex-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_regex-1.74.0/libboost_regex-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_serialization-1.74.0/boost_serialization-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_serialization-1.74.0/libboost_serialization-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_serialization-1.74.0/boost_serialization-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_serialization-1.74.0/libboost_serialization-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_program_options-1.74.0/libboost_program_options-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_program_options-1.74.0/libboost_program_options-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_program_options-1.74.0/boost_program_options-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_program_options-1.74.0/boost_program_options-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_date_time-1.74.0/libboost_date_time-variant-static.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_date_time-1.74.0/boost_date_time-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_date_time-1.74.0/libboost_date_time-variant-shared.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_date_time-1.74.0/boost_date_time-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_headers-1.74.0/boost_headers-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_headers-1.74.0/boost_headers-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_python-1.74.0/boost_python-config-version.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_python-1.74.0/boost_python-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_python-1.74.0/libboost_python-variant-static-py3.11.cmake
/usr/lib/x86_64-linux-gnu/cmake/boost_python-1.74.0/libboost_python-variant-shared-py3.11.cmake
```

After installing `libboost-filesystem-dev`, it fixed that issue.

I can push a quick PR to improve the finding of boost. Since Ubuntu 18.04 and other's are EOL, it should be safe to bump some behavior up. 

Sure, it might be ok for the general README to recommend users install `libboost-all-dev`, however I try to keep my build dependencies to a minimum.

closed 2023-06-24 by zkingston