ompl/ompl #1068: Advice on optimizing Path Planning using ompl app

IEI245MU5, 2023-06-26:
Hello,

I am experimenting with Path Planning using ompl.app. However I am unable to get a smooth solution path unlike using the ompl.app GUI, using the same problem setup and planner (RRTConnect).
The states from the output solution path of my function are very few and seemingly jump randomly, ignoring the Path Length Optimization Objective.
Here is the function I am using, it is mainly from the 3D Rigid Body Planning Demo:

```
bool plan(std::string robot_cad_path, 
                std::string env_cad_path,
                std::vector<double> start, 
                int bounds, 
                std::string solution_directory){
    // plan in SE3
    app::SE3RigidBodyPlanning setup;

    // load the robot and the environment
    setup.setRobotMesh(robot_cad_path);
    setup.setEnvironmentMesh(env_cad_path);
    
    // increase bounds to be larger than set environment
    setup.setBoundsAddition(bounds);
 
    // define start state
    base::ScopedState<base::SE3StateSpace> start(setup.getSpaceInformation());
    start->setX(start[0]);
    start->setY(start[1]);
    start->setZ(start[2]);
    start->rotation().setIdentity();
 
    // define goal state
    base::ScopedState<base::SE3StateSpace> goal(start);
    goal->setX(bounds);
    goal->setY(bounds);
    goal->setZ(bounds);
    goal->rotation().setIdentity();
 
    // set the start & goal states
    setup.setStartAndGoalStates(start, goal);
 
    // setting collision checking resolution
    setup.getSpaceInformation()->setStateValidityCheckingResolution(0.005);

    // select Planner and Optimization Objective
    setup.setOptimizationObjective(std::make_shared<ompl::base::PathLengthOptimizationObjective>(setup.getSpaceInformation()));
    setup.setPlanner(std::make_shared<geometric::RRTConnect>(setup.getSpaceInformation()));
 
    // try to solve the problem
    if (setup.solve(300))
    {

        // save the solution path to a text file
        std::ofstream pathFile = std::ofstream(solution_directory + "/path.txt");
        setup.getSolutionPath().printAsMatrix(pathFile);
        pathFile.close();
        setup.getSolutionPath().printAsMatrix(std::cout);

        return true;
    }

    return false;
}
```
Here are the solution path of my function and the one from the GUI in comparison:
[path.txt](https://github.com/ompl/ompl/files/11870558/path.txt)
[path_GUI.txt](https://github.com/ompl/ompl/files/11870560/path_GUI.txt)

I would be very grateful for advice on how to optimize my function to produce a better solution path.

zkingston, 2023-06-26:
The reason why the GUI is producing better paths is that it is simplifying the resulting path. You can simplify your path by inserting:
```cpp
setup.simplifySolution();
```
before you call `getSolutionPath()`.

IEI245MU5, 2023-06-28:
When using the Path Simplifier, I often receive a warning:
`Warning: Solution path may slightly touch on an invalid region of the state space`

Assuming the movement between states to be linear, the presented solution moves the robot through solid objects.
Do I have to  `setStateValidityCheckingResolution()` lower?