ompl/ompl #1069: FMT* missing optimization termination condition causing it to stop early

scchow, 2023-07-06:
I was testing some optimal planners and noticed FMT* seems to always terminate early whereas RRT* and AIT* takes most of the planning time to improve their solutions. Looking at the termination conditions for each of the planners, I noticed that  [AIT*](https://github.com/ompl/ompl/blob/2ecd327ca66890e84dd4e3649503e0eb765a13c8/src/ompl/geometric/planners/informedtrees/src/AITstar.cpp#L243-L246) and [RRT*](https://github.com/ompl/ompl/blob/2ecd327ca66890e84dd4e3649503e0eb765a13c8/src/ompl/geometric/planners/rrt/src/RRTstar.cpp#L539-L542) seem to check whether `objective_->isSatisfied(cost)` before terminating, whereas [FMT* ](https://github.com/ompl/ompl/blob/2ecd327ca66890e84dd4e3649503e0eb765a13c8/src/ompl/geometric/planners/fmt/src/FMT.cpp#L348-L349) seems to break as soon as a goal is found.

Given that FMT* is also an asymptotically optimal planner, should FMT* also check whether the objective is satisfied before terminating?

ChamzasKonstantinos, 2023-07-07:
Hi @scchow  FMT* is asymptotically optimal but for a **predifined fixed** set of samples. In contrast to other planners it does not add more samples given more time. The optimality lies in the fact, that for this fixed number of samples, it will find the  optimal path if given enough time. The asymptotic part is achieved by increasing the number of samples e.g. by using the [setNumSamples()](https://ompl.kavrakilab.org/classompl_1_1geometric_1_1FMT.html#afb43613a5f83ade6f8e90a0fa2afa544) However, you would have to call the planner again after you increase the number of samples and planning would start from scratch.

scchow, 2023-07-07:
Hi @ChamzasKonstantinos 
Ah, that makes sense! I had thought that FMT* would incrementally increase the number of samples as planning time permits while reusing the existing information in the tree, but now I realize I was probably thinking of a variation of FMT* (possibly [IAFMT*](https://ieeexplore.ieee.org/document/9091952)). Thanks for the response!

closed 2023-07-07 by scchow