ompl/ompl #1078: Make update_bindings error cannot find bindings/base

lydonchandra, 2023-07-19:
Checking out `omplapp` `main` branch recursively today,
python `make update_bindings` failed:
[image]

It builds OK When main is checkout just after #980 is committed. 
So somewhere along the commits, somethings broke again ?

Ubuntu 20.04

1850298154, 2023-07-22:
I meet a problem in Ubuntu22 . I try to use python to use ompl library (use .so files in ompl). 
```shell
(base) z@z:/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB$ ipython3 test.py 
['/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB', '/home/z/anaconda3/bin', '/home/z/anaconda3/lib/python39.zip', '/home/z/anaconda3/lib/python3.9', '/home/z/anaconda3/lib/python3.9/lib-dynload', '', '/home/z/anaconda3/lib/python3.9/site-packages', '/home/z/anaconda3/lib/python3.9/site-packages/locket-0.2.1-py3.9.egg', '/home/z/anaconda3/lib/python3.9/site-packages/IPython/extensions', '/home/z/.ipython', '/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/../opengjk', '/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB']
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/test.py in <module>
      4 import shutil
      5 import numpy as np
----> 6 from uav import *
      7 from others import *
      8 from run import *

/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/uav.py in <module>
      3 from scipy import linalg as lg
      4 from geometry import *
----> 5 from path_planning import *
      6 import copy
      7 import time

/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/path_planning.py in <module>
      2 import numpy as np
      3 try:
----> 4     from ompl import util as ou
      5     from ompl import base as ob
      6     from ompl import geometric as og

/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/ompl/util/__init__.py in <module>
      1 from os.path import abspath, dirname
      2 from ompl import dll_loader
----> 3 dll_loader('ompl', dirname(abspath(__file__)))
      4 from ompl.util._util import *
      5 import inspect

/mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/ompl/__init__.py in dll_loader(lib, path)
     15     if not isfile(fname):
     16         fname = find_library(lib)
---> 17     ctypes.CDLL(fname, ctypes.RTLD_GLOBAL)
     18 
     19 class PlanningAlgorithms(object):

~/anaconda3/lib/python3.9/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    380 
    381         if handle is None:
--> 382             self._handle = _dlopen(self._name, mode)
    383         else:
    384             self._handle = handle

OSError: /home/z/anaconda3/lib/python3.9/site-packages/matplotlib/../../../libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/ompl/util/libompl.so)
```

I used `dll` and `ll` to check what is wrong.

```shell
(base) z@z:~$ ldd /usr/local/lib/libompl.so
	linux-vdso.so.1 (0x00007fff821d1000)
	libode.so.8 => /lib/x86_64-linux-gnu/libode.so.8 (0x00007f584cc9e000)
	libboost_serialization.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0 (0x00007f584cc5c000)
	libboost_filesystem.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0 (0x00007f584cc3c000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f584c000000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f584c319000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f584c2f9000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f584bc00000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f584cdd5000)
	libccd.so.2 => /lib/x86_64-linux-gnu/libccd.so.2 (0x00007f584cc2e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f584cc29000)
(base) z@z:~$ ll  /lib/x86_64-linux-gnu/libstdc++.so.6 
lrwxrwxrwx 1 root root 19  7月 18 20:31 /lib/x86_64-linux-gnu/libstdc++.so.6 -> libstdc++.so.6.0.30
(base) z@z:~$ find / -name "libstdc++.so.6.0.30" 2>/dev/null
/snap/gnome-42-2204/120/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30
/snap/core22/817/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30
/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30
(base) z@z:~$ find / -name "libstdc++.so.6.0.29" 2>/dev/null
(base) z@z:~$ 

```

I have `libstdc++.so.6.0.30` , but i don't have `libstdc++.so.6.0.29`.
How can i do to solve it ? 

Thank you for you advice !

1850298154, 2023-07-22:
I sovled above problem. But I meet a new problem.
How can I solve it？ thank you！

ubuntu22   anaconda 2023-07
```shell

File /mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/ompl/util/__init__.py:4
      2 from ompl import dll_loader
      3 dll_loader('ompl', dirname(abspath(__file__)))
----> 4 from ompl.util._util import *
      5 import inspect
      7 def OMPL_DEBUG(text):

SystemError: initialization of _util raised unreported exception

```