ompl/ompl #1079: SystemError: initialization of _util raised unreported exception

1850298154, 2023-07-23:
[image]

File /mnt/MPC-MOTION-PLANING/github-share/MPC-MP-zyt-note/IMPC-zOB/ompl/util/__init__.py:4
      2 from ompl import dll_loader
      3 dll_loader('ompl', dirname(abspath(__file__)))
----> 4 from ompl.util._util import *
      5 import inspect
      7 def OMPL_DEBUG(text):

SystemError: initialization of _util raised unreported exception

How to solve it ? 
Thank you!

zkingston, 2023-07-31:
Can you post the full output of `make update_bindings`?

lostDeers, 2024-01-17:
I met the same problem. Here is the full output of execute `make update_bindings` in `ompl-1.6.0/build/Release`.
```
[  0%] Preparing C++ header file for Python binding generation for module control
[  0%] Built target control.h
[  0%] Preparing C++ header file for Python binding generation for module base
[  0%] Built target base.h
[100%] Preparing C++ header file for Python binding generation for module util
[100%] Built target util.h
[100%] Creating C++ code for Python module util (see pyplusplus_util.log)
/usr/local/lib/python3.10/dist-packages/pyplusplus/binary_parsers/parsers.py:7: DeprecationWarning: invalid escape sequence 'm'
  """
std::vector<double>::reference std::vector<double>::at(std::size_t
__n) [member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<double>::reference std::vector<double>::front() [member
function] > warning W1008: The function returns non-const reference to
"Python immutable" > type. The value cannot be modified from Python.
std::vector<double>::reference std::vector<double>::back() [member
function] > warning W1008: The function returns non-const reference to
"Python immutable" > type. The value cannot be modified from Python.
std::vector<double>::pointer
std::vector<double>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const double *, std::vector<double>>
__first, __gnu_cxx::__normal_iterator<const double *,
std::vector<double>> __last) [member function] > compilation error
W1050: The function returns "std::vector<double>::pointer" > type. You
have to specify a call policies.Be sure to take a look on `Py++` >
defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer
__first, std::vector<double>::pointer __last,
std::vector<double>::pointer __result,
std::vector<double>::_Tp_alloc_type & __alloc, std::true_type arg4)
[member function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer arg0,
std::vector<double>::pointer arg1, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & arg3, std::false_type
arg4) [member function] > compilation error W1050: The function
returns "std::vector<double>::pointer" > type. You have to specify a
call policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_relocate(std::vector<double>::pointer __first,
std::vector<double>::pointer __last, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & __alloc) [member
function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies.Be sure to take a look on `Py++` > defined call policies
std::vector<unsigned int>::reference std::vector<unsigned
int>::at(std::size_t __n) [member function] > warning W1008: The
function returns non-const reference to "Python immutable" > type. The
value cannot be modified from Python.
std::vector<unsigned int>::reference std::vector<unsigned
int>::front() [member function] > warning W1008: The function returns
non-const reference to "Python immutable" > type. The value cannot be
modified from Python.
std::vector<unsigned int>::reference std::vector<unsigned int>::back()
[member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<unsigned int>::pointer std::vector<unsigned
int>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const unsigned int *,
std::vector<unsigned int>> __first, __gnu_cxx::__normal_iterator<const
unsigned int *, std::vector<unsigned int>> __last) [member function] >
compilation error W1050: The function returns "std::vector<unsigned >
int>::pointer" type. You have to specify a call policies.Be sure to
[... 2879 lines cut ...]
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/lostdeers/robo/RoboGen/ompl-1.6.0/build/Release
[100%] Built target update_control_bindings
[100%] Built target update_bindings
```

abaicoming, 2025-03-06:
I encountered the same issue. May I ask if you have resolved it?

lostDeers, 2025-03-12:
I used prebuild binary version from official and solve the problem.

abaicoming, 2025-03-23:
Thank you for the reminder. However, I found that this issue seems to be caused by a conflict between "OMPL" and "Anaconda." The problem was resolved after I tried launching without Anaconda:
`conda deactivate`

lostDeers, 2025-03-24:
In some cases it will happen without Anaconda, but that's rarely.