ompl/ompl #1083: No file generated after running 'make update_bindings'

WangCH69, 2023-07-28:
**1、Following are my system setups:**
OS: Ubuntu 18.04 bionic
Python version: 3.8
Conda Environment: YES

**2、Issue description:**
I first used 'cmake -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" ../..' to build the CMakelist file. Then I used 'make update_bindings' command to build python bindings.
However, a weird thing happened. In the ubuntu terminal I only received a feedback 'Built target update_bindings' and there is no corresponding 'XXX.so' file such as '_util.so' in the directory 'py-bindings/ompl'. So I guess there must be something wrong with the binding generation, but I couldn't locate the exact issue.

**3、The output after running 'Cmake ../.. '**
(LWang_2) dell@dell-Precision-5820-Tower-X-Series:~/WCH-Laowang/IMPC/IMPC-OB/ompl/build/Release$ cmake -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" ../..
-- PYTHON PATH : /usr/local/anaconda3/envs/LWang_2/bin/python3
-- Build type: Release
-- Checking for module 'libspot'
-- No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
-- The following features have been enabled:

OMPL_VERSIONED_INSTALL, Whether to install header files in
/include/ompl-X.Y/ompl, where X and Y are the major and minor
version numbers
OMPL_BUILD_TESTS, Whether to build the unit tests.
OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.
-- The following OPTIONAL packages have been found:

Threads, https://en.wikipedia.org/wiki/POSIX_Threads
Pthreads is sometimes needed, depending on OS / compiler.
pypy, https://pypy.org/
Used to speed up the generation of python bindings.
flann (required version >= 1.8.3), https://github.com/mariusmuja/flann
If detetected, FLANN can be used for nearest neighbor queries by OMPL.
Doxygen, http://doxygen.org/
Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org/).
Python, https://www.python.org/
Used for python bindings.
castxml, https://github.com/CastXML/CastXML
Used to generate Python bindings.
PkgConfig, https://www.freedesktop.org/wiki/Software/pkg-config/
Used to find (compilation flags for) dependencies.
yaml-cpp, https://github.com/jbeder/yaml-cpp
Used for parsing input data for the PlanarManipulator demo.
-- The following REQUIRED packages have been found:

Eigen3, http://eigen.tuxfamily.org/
A linear algebra library used throughout OMPL.
Boost (required version >= 1.58), https://www.boost.org/
Used throughout OMPL for data serialization, graphs, etc.
_-- The following features have been disabled:

OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests._
-- The following OPTIONAL packages have not been found:

Triangle, http://www.cs.cmu.edu/~quake/triangle.html
Used to create triangular decompositions of polygonal 2D environments.
spot, http://spot.lrde.epita.fr/
Used for constructing finite automata from LTL formulae.
-- The following Python modules were found:

numpy pyplusplus pygccxml
-- Configuring done (0.6s)
-- Generating done (0.1s)
-- Build files have been written to: /home/dell/WCH-Laowang/IMPC/IMPC-OB/ompl/build/Release

I found out that the feature 'OMPL_BUILD_PYBINDINGS' had been disabled, which never happened in all other possible issues I've read in the issue section. I guess this must be the reason, but I just don't know how it ends up like this. Could this be related to my python package setup?

Been struggling with this issue for some days. Any help would be appreciated! Thanks!

WangCH69, 2023-07-28:
Also here's the CMakeCache.txt:
# This is the CMakeCache file.
# For build in directory: /home/dell/WCH-Laowang/IMPC/IMPC-OB/ompl/build/Release
# It was generated by CMake: /usr/local/anaconda3/envs/LWang_2/lib/python3.8/site-packages/cmake/data/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//No help, variable specified on the command line.
BOOST_ROOT:PATH=/usr/local/anaconda3/envs/LWang_2

//The directory containing a CMake configuration file for Boost.
Boost_DIR:PATH=Boost_DIR-NOTFOUND

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY38_LIBRARY_DEBUG-NOTFOUND

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY38_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy3.so

//Boost numpy3 library (release)
Boost_NUMPY3_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy3.so

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py38 library (debug)
Boost_PYTHON-PY38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY38_LIBRARY_DEBUG-NOTFOUND

//Boost python-py38 library (release)
Boost_PYTHON-PY38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY38_LIBRARY_RELEASE-NOTFOUND

//Boost python38 library (debug)
Boost_PYTHON38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON38_LIBRARY_DEBUG-NOTFOUND

//Boost python38 library (release)
Boost_PYTHON38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON38_LIBRARY_RELEASE-NOTFOUND

//Boost python3 library (debug)
Boost_PYTHON3_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python3.so
[... 737 lines cut ...]
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//linker supports push/pop state
_CMAKE_LINKER_PUSHPOP_STATE_SUPPORTED:INTERNAL=TRUE
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local/anaconda3/envs/LWang_2
__pkg_config_arguments_FLANN:INTERNAL=flann
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann
pkgcfg_lib_FLANN_flann-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann_cpp
pkgcfg_lib_FLANN_flann_cpp-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=

zkingston, 2024-04-23:
Closing due to age.

closed 2024-04-23 by zkingston