ompl/ompl #1090: AttributeError: module 'ompl.geometric' has no attribute 'xxx'

im-renpei, 2023-08-17:
Hi developers,

Thanks for your contributions to this library, it really is a powerful tool in robotics. @zkingston @mamoll 

I am trying to **RRTstar-Connect** and **Informed-RRTstar-Connect**, which have been implemented by @rezamasha in https://github.com/ompl/ompl/pull/743, so I simply copied the changes involved into the OMPL 1.6.0 code, and used the script _install-ompl-ubuntu-projectx.sh_ for installation; the installation log is recorded in _ompl_installation_log.txt_.

The entire installation process does not seem to have any problems. After the installation is complete, I can also successfully execute _demos/RigidBodyPlanning.py_, but when I replace **RRTConnect** (https://github.com/ompl/ompl/blob/main/demos/RigidBodyPlanning.py#L119) with **InformedRRTstarConnect** or **RRTstarConnect**, I get the following error:

```bash
Traceback (most recent call last):
  File "demos/RigidBodyPlanning.py", line 143, in <module>
    planTheHardWay()
  File "demos/RigidBodyPlanning.py", line 119, in planTheHardWay
    planner = og.InformedRRTstarConnect(si)
AttributeError: module 'ompl.geometric' has no attribute 'InformedRRTstarConnect'
```
```bash
Traceback (most recent call last):
  File "demos/RigidBodyPlanning.py", line 143, in <module>
    planTheHardWay()
  File "demos/RigidBodyPlanning.py", line 119, in planTheHardWay
    planner = og.RRTstarConnect(si)
AttributeError: module 'ompl.geometric' has no attribute 'RRTstarConnect'
```

Am I doing something wrong or missing something?

Thank you in advance.

[install-ompl-ubuntu-projectx.sh.txt](https://github.com/ompl/ompl/files/12367375/install-ompl-ubuntu-projectx.sh.txt)
[ompl_installation_log.txt](https://github.com/ompl/ompl/files/12367383/ompl_installation_log.txt)

im-renpei, 2023-08-18:
Hi @Jaeyoung-Lim , do you have any suggestions for this problem? I see that you have been able to successfully use RRTstarConnect planner.

zkingston, 2023-11-28:
You need to add the relevant headers to the `headers_geometric.txt` file under the `py-bindings`, and then regenerate the python bindings.