ompl/ompl #1096: Always planning up to solving time

mojervico, 2023-09-01:
Hi,

I'm using the simple planners such as RRT/RRTConnect/PRM etc, an i'm calling the function solve from the simpleSetupPtr like thata ss->solve(solving_time).

My probem/doubts is that no matter the value of solving_time I choose, the planner always reach its maximum solving_time. I mean if I choose a solvin time of 1 second, the planner is able to solver the problem in 1 second. However, if for the same problem (same states and seed) I choose I solving time of 10 seconds, it takes up to 10 seconds to solve a problem that can be solve in 1. I would like the planner to return as soon as it finds the solution. Am I missing anything?

Best Regards

IEI245MU5, 2023-09-10:
Hello,

By default the planner always terminates after finding the first solution.

Although when the planner reaches its maximum solving time, it sometimes approximates the solution path and solve() returns true even when these approximated solution might not actually be correct. If you want to have an exact solution you most likely have to increase the solving time and use

`if (ss.solve(solving_time)==ompl::base::PlannerStatus::EXACT_SOLUTION)`

Best Regards
Kaiwen

zkingston, 2023-10-09:
The above comment is correct. You should check if you are getting exact solutions or approximate solutions - the behavior described (if you are not using an AO planner) sounds like the planner is failing to find a solution and is returning an approximate solution. If you are using an AO planner (one with a * after the name, e.g., PRM*), these planners will use the entire allocated solution time.

closed 2023-10-09 by zkingston