ompl/ompl #1098: boost_python-py311 not found on macOS

fferri, 2023-09-13:
Python 3.11.5 is installed, as well as boost-python3 1.82.0:

```
❯ brew ls | grep python
boost-python3
python@3.11

❯ ls $HOMEBREW_CELLAR/*python*
/opt/homebrew/Cellar/boost-python3:
1.82.0

/opt/homebrew/Cellar/python@3.11:
3.11.5
```

but when running CMake, boost_python-py311==1.82.0 is reported as not found:

```
 -- The following OPTIONAL packages have not been found:

  * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
    Used to create triangular decompositions of polygonal 2D environments.
  * spot, <http://spot.lrde.epita.fr>
    Used for constructing finite automata from LTL formulae.
  * boost_python-py311 (required version == 1.82.0)
  * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
    Used for parsing input data for the PlanarManipulator demo.

 The following Python modules were found:

  * numpy pyplusplus pygccxml

 Configuring done (2.4s)
```

zkingston, 2023-09-22:
Please see #1100 and let me know if this addresses compilation issues.

fferri, 2023-09-22:
Nope.
I tried apple-m1-python-bindings branch.
Some errors have been fixed (the one about `boost::type`).
Some others are still there (ValueError: too many values to unpack (expected 4)).
Ultimately, only _base.so and _util.so are built.

zkingston, 2023-09-22:
Ah, I know the error about value unpacking - will be pushing an update to pyplusplus soon.

zkingston, 2023-09-22:
See https://github.com/ompl/pyplusplus/pull/1

fferri, 2023-09-22:
I installed pyplusplus git branch mac-update, but something's not right:

```
❯ make clean_bindings
Built target clean_bindings

❯ make update_bindings
[  0%] Preparing C++ header file for Python binding generation for module control
[  0%] Preparing C++ header file for Python binding generation for module base
[  0%] Preparing C++ header file for Python binding generation for module geometric
[  0%] Preparing C++ header file for Python binding generation for module tools
[ 50%] Preparing C++ header file for Python binding generation for module util
[ 50%] Built target tools.h
[ 50%] Built target util.h
[ 50%] Built target control.h
[ 50%] Built target geometric.h
[ 50%] Built target base.h
[ 50%] Creating C++ code for Python module util (see pyplusplus_util.log)
Traceback (most recent call last):
  File "/Users/me/Dev/ompl/py-bindings/generate_bindings.py", line 43, in <module>
    from pyplusplus.module_builder import call_policies
ImportError: cannot import name 'call_policies' from 'pyplusplus.module_builder' (unknown location)
-- Build type: Release
warning: target 'Boost::system' is not absolute...
warning: target 'Boost::system' does not exist...
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
-- Code for module util not found; type "make update_bindings"
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- Checking for module 'yaml-cpp'
--   No package 'yaml-cpp' found
-- Could NOT find yaml-cpp (missing: YAMLCPP_LIBRARIES YAMLCPP_INCLUDE_DIRS)
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).
 * boost_numpy (required version == 1.82.0)
 * Python, <https://www.python.org>
   Used for python bindings.
 * boost_python (required version == 1.82.0)
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * boost_unit_test_framework (required version == 1.82.0)
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.

-- The following REQUIRED packages have been found:

 * boost_serialization (required version == 1.82.0)
 * boost_atomic (required version == 1.82.0)
[... 336 lines cut ...]
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done (0.5s)
-- Generating done (0.2s)
-- Build files have been written to: /Users/me/Dev/ompl/build
[100%] Built target update_tools_bindings
[100%] Built target update_bindings

❯ make py_ompl
Built target py_ompl
```

note the error:

```
Traceback (most recent call last):
  File "/Users/me/Dev/ompl/py-bindings/generate_bindings.py", line 43, in <module>
    from pyplusplus.module_builder import call_policies
ImportError: cannot import name 'call_policies' from 'pyplusplus.module_builder' (unknown location)
```

zkingston, 2023-09-22:
Just a sanity check, can you delete and re-initialize the build directory after reinstalling `pyplusplus`?

fferri, 2023-09-22:
Yes, that worked. Good catch!

```
❯ ls lib
_base.so            _geometric.so       _util.so            libompl.17.dylib    py_std_function.so
_control.so         _tools.so           libompl.1.6.0.dylib libompl.dylib
```

```
❯ PYTHONPATH=$PWD/lib:$PWD/../py-bindings bpython
bpython version 0.24 on top of Python 3.11.5 /opt/homebrew/opt/python@3.11/bin/python3.11
>>> from ompl import geometric as og
>>> from ompl import base as ob
>>>
```

closed 2023-10-09 by zkingston

ccm010101, 2024-11-13:
> Yes, that worked. Good catch!
>
> ```
> ❯ ls lib
> _base.so            _geometric.so       _util.so            libompl.17.dylib    py_std_function.so
> _control.so         _tools.so [...]

@fferri 
Can you share how you re-installed pyplusplus?

I use mac M1, and have

```
(ompl) Doonams-MacBook-Pro:ompl chanmichoi$ brew ls | grep python
boost-python3
python@3.12
python@3.13

(ompl) Doonams-MacBook-Pro:ompl chanmichoi$ ls $HOMEBREW_CELLAR/*python*
/opt/homebrew/Cellar/boost-python3:
1.86.0

/opt/homebrew/Cellar/python@3.12:
3.12.7_1

/opt/homebrew/Cellar/python@3.13:
3.13.0_1
```

I encounter the same error message like yours
`ImportError: cannot import name 'call_policies' from 'pyplusplus.module_builder' (unknown location)`

For me, I did
```
git clone https://github.com/ompl/pyplusplus.git
cd pyplusplus
python setup.py install
```