ompl/ompl #1099: The py-bindings were not successfully generated

zhe-beginner, 2023-09-19:
Dear program developers and colleagues, I encountered some issues while installing omplapp. I would greatly appreciate it if I could get your help.

My system is ubuntu 20.04, which has been installed several times through source code but failed. I was worried about system environmental pollution, so I reinstalled the ubuntu 20.04 system and decided to install it through the install-ompl-ubuntu.sh script. After reinstalling the system, I followed the following steps to install it.

> 1   Download the OMPL installation script(https://ompl.kavrakilab.org/install-ompl-ubuntu.sh)
> 2   chmod u+x install-ompl-ubuntu.sh
> 3   ./install-ompl-ubuntu.sh –app

Everything seems to be going smoothly

Unfortunately, the py-bindings file was not successfully generated. The relevant logs under the command of compile-1.6.0-Source/build/Release are as follows:

1. The content in pyplusplus_app.log is：

Traceback (most recent call last):
  File "/home/l/omplapp/omplapp-1.6.0-Source/py-bindings/generate_bindings.py", line 16, in <module>
    from pygccxml.declarations.runtime_errors import declaration_not_found_t
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/__init__.py", line 45, in <module>
    import importlib_metadata as metadata
  File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/__init__.py", line 6, in <module>
    import zipp
  File "/usr/local/lib/python3.8/dist-packages/zipp/__init__.py", line 9, in <module>
    from .py310compat import text_encoding
  File "/usr/local/lib/python3.8/dist-packages/zipp/py310compat.py", line 5
    def _text_encoding(encoding, stacklevel=2, /):  # pragma: no cover
                                               ^
SyntaxError: invalid syntax (expected ')')

2. The contents in pyplusplus_base.log  pyplusplus_control.log  pyplusplus_geometric.log  pyplusplus_util.log are the same,all of which are:

Traceback (most recent call last):
  File "/home/l/omplapp/omplapp-1.6.0-Source/ompl/py-bindings/generate_bindings.py", line 41, in <module>
    from pygccxml import declarations
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/__init__.py", line 45, in <module>
    import importlib_metadata as metadata
  File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/__init__.py", line 6, in <module>
    import zipp
  File "/usr/local/lib/python3.8/dist-packages/zipp/__init__.py", line 9, in <module>
    from .py310compat import text_encoding
  File "/usr/local/lib/python3.8/dist-packages/zipp/py310compat.py", line 5
    def _text_encoding(encoding, stacklevel=2, /):  # pragma: no cover
                                               ^
SyntaxError: invalid syntax (expected ')')

I don't know where the problem occurred during the installation process, and I look forward to receiving your help.

Tass0sm, 2023-09-19:
Also getting this error

zhe-beginner, 2023-09-20:
The latest progress has found that it may be due to the high version of the **zipp** Python package .When I run 

> ./install-ompl-ubuntu.sh –app

The system defaults to installing zipp 3.17.0. The relevant statement shows[https://github.com/jaraco/zipp/issues/100]
Zip=3.17.0 and zip=3.16.0 are not supported Python>=3.8,

So I uninstalled zip 3.17.0 using 

> sudo - H pip3 uninstall zipp

 and installed zip 3.15.0 using 

> sudo - H pip3 install zip==3.15.0

The above issue was resolved, but new issues arose.
When I run 

> make -j 1 update_bindings

 In the **omplapp/omplap-1.6.0-Source/build/Release** directory, the relevant logs under the command of **omplapp/omplap-1.6.0-Source/build/Release** are as follows:

    The contents in pyplusplus_base.log pyplusplus_control.log pyplusplus_geometric.log pyplusplus_util.log are the same,all of which are:

> Traceback (most recent call last):
>   File "/home/l/omplapp/omplapp-1.6.0-Source/ompl/py-bindings/generate_bindings.py", line 41, in <module>
>     from pygccxml import declarations
>   File "/usr/local/lib/python3.8/dist-packages/pygccxml/__init__.py", line 45, in <module>
>     import importlib_metadata as metadata
>   File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/__init__.py", line 18, in <module>
>     from . import _adapters, _meta, _py39compat
>   File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/_meta.py", line 1, in <module>
>     from typing import Protocol
> ImportError: cannot import name 'Protocol'

Relevant information display, **Protocol** can be used in Python3.8 through  **from typing import Protocol** .
It is really strange.
I don't know where the problem occurred during the installation process, and I look forward to receiving your help.

Tass0sm, 2023-09-21:
I found using `make -n update_bindings` that it was running pypy3 to generate the bindings. pypy3 didn't seem to work on my system so I reran cmake as recommended to disable using pypy3.

cd build/Release
cmake -DCMAKE_DISABLE_FIND_PACKAGE_PYPI=ON ../..

Then I removed python 2 from my system and also changed the shebang in generate_bindings.py to be `#!/usr/bin/env python3` instead of `#!/usr/bin/env python`

I'll post another comment if I find why pypy3 doesn't work

zhe-beginner, 2023-09-22:
Thank you for your suggestion.
In _./install-ompl-ubuntu.sh_ file. I change 

> cmake ../.. -DPYTHON_EXEC=/usr/bin/python${PYTHONV}

to

> cmake -DCMAKE_DISABLE_FIND_PACKAGE_pypy=ON ../.. -DPYTHON_EXEC=/usr/bin/python${PYTHONV}

and did not remove python 2，
Then run ./install-ompl-ubuntu.sh –app
Some other problems occurred, but have been resolved by [https://github.com/ompl/ompl/commit/120aa00e3031d778c6a7e051b2b84633923e701d](url).

Although your question of "why pypy3 doesn't work" is still not solved, the gui can at least be used.

Your discovery helped me a lot, thank you again,
I will post my latest findings on the issue here

zkingston, 2023-09-22:
If you are compiling directly from the installation script, can you please use the latest commit on this git repository instead? There are some documented issues with Python binding generation using the install script. Let me know if errors persist.

Tass0sm, 2023-09-23:
I encountered this particular issue on both 1.6 and the latest commit I believe. After resolving this error, I did have to use the latest commit due to the issue fixed by #980. While compiling the latest commit, I also encountered the following error:

```
CMake Error in CMakeLists.txt:
        Imported target "Boost::unit_test_framework" includes non-existent path
      
          "/include"
      
        in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
      
        * The path was deleted, renamed, or moved to another location.
      
        * An install or uninstall procedure did not complete successfully.
      
        * The installation package was faulty and references files it does not
        provide.
  ```
  
I didn't have time to find the root cause, so I just created a symlink from `/include` to `/usr/include` and there were no more issues after that.

sakjdas, 2023-10-18:
> make -n update_bindings

> Thank you for your suggestion. Based on your suggestion, in `install-ompl-ubuntu.sh` file, I changed
> 
> > cmake ../.. -DPYTHON_EXEC=/usr/bin/python${PYTHONV}
> 
> to
> 
> > cmake -DCMAKE_DISABLE_FIND_PACKAGE_pypy=ON ../.. -DPYTHON_EXEC=/usr/bin/python${PYTHONV}
> 
> and did not remove python 2， and also changed the shebang in `generate_bindings.py` to be `/usr/bin/env python3` instead of `/usr/bin/env python`.
> 
> Then run ./install-ompl-ubuntu.sh –app Some other problems occurred, but have been resolved by [Added try/except for string SpecificParam](https://github.com/ompl/ompl/commit/120aa00e3031d778c6a7e051b2b84633923e701d) and [Ubuntu20.04安装OMPL安装指南](https://blog.csdn.net/qq_38606680/article/details/129028660).
> 
> Although your question of "why pypy3 doesn't work" is still not solved, the gui can at least be used.
> 
> Your discovery helped me a lot, thank you again, I will post my latest findings on the issue here.

Hi, can I ask you how do you solve the protocol problem? because I did what you have said here, but it does't work. Have you done any other steps? Thanks in advance!

Tass0sm, 2023-11-20:
> I've installed zipp 3.15.0

What python version are you using?

zkingston, 2023-11-28:
Consolidating with #1102 - please move discussion there.