ompl/ompl #1102: Do `` make -j `nproc` update_bindings `` but get ImportError: cannot import name 'Protocol'**

sakjdas, 2023-10-18:
Hi everyone. I try to install ompl and when I do `` make -j `nproc` update_bindings ``, I got the following error message.

**Traceback (most recent call last):
File "/home/l/omplapp/omplapp-1.6.0-Source/ompl/py-bindings/generate_bindings.py", line 41, in
from pygccxml import declarations
File "/usr/local/lib/python3.8/dist-packages/pygccxml/init.py", line 45, in
import importlib_metadata as metadata
File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/init.py", line 18, in
from . import _adapters, _meta, _py39compat
File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/_meta.py", line 1, in
from typing import Protocol
ImportError: cannot import name 'Protocol'**

I try to import typing and it works, It seems that there is no Protocol inside typing. But I googled and it shows that protocol was added into typing in python3.8. So I am very confused and paid a lot time on this problem. Does someone have any idea about how to solve this problem? I appriciated any comment and thanks in advance!

zkingston, 2023-11-09:
Two things:
- It appears that you downloaded the source zip for OMPL.app, could you clone the latest version of OMPL?
- Since you are using Python 3.8, can you use [v2.2.1](https://github.com/CastXML/pygccxml/releases/tag/v2.2.1) of PyGCCXML?

Then clean and re-build

Radiance-nt, 2023-11-09:
Finnaly, the libs (including _util.so, _based.so) have been generated. Although I made it in v1.5.2 version. 

I faced some problem when compiling latest OMPL (v1.6.0), and switched to v1.5.2.
I created a virtual env in build/venv, and a folder `build/local` as the installation directory, `build/Release` as the build directory.
After activating the venv, run `pip install numpy wheel pyplusplus pygccxml==2.2.1` to install the required libs.
Before cmaking, the required pkgs that should be installed with `apt` must be required as well.

After preparation, run cmake with the following commands,
```
CXXFLAGS="-std=c++17  -fsized-deallocation" cmake -DCMAKE_CXX_COMPILER=/usr/bin/g++-9 -DPYTHON_EXEC=`which python` -DOMPL_BUILD_PYBINDINGS=ON -DCMAKE_INSTALL_PREFIX=$(dirname $PWD)/local ../../
```
I think this would help to isolate enviroments and avoid underlying issues. By the way, I tried g++-10 but it did not work for my case.
Another issue is the bindings of OMPL (v1.5.2) have problems compiling with c++17 standard (which is needed for Spot), so I make that target use c++14 standard , 
I add this standard to something like `add_custom_target(update_bindings)` in a cmake file.

After that, I ran `make update_bindings` and `make`, the `*.so` files successfully came out in `build/Release/lib`.

Overall [v2.2.1] of PyGCCXML seems to be the key solving problems along with python3.8. Thank everyone who gave helpful suggestions. Although my installation procedures are so complex, at least the py-bindings can work.

JeelChatrola, 2023-11-10:
Two things:

It appears that you downloaded the source zip for OMPL.app, could you clone the latest version of OMPL?
Since you are using Python 3.8, can you use [v2.2.1](https://github.com/CastXML/pygccxml/releases/tag/v2.2.1) of PyGCCXML?
Then clean and re-build

I tried this but still end up with same problem

`Traceback (most recent call last):
  File "/home/jeelros/Downloads/omplapp-1.6.0-Source/py-bindings/generate_bindings.py", line 16, in <module>
    from pygccxml.declarations.runtime_errors import declaration_not_found_t
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/__init__.py", line 45, in <module>
    import importlib_metadata as metadata
  File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/__init__.py", line 18, in <module>
    from . import _adapters, _meta, _py39compat
  File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/_meta.py", line 1, in <module>
    from typing import Protocol
ImportError: cannot import name 'Protocol'
-- Build type: Release
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimpTargets.cmake:54 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimp-config.cmake:1 (include)
  CMakeLists.txt:139 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- Code for module util not found; type "make update_bindings"
-- Code for module app not found; type "make update_bindings"
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * OpenGL, <http://opengl.org>
   Used to render graphics in ompl_app.
 * pypy, <http://pypy.org>
   Used to speed up the generation of python bindings.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., http://ompl.kavrakilab.org).
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.
 * Python, <http://python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.

-- The following REQUIRED packages have been found:

 * assimp, <http://assimp.org>
   Used in ompl_app for reading meshes representing robots and environments.
 * ccd, <https://github.com/danfis/libccd>
   Collision detection library used by fcl.
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * octomap
 * fcl, <https://github.com/flexible-collision-library/fcl>
   The default collision checking library.
 * Boost, <http://boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Drawstuff, <http://ode.org>
   Part of the ODE source distribution, used in one demo program.
 * PQP, <http://gamma.cs.unc.edu/SSV/>
   Used as an alternative, additional collision checking library (the default is FCL).

-- The following Python modules were found:

 * PyQt5 OpenGL numpy pyplusplus pygccxml flask celery

-- Configuring done
-- Generating done
-- Build files have been written to: /home/jeelros/Downloads/omplapp-1.6.0-Source/build/Release
[100%] Built target update_app_bindings
[100%] Built target update_bindings
`