ompl/ompl #1107: Cannot link to Flann on Ubuntu 22.04

galou, 2023-10-30:
I cannot compile OMPL in Ubuntu 22.04 since 9efd7aad6fb (2023-10-26): `ld.lld: error: unable to find library -lflann::flann`.

To solve the issue, I used `${FLANN_LIBRARIES}` instead of `flann::flann`, i.e. I replaced `"$<$<BOOL:${OMPL_HAVE_FLANN}>:flann::flann>"` with `"$<$<BOOL:${OMPL_HAVE_FLANN}>:${FLANN_LIBRARIES}>"` in `src/ompl/CMakeLists.txt`.

cohnt, 2023-11-09:
I also faced this issue, and can confirm that the fix you've suggested works on my system.

kylc, 2023-11-11:
I am **not** able to reproduce using the minimal Docker setup below. Maybe you have an old FLANN version around being picked up by CMake?

```
docker run --mount type=bind,source="$(pwd)",target=/ompl --rm -it ubuntu:22.04

export DEBIAN_FRONTEND=noninteractive

apt-get update
apt-get -y install git liblz4-dev g++ cmake pkg-config libboost-serialization-dev libboost-filesystem-dev libboost-system-dev libboost-program-options-dev libboost-test-dev libeigen3-dev libode-dev wget libyaml-cpp-dev

git clone https://github.com/flann-lib/flann.git --branch 1.9.2
mkdir -p flann/build && cd flann/build
cmake -DBUILD_EXAMPLES=OFF .. && cmake --build . && cmake --install .

mkdir -p /ompl/build/Release && cd /ompl/build/Release/
cmake ../.. && cmake --build . --parallel
```

cohnt, 2023-11-11:
I have flann 1.6.13 installed from pip.

galou, 2023-11-13:
I have `libflann-dev/jammy,now 1.9.1+dfsg-11 amd64` install with apt.

kylc, 2023-11-13:
I think these older FLANN versions are no longer supported after the recent commits. The README now links to [Flann v1.9.2](https://github.com/flann-lib/flann/tree/1.9.2) and the CMake checks for at least the same.

cohnt, 2023-11-13:
Since FLANN is an optional dependency, shouldn't OMPL be able to build if only an older version is available, by just not including it?

kylc, 2023-11-13:
Yes, I think it should ignore older versions. I tried adding both of `apt install libflann-dev` (for 1.9.1) and `pip install flann==1.6.13` to the Docker script above, and neither have any effect on the OMPL build.

Are you possibly building from an old `build/` directory with a stale CMake cache?

cohnt, 2023-11-13:
Clearing the build cache solved the problem for me. Thanks!

closed 2023-11-27 by zkingston