ompl/ompl #1109: Bloom 1.6.0 binaries for ROS 2 Iron

Yadunund, 2023-11-09:
Hi @mamoll 

Could I trouble you to bloom a `1.6.0` release for ROS 2 Iron similar to [Humble](https://github.com/ros/rosdistro/pull/35871)? Currently the version of `ompl` for Iron is [1.5.2](https://github.com/ros/rosdistro/blob/7bd100ab29bc6e8d5f134bd1941d83381c005876/iron/distribution.yaml#L3550).

I'm noticing some performance differences in my application once I switched to Iron from Humble and wondering if this could be attributed to the older version of the binaries released.

It would be great if you could also bloom a release to Rolling as that would ensure the latest version being available on Jazzy when that i out :)

Yadunund, 2023-12-18:
@mamoll friendly ping.

Yadunund, 2024-01-09:
@zkingston are you able to help with this issue?

zkingston, 2024-01-15:
Yes, I would be willing to help with this issue. I'll ask Mark and get the ball rolling here.

Yadunund, 2024-01-16:
> Yes, I would be willing to help with this issue. I'll ask Mark and get the ball rolling here.

Great! Thanks a lot for your help!

Yadunund, 2024-02-02:
Hi @zkingston, friendly ping to enquire on the status here. When can we expect the newer versions to be bloomed?

Yadunund, 2024-04-22:
@zkingston @mamoll another reminder.

We've already branched into `jazzy` from `rolling` and the version there is also the older `1.5.2`. Kindly bloom `1.6.0` for `rolling`, `iron`, and `jazzy`. If there is any way I can help, please let me know. I assume it should be straightforward since you have `1.6.0` for `humble`.

mamoll, 2024-04-26:
@Yadunund I won't be able to get to this for another couple of weeks. If you want to take a stab at it, go ahead. The main bottlenecks for me are (1) setting up a VM and (2) trying to find the bloom steps. Searching my email, I believe the right command after you have set up your env is simply:

    bloom-release ompl -r jazzy -t jazzy

Yadunund, 2024-04-27:
Hi @mamoll,

Thanks for the reply. I'm familiar with running the bloom command. It should be 
```bash
bloom-release --rosdistro iron ompl # replace iron with rolling or jazzy
```

One thing I'm confused by is that the configured [devel branch to release from](https://github.com/ros2-gbp/ompl-release/blob/b38350367cb4fb38671b64d5d61282eceafe80b1/tracks.yaml#L110) is `1.5.2`. However I see no such branch in this repository? Any change is was accidentally deleted? We'll need to bloom from the same branch to iron, rolling and jazzy since the `1.6.0` tag was present in this branch.

Another tricky part is that I won't be able to bloom a release unless I'm authorized to do so -> ie, without being added to the [ompl team in ros2-gbp](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/e83a3ac620818eafcff326820b385b5b7e0c7db2/ompl.tf#L3). If you're comfortable, I can request to be added to that group.

mamoll, 2024-04-28:
There is [a tag for 1.5.2](https://github.com/ompl/ompl/releases/tag/1.5.2). I don't quite understand why 1.6.0 and 1.5.2 need to be in the same branch, but they likely are.

Please submit a request to be added to the ompl team in ros2-gbp. I'd be happy to add you.

Yadunund, 2024-08-12:
Closing the loop here,
* I've updated `devel_branch` for the active distros in track.yaml to `main`: https://github.com/ros2-gbp/ompl-release/commit/1699c86188351cb1a9e3388b3d64e2fa6885ad73, https://github.com/ros2-gbp/ompl-release/commit/2a692cf8d801d005aece3e51f97f96a79a956d27 
* Then bloomed `1.6.0` binaries for 
  * Iron:  https://github.com/ros/rosdistro/pull/42064, https://github.com/ros/rosdistro/pull/42426
  * Jazzy: https://github.com/ros/rosdistro/pull/42425
  * Rolling: https://github.com/ros/rosdistro/pull/42424
 
 In the future when Rolling is migrated to a new ROS distribution, eg, K-turtle, the version on `ompl` on Rolling will also be released into K-tutrle thereby preventing the issue that we saw with Iron and Jazzy. We don't have to update the `tracks.yaml` in the foreseeable future.

closed 2024-08-12 by Yadunund

mamoll, 2024-08-12:
Thanks, @Yadunund!