ompl/ompl #1110: ModuleNotFoundError: No module named 'ompl.util._util'

MaTianji-nuaa-hit, 2023-11-10:
After the installation,when I run 
> python3 ompl_app.py
The relevant logs of the command are as follows:
>ModuleNotFoundError: No module named 'ompl.util._util'
I don't know where the problem occurred during the installation process or during thje generation of py.

guankou, 2023-11-21:
i have the same question

zkingston, 2023-11-28:
Are you trying to build the latest version of OMPL? Please checkout the latest commit. Also make sure you are using v2.2.1 of PyGCCXML.

tl123987, 2023-12-01:
i have the same question
[image]

hshose, 2024-01-12:
Same issue here on up-to-date arch and python 3.11
I've tried `main` branch and  `1.6.0` release; also when downgrading to PyGCCXML v2.2.1, `ompl.util._util` is not found in any case when running in python `from ompl import base`.

zkingston, 2024-01-15:
For further debugging, please post output of the `update_bindings` step. Also, if you just need base OMPL Python bindings, please try using the pre-built wheels: https://github.com/ompl/ompl/releases/tag/prerelease

0Z17, 2024-01-16:
I also encountered the same issue and resolved it by upgrading PyPy to version ```7.3.15.``` I will provide more details so that users facing similar problems can refer to them and contribute to better project maintenance.

My machine is running Ubuntu ```20.04```, PyPy version ```7.3.15```, and pygccxml version ```2.4.0```.

During the initial compilation process using the ```install-ompl-ubuntu.sh``` script, some errors were encountered. One of them was reported in the ```build/Release/pyplusplus_util.log``` file as follows:

```
Traceback (most recent call last):
  File "/home/z017/Projects/OMPL/omplapp-1.6.0-Source/ompl/py-bindings/generate_bindings.py", line 41, in <module>
    from pygccxml import declarations
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/__init__.py", line 45, in <module>
    import importlib_metadata as metadata
  File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/__init__.py", line 1
    from __future__ import annotations
    ^
SyntaxError: future feature annotations is not defined
```

After some research, I discovered that it was a PyPy version issue. PyPy 7.3.1 (corresponding to Python 3.6) does not support the import of ```annotations``` from ```__future__ ```, which is available in Python 3.7.

Therefore, I upgraded PyPy to version ```7.3.15```, reran the ```install-ompl-ubuntu.sh``` script, successfully compiled, and completed the generation of bindings.

AbhishekGY, 2024-04-20:
> I also encountered the same issue and resolved it by upgrading PyPy to version `7.3.15.` I will provide more details so that users facing similar problems can refer to them and contribute to better project maintenance.
> 
> My machine is running Ubuntu `20.04`, PyPy version `7.3.15`, and pygccxml version `2.4.0`.
> 
> During the initial compilation process using the `install-ompl-ubuntu.sh` script, some errors were encountered. One of them was reported in the `build/Release/pyplusplus_util.log` file as follows:
> 
> ```
> Traceback (most recent call last):
>   File "/home/z017/Projects/OMPL/omplapp-1.6.0-Source/ompl/py-bindings/generate_bindings.py", line 41, in <module>
>     from pygccxml import declarations
>   File "/usr/local/lib/python3.8/dist-packages/pygccxml/__init__.py", line 45, in <module>
>     import importlib_metadata as metadata
>   File "/usr/local/lib/python3.8/dist-packages/importlib_metadata/__init__.py", line 1
>     from __future__ import annotations
>     ^
> SyntaxError: future feature annotations is not defined
> ```
> 
> After some research, I discovered that it was a PyPy version issue. PyPy 7.3.1 (corresponding to Python 3.6) does not support the import of `annotations` from `__future__ `, which is available in Python 3.7.
> 
> Therefore, I upgraded PyPy to version `7.3.15`, reran the `install-ompl-ubuntu.sh` script, successfully compiled, and completed the generation of bindings.
> 
> This is my first time responding to an issue, so if there are any mistakes or areas for improvement, I appreciate your understanding and welcome any feedback.

@0Z17 Can you please explain how you upgraded to 7.3.15.? I tried using apt upgrade but it says that the latest version is 7.3.1 and its already installed on my system.

0Z17, 2024-04-20:
@AbhishekGY I can't quite remember exactly how I upgraded to 7.3.15, but I encountered the same issue where I couldn't update to 7.3.15 through apt upgrade. If I remember correctly, I downloaded it directly from the official website. You can try visiting this link: https://www.pypy.org/download.html.

Hope this helps!

AbhishekGY, 2024-04-20:
@0Z17 I'll give that a try. Thank you very much!

296620546, 2024-04-22:
> @0Z17 I'll give that a try. Thank you very much! 我会给予一个尝试。非常感谢您！

I have encountered the same issue. Has it been resolved?

Olumat, 2024-07-01:
I just need base ompl python bindings, but the wheels don't work on WSL 1. Same issue here, I can only upgrade PyPy to Version 7.3.16 and I have the same error still. I saw that this is a common issue when trying to install ompl. Does anyone have insight what is actually wrong?

eunseon02, 2024-07-16:
> I just need base ompl python bindings, but the wheels don't work on WSL 1. Same issue here, I can only upgrade PyPy to Version 7.3.16 and I have the same error still. I saw that this is a common issue when trying to install ompl. Does anyone have insight what is actually wrong?

have you solved it?

eunseon02, 2024-07-16:
i solved this issue by installing the whl files corresponding to my python --version
https://github.com/ompl/ompl/releases/tag/prerelease

anpom21, 2024-10-14:
Hi i tried to install the whl files from there using pip but it didnt resolve my issue?
Running ubuntu 24.04 and python3.12
`from ompl import base as ob
  File "/usr/lib/python3/dist-packages/ompl/base/__init__.py", line 1, in <module>
    from ompl import util
  File "/usr/lib/python3/dist-packages/ompl/util/__init__.py", line 4, in <module>
    from ompl.util._util import *
ModuleNotFoundError: No module named 'ompl.util._util'`
Any other suggestions?

zhaianbang, 2025-01-06:
```
ompl的segment fault的问题，解决方案，已解决：
运行genesis，直接ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl，运行grasp_bottle程序出现ompl的segment fault的问题，解决步骤如下：
（重点：注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决）
用python=3.10的conda环境：
conda create genesis0106 python=3.10

conda activate enesis0106

pip install torch==2.5.1 torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118

sudo apt-get -y install g++ cmake pkg-config libboost-serialization-dev libboost-filesystem-dev libboost-system-dev libboost-program-options-dev libboost-test-dev libeigen3-dev libode-dev wget libyaml-cpp-dev
export CXX=g++
export MAKEFLAGS="-j `nproc`"
pip install -vU https://github.com/CastXML/pygccxml/archive/develop.zip pyplusplus
sudo apt-get -y install castxml libboost-python-dev libboost-numpy-dev pypy3

git clone --recurse-submodules https://github.com/ompl/ompl.git -b 1.6.0
mkdir -p ompl/build
cd ompl/build

cmake .. -DPYTHON_EXEC=$(which python) -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" -DPYTHONPATH=$CONDA_PREFIX/lib/python3.10/site-packages

注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决：
/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/pyplusplus_util.log

如果出现cannot import name 'Protocol' from 'typing'错误：
找到对应的行，然后注释掉typing，替换加入from typing_extensions import Protocol
pip install typing_extensions==4.5.0
pip install pygccxml==2.1.0

修改注释/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/py-bindings/generate_bindings.py中194行的：

    # self.ompl_ns.class_(f'SpecificParam< {self.string_decl} >').rename('SpecificParamString')
    try:
        self.ompl_ns.class_(f'SpecificParam< std::string >').rename('SpecificParamString')
    except:
        self.ompl_ns.class_(f'SpecificParam< std::basic_string< char > >').rename('SpecificParamString')

然后执行：
make update_bindings
（如果出现No rule for update_bindings报错，执行conda install -c conda-forge boost boost-cpp）

再执行：
make -j8
sudo make install

欢迎交流ab_zhai@foxmail.com
```

danjayy, 2025-05-08:
I encountered similar issue when trying to install OMPL with the python bindings in an Ubuntu 20.04 docker container. After much debugging and reading the comments here, I found that the issue was with the library dependencies. The steps for installation (in a docker container or locally) are in my repo: [https://github.com/danjayy/OMPL_Python_bindings/blob/main/README.md](ompl_python_bindings)