ompl/ompl #1112: Issue in saving/reading PRM Planner Data

shloksobti, 2023-11-16:
There is an issue in saving and reading `PlannerData` for PRM. When I construct a Roadmap and then save it - say it has two disjoint sets - then when I read it, it always reads in as having a single component. This is problematic because when planning with precomputed roadmaps, it throws an uncaught exception when it checks for a solution to exist (by checking if start and goal are in the same component).

Minimum Reproducible Example:
In this example, we plan in a simple R2 space which is disjoint so you would expect 2 components.
```
#include <ompl/base/PlannerDataStorage.h>
#include <ompl/base/spaces/RealVectorStateSpace.h>
#include <ompl/geometric/SimpleSetup.h>
#include <ompl/geometric/planners/prm/PRM.h>

namespace ob = ompl::base;
namespace og = ompl::geometric;

#define foreach BOOST_FOREACH

class MyPRM : public ompl::geometric::PRM {
 public:
  MyPRM(const ompl::base::PlannerData& data, bool starStrategy = false)
      : ompl::geometric::PRM(data, starStrategy) {}

  std::size_t getNumComponents() {
    std::set<int> uniqueParents;
    for (int i = 0; i < milestoneCount(); ++i)
      uniqueParents.insert(disjointSets_.find_set(i));

    return uniqueParents.size();
  }
};

bool isStateValid(const ob::State* state) {
  auto values = state->as<ob::RealVectorStateSpace::StateType>()->values;

  if (values[0] < -1 || values[0] > 1) {
    return true;
  }
  return false;
}

void plan_and_save() {
  auto space(std::make_shared<ob::RealVectorStateSpace>(2));

  ob::RealVectorBounds bounds(2);
  bounds.setLow(-2);
  bounds.setHigh(2);
  space->setBounds(bounds);

  auto si(std::make_shared<ob::SpaceInformation>(space));
  si->setStateValidityChecker(isStateValid);
  si->setup();

  ob::ScopedState<> start(space);
  start.random();
  ob::ScopedState<> goal(space);
  goal.random();

  auto pdef(std::make_shared<ob::ProblemDefinition>(si));

  pdef->setStartAndGoalStates(start, goal);

  auto planner(std::make_shared<MyPRM>(si));
  planner->setProblemDefinition(pdef);
  planner->setup();

  planner->constructRoadmap(ompl::base::timedPlannerTerminationCondition(1.0));

  std::cout << "Num Components: " << planner->getNumComponents() << std::endl;

  // Save planner Data!
  ompl::base::PlannerData data(si);
  planner->getPlannerData(data);
  ompl::base::PlannerDataStorage storage;
  std::string filename = "/home/test_prm";
  storage.store(data, filename.c_str());
  std::cout << "Saved!\n";

  return;
}

void load_and_plan() {
  auto space(std::make_shared<ob::RealVectorStateSpace>(2));

  ob::RealVectorBounds bounds(2);
  bounds.setLow(-2);
  bounds.setHigh(2);
  space->setBounds(bounds);

  auto si(std::make_shared<ob::SpaceInformation>(space));
  si->setStateValidityChecker(isStateValid);
  si->setup();

  ob::ScopedState<> start(space);
  start.random();
  ob::ScopedState<> goal(space);
  goal.random();

  auto pdef(std::make_shared<ob::ProblemDefinition>(si));

  pdef->setStartAndGoalStates(start, goal);

  // Load planner Data!
  std::string filename = "/home/test_prm";
  ompl::base::PlannerData data(si);
  ompl::base::PlannerDataStorage storage;
  storage.load(filename.c_str(), data);

  auto planner(std::make_shared<MyPRM>(data));
  planner->setProblemDefinition(pdef);
  planner->setup();

  std::cout << "Num Components: " << planner->getNumComponents() << std::endl;

  return;
}

int main() {
  plan_and_save(); // Outputs 2
  load_and_plan(); // Outputs 1
  return 0;
}

```

shloksobti, 2023-11-16:
Potential fix: https://github.com/ompl/ompl/pull/1114

mamoll, 2023-11-22:
Thanks for the example code and the PR. The PR needs a very minor cleanup, but I did approve it. Thanks for the fix!

closed 2023-11-22 by shloksobti