ompl/ompl #1113: STRRT* support

JojoSim0n, 2023-11-16:
Hi, i want to use the st-rrt* planner from the ompl lib, but it seems it is not integrated in the python version? Or if it is, whats its planner name ?

zkingston, 2023-11-27:
ST-RRT* is available in Python with `ompl.geometric.STRRTstar()`

closed 2023-11-27 by zkingston

JojoSim0n, 2023-11-28:
Apparently using ompl.geometric.STRRTstar() returns `AttributeError: module 'ompl.geometric' has no attribute 'STRRTstar'. Did you mean 'SORRTstar'` so this question is not closed

zkingston, 2023-11-28:
Python bindings for ST-RRT* were added with #1009. Have you built OMPL with the latest version from the repo?

JojoSim0n, 2023-11-29:
at first i just used the install script from the ompl website, noticing that this is from january, i cloned the newest version of the repo and build python bindings but receiving error for geometric:

 `file "_Motion__value_traits.pypp.hpp" - updated( 0.000162 seconds )
Traceback (most recent call last):
  File "/home/johannes/ompl/py-bindings/generate_bindings.py", line 995, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/johannes/ompl/py-bindings/generate_bindings.py", line 652, in __init__
    replacement)
  File "/home/johannes/ompl/py-bindings/ompl/bindings_generator.py", line 156, in __init__
    self.mb.split_module('bindings/' + name, use_files_sum_repository=True)
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/module_builder/boost_python_builder.py", line 389, in split_module
    , encoding=self.encoding)
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/file_writers/__init__.py", line 37, in write_multiple_files
    mfs.write()
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/file_writers/multiple_files.py", line 395, in write
    self.split_values_traits()
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/file_writers/multiple_files.py", line 321, in split_values_traits
    for trait in self.__value_traits: self.split_value_traits(trait)
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/file_writers/multiple_files.py", line 317, in split_value_traits
    , value_traits.create() ) )
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/file_writers/multiple_files.py", line 56, in write_file
    raise RuntimeError( os.linesep.join(msg) )
RuntimeError: `Py++` is going to write different content to the same file(bindings/geometric/_Motion__value_traits.pypp.hpp).
The following is a short list of possible explanations for this behaviour:
* `Py++` bug, in this case, please report it
* module_builder_t contains two or more classes with the same alias
* module_builder_t contains two or more classes with the same wrapper alias
Please carefully review `Py++` warning messages. It should contain an additional information.`

zkingston, 2023-11-29:
Please post the entire debug message and specifics about your machine - what versions of software are you using, what OS, etc.

JojoSim0n, 2023-12-11:
System: Ubuntu 22.04
python version: 3.10.12 / g++: 11.4.0
Installation: followed the instructions on github, download repo, mkdir, cmake, make update_bindings ... 
pyplusplus_geometric.log:
`In file included from /tmp/tmpvptj28e_.h:1:
In file included from ./bindings/geometric.h:340:
In file included from /home/johannes/Documents/ompl/src/ompl/base/SpaceInformation.h:43:
In file included from /home/johannes/Documents/ompl/src/ompl/base/StateSpace.h:43:
In file included from /home/johannes/Documents/ompl/src/ompl/base/ProjectionEvaluator.h:49:
In file included from /usr/include/eigen3/Eigen/Core:250:
/usr/include/eigen3/Eigen/src/Core/arch/Default/GenericPacketMathFunctions.h:623:16: warning: unknown attribute 'optimize' ignored [-Wunknown-attributes]
__attribute__((optimize("-fno-unsafe-math-optimizations")))
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
/usr/local/lib/python3.10/dist-packages/pyplusplus/binary_parsers/parsers.py:7: DeprecationWarning: invalid escape sequence 'm'
  """
Replacing member function ::ompl::geometric::PathGeometric::print
Replacing member function ::ompl::geometric::SimpleSetup::print
Replacing member function ::ompl::geometric::PathHybridization::print
Replacing member function ::ompl::tools::SelfConfig::print
Replacing member function ::ompl::base::OptimizationObjective::print
Replacing member function ::ompl::base::GoalRegion::print
Replacing member function ::ompl::base::Goal::print
Replacing member function ::ompl::base::ProblemDefinition::print
Replacing member function ::ompl::base::ScopedState<>::print
Replacing member function ::ompl::base::Path::print
Replacing member function ::ompl::base::ProjectionMatrix::print
Replacing member function ::ompl::base::ParamSet::print
Replacing member function
::ompl::geometric::PathGeometric::printAsMatrix
Replacing member function ::ompl::base::TimeStateSpace::printSettings
Replacing member function
::ompl::base::RealVectorStateSpace::printSettings
Replacing member function ::ompl::base::Planner::printSettings
Replacing member function
::ompl::base::SpaceInformation::printSettings
Replacing member function ::ompl::base::StateSpace::printSettings
Replacing member function
::ompl::base::CompoundStateSpace::printSettings
Replacing member function
::ompl::base::ProjectionEvaluator::printSettings
Replacing member function ::ompl::geometric::SPARStwo::printDebug
Replacing member function ::ompl::geometric::SPARS::printDebug
still alive after 60.0608 seconds!
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer
__first, std::vector<double>::pointer __last,
std::vector<double>::pointer __result,
std::vector<double>::_Tp_alloc_type & __alloc, std::true_type arg4)
[member function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer arg0,
std::vector<double>::pointer arg1, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & arg3, std::false_type
arg4) [member function] > compilation error W1050: The function
returns "std::vector<double>::pointer" > type. You have to specify a
call policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<double>::pointer
std::vector<double>::_S_relocate(std::vector<double>::pointer __first,
std::vector<double>::pointer __last, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & __alloc) [member
function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies.Be sure to take a look on `Py++` > defined call policies
static std::vector<unsigned int>::pointer std::vector<unsigned
int>::_S_do_relocate(std::vector<unsigned int>::pointer __first,
std::vector<unsigned int>::pointer __last, std::vector<unsigned
int>::pointer __result, std::vector<unsigned int>::_Tp_alloc_type &
[... 1372 lines cut ...]
file "SPARStwo.pypp.cpp" - updated( 0.000443 seconds )
file "SST.pypp.hpp" - updated( 0.000174 seconds )
file "SST.pypp.cpp" - updated( 0.000283 seconds )
file "STRIDE.pypp.hpp" - updated( 0.000162 seconds )
file "STRIDE.pypp.cpp" - updated( 0.000308 seconds )
file "STRRTstar.pypp.hpp" - updated( 0.000217 seconds )
file "STRRTstar.pypp.cpp" - updated( 0.000363 seconds )
file "SimpleSetup.pypp.hpp" - updated( 0.000204 seconds )
file "SimpleSetup.pypp.cpp" - updated( 0.000309 seconds )
file "TRRT.pypp.hpp" - updated( 0.000160 seconds )
file "TRRT.pypp.cpp" - updated( 0.000260 seconds )
file "_geometric_free_functions.pypp.hpp" - updated( 0.000145 seconds
)
file "_geometric_free_functions.pypp.cpp" - updated( 0.000147 seconds
)
file "_geometric.main.cpp" - updated( 0.000279 seconds )`

creating error:
python3  
`>>>from ompl import geometric as og
 >>> p = og.STRRTstar
 Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'ompl.geometric' has no attribute 'STRRTstar'. Did you mean: 'SORRTstar'?
'