ompl/ompl #1116: Building Python bindings fails on Ubuntu 22.04

panthuncia, 2023-11-18:
When attempting to build in Ubuntu 22.04, I cannot successfully build Python bindings. I can import "ompl" in Python, but many things are missing. 

I see that I am getting errors in Python binding building about pygccxml, specifically:

/usr/local/lib/python3.10/dist-packages/pyplusplus/binary_parsers/parsers.py:7: DeprecationWarning: invalid escape sequence 'm'
  """
Traceback (most recent call last):
  File "/home/jetson/ompl/py-bindings/generate_bindings.py", line 995, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/jetson/ompl/py-bindings/generate_bindings.py", line 652, in __init__
    replacement)
  File "/home/jetson/ompl/py-bindings/ompl/bindings_generator.py", line 142, in __init__
    self.std_ns = self.mb.namespace('std')
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/module_builder/module_builder.py", line 267, in namespace
    , recursive=recursive )
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/namespace.py", line 121, in namespace
    recursive=recursive)
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 92, in get_single
    raise runtime_errors.multiple_declarations_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.multiple_declarations_found_t: Multiple declarations have been found. Matcher: [(decl type==namespace_t) and (name==std)]

I have version 2.4.0 of pygccxml.

Here is my build log:

jetson@nano:~$ ./install-ompl-ubuntu.sh -g -p
Hit:1 https://linux.teamviewer.com/deb stable InRelease
Hit:2 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  gcc-10-base
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++ is already the newest version (4:11.2.0-1ubuntu1).
libboost-filesystem-dev is already the newest version (1.74.0.3ubuntu7).
libboost-program-options-dev is already the newest version (1.74.0.3ubuntu7).
libboost-system-dev is already the newest version (1.74.0.3ubuntu7).
libyaml-cpp-dev is already the newest version (0.7.0+dfsg-8build1).
pkg-config is already the newest version (0.29.2-1ubuntu3).
wget is already the newest version (1.21.2-2ubuntu1).
libboost-serialization-dev is already the newest version (1.74.0.3ubuntu7).
libboost-test-dev is already the newest version (1.74.0.3ubuntu7).
libeigen3-dev is already the newest version (3.4.0-2ubuntu2).
libode-dev is already the newest version (2:0.16.2-1).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
The following package was automatically installed and is no longer required:
  gcc-10-base
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.4).
The following package was automatically installed and is no longer required:
  gcc-10-base
Use 'sudo apt autoremove' to remove it.
[... 609 lines cut ...]
 * boost_headers (required version == 1.74.0)
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.9.2), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * boost_python-py310 (required version == 1.74.0)

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done (1.6s)
-- Generating done (0.5s)
-- Build files have been written to: /home/jetson/ompl/build/Release
[100%] Built target update_tools_bindings
[100%] Built target update_bindings

Any ideas what might be causing this?

zkingston, 2023-11-28:
Try using [pygccxml 2.2.1](https://github.com/CastXML/pygccxml/releases/tag/v2.2.1)

patrick-5546, 2024-04-04:
I think I am running into the same problem on the raspberry pi 4 also running Ubuntu 22.04:
- Updated the dockerfile to comment out libspot (no arm package) and ping pygccxml: https://github.com/patrick-5546/ompl/blob/patrick-5546/fix-arm-build/scripts/docker/ompl.Dockerfile
- Ran with `docker build . -f scripts/docker/ompl.Dockerfile -t ompl`
- The build completes but some binaries aren't there
    - For example, when I compile on x86 `/usr/lib/python3/dist-packages/ompl/util/_util.so` exists, but not on arm
- Build log: [ompl-build.log](https://github.com/ompl/ompl/files/14860041/ompl-build.log)

patrick-5546, 2024-04-05:
Looking into this more, I believe the relevant error in the log is this
```
#11 15.98 ERROR error occured, while parsing element with name "Field" and attrs "['id', 'name', 'type', 'context', 'access', 'offset']".
#11 15.98 Error: 'line'.
#11 15.98 Error: can't generate code for module util
```

Tracing back where this error appears in the code:
https://github.com/ompl/ompl/blob/763fc2f1e5591a42b0a95316cdf9dd10a6ae20cd/py-bindings/generate_bindings.py#L998
https://github.com/ompl/ompl/blob/763fc2f1e5591a42b0a95316cdf9dd10a6ae20cd/py-bindings/bindings_generator.py.in#L132-L139
https://github.com/ompl/ompl/blob/763fc2f1e5591a42b0a95316cdf9dd10a6ae20cd/py-bindings/bindings_generator.py.in#L45

However I don't know where to go from there

Following these related issues, I tried building castxml from source using clang/llvm 11 but I am still getting the same error
- https://github.com/ompl/ompl/issues/725#issuecomment-658925712
- https://lists.gnu.org/archive/html/discuss-gnuradio/2021-12/msg00106.html

chuangwang1991, 2024-04-23:
> Try using [pygccxml 2.2.1](https://github.com/CastXML/pygccxml/releases/tag/v2.2.1)

```python
-- Build type: Release
-- Code for module util not found; type "make update_bindings"
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * spot, <http://spot.lrde.epita.fr/>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org/>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org/).
 * boost_numpy (required version == 1.74.0)
 * Python, <https://www.python.org/>
   Used for python bindings.
 * boost_python (required version == 1.74.0)
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * boost_unit_test_framework (required version == 1.74.0)
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * boost_serialization (required version == 1.74.0)
 * boost_filesystem (required version == 1.74.0)
 * boost_system (required version == 1.74.0)
 * boost_program_options (required version == 1.74.0)
 * Eigen3, <http://eigen.tuxfamily.org/>
   A linear algebra library used throughout OMPL.
 * boost_headers (required version == 1.74.0)
 * Boost (required version >= 1.58), <https://www.boost.org/>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * pypy, <https://pypy.org/>
   Used to speed up the generation of python bindings.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.9.2), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * boost_python-py310 (required version == 1.74.0)

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /root/ompl/ompl/build/Release
root@localhost:~/ompl/ompl/build/Release# make -j 4 update_bindings
[  0%] Preparing C++ header file for Python binding generation for module base
[  0%] Preparing C++ header file for Python binding generation for module tools
[  0%] Built target tools.h
[ 50%] Preparing C++ header file for Python binding generation for module util
[ 50%] Preparing C++ header file for Python binding generation for module control
[ 50%] Built target base.h
[ 50%] Built target control.h
[ 50%] Built target util.h
[ 50%] Preparing C++ header file for Python binding generation for module geometric
[ 50%] Creating C++ code for Python module util (see pyplusplus_util.log)
[ 50%] Built target geometric.h
Traceback (most recent call last):
  File "/root/ompl/ompl/py-bindings/generate_bindings.py", line 1038, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/root/ompl/ompl/py-bindings/generate_bindings.py", line 998, in __init__
    code_generator_t.__init__(self, 'util', None, replacement, 1)
  File "/root/ompl/ompl/py-bindings/ompl/bindings_generator.py", line 142, in __init__
    self.std_ns = self.mb.namespace('std')
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/module_builder/module_builder.py", line 265, in namespace
    return self.global_ns.namespace( name=name
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/namespace.py", line 117, in namespace
    self._find_single(
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 92, in get_single
    raise runtime_errors.multiple_declarations_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.multiple_declarations_found_t: Multiple declarations have been found. Matcher: [(decl type==namespace_t) and (name==std)]
``` 

I met the same problem on ubuntu 22.04 arm64 with castxml 0.4.8 and pygccxml 2.4.0
(castxml 0.4.4 build failed on arm64....., maybe the llvm and clang problem, so I just download the prebuilt version)

While I build successfully on ubuntu 22.04 x86 with castxml 0.4.4 and pygccxml 2.4.0

Any advice will be appreciated, thanks a lot

chuangwang1991, 2024-04-23:
> Looking into this more, I believe the relevant error in the log is this
> 
> ```
> #11 15.98 ERROR error occured, while parsing element with name "Field" and attrs "['id', 'name', 'type', 'context', 'access', 'offset']".
> #11 15.98 Error: 'line'.
> #11 15.98 Error: can't generate code for module util
> ```
> 
> Tracing back where this error appears in the code:
> 
> https://github.com/ompl/ompl/blob/763fc2f1e5591a42b0a95316cdf9dd10a6ae20cd/py-bindings/generate_bindings.py#L998
> 
> 
> https://github.com/ompl/ompl/blob/763fc2f1e5591a42b0a95316cdf9dd10a6ae20cd/py-bindings/bindings_generator.py.in#L132-L139
> 
> 
> https://github.com/ompl/ompl/blob/763fc2f1e5591a42b0a95316cdf9dd10a6ae20cd/py-bindings/bindings_generator.py.in#L45
> 
> However I don't know where to go from there
> 
> Following these related issues, I tried building castxml from source using clang/llvm 11 but I am still getting the same error
> 
> * [Could not build OMPL with Python bindings on Raspberry Pi 4 #725 (comment)](https://github.com/ompl/ompl/issues/725#issuecomment-658925712)
> * https://lists.gnu.org/archive/html/discuss-gnuradio/2021-12/msg00106.html

Just build ompl from github main branch, not 1.6.0

or you can change the '' CMakeModules/FindPython.cmake'' at line 86 (line 80 in latest master) there are predefined file path suffixes for python libraries for a few CPU architectures, default for - x86_64 and i386, need to add arm64, just like:
```python
PATH_SUFFIXES "" "python${PYTHON_VERSION}/config" ''aarch64-linux-gnu''
```

patrick-5546, 2024-04-24:
> Just build ompl from github main branch, not 1.6.0

I am building the main branch. In Python, I can `import ompl` but not `from ompl import util`.

I tried adding the architectures to the other `PATH_SUFFIXES` in the file, but that didn't work either: https://github.com/patrick-5546/ompl/commit/7ad1e5fdcdcf7cc3cb51cd46fae1f442de6abe09

chuangwang1991, 2024-04-25:
> > Just build ompl from github main branch, not 1.6.0
> 
> I am building the main branch. In Python, I can `import ompl` but not `from ompl import util`.
> 
> I tried adding the architectures to the other `PATH_SUFFIXES` in the file, but that didn't work either: [patrick-5546@7ad1e5f](https://github.com/patrick-5546/ompl/commit/7ad1e5fdcdcf7cc3cb51cd46fae1f442de6abe09)

It's still a open issue...

I have tried theOMPL  main branch, and 1.6.0, 

castxml 0.4.4 with pygccxml 2.2.1,  building the 'util.h' faied, just like:

```python
#11 15.98 ERROR error occured, while parsing element with name "Field" and attrs "['id', 'name', 'type', 'context', 'access', 'offset']".
#11 15.98 Error: 'line'.
#11 15.98 Error: can't generate code for module util
```

castxml 0.4.4 with pygccxml 2.4.0, 
castxml 0.4.8 with pygccxml 2.4.0, 
castxml 0.5.0 with pygccxml 2.5.0, 

 building faied, just like:
```python
[ 50%] Built target util.h
[ 50%] Preparing C++ header file for Python binding generation for module geometric
[ 50%] Creating C++ code for Python module util (see pyplusplus_util.log)
[ 50%] Built target geometric.h
Traceback (most recent call last):
  File "/root/ompl/ompl/py-bindings/generate_bindings.py", line 1038, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/root/ompl/ompl/py-bindings/generate_bindings.py", line 998, in __init__
    code_generator_t.__init__(self, 'util', None, replacement, 1)
  File "/root/ompl/ompl/py-bindings/ompl/bindings_generator.py", line 142, in __init__
    self.std_ns = self.mb.namespace('std')
  File "/usr/local/lib/python3.10/dist-packages/pyplusplus/module_builder/module_builder.py", line 265, in namespace
    return self.global_ns.namespace( name=name
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/namespace.py", line 117, in namespace
    self._find_single(
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 92, in get_single
    raise runtime_errors.multiple_declarations_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.multiple_declarations_found_t: Multiple declarations have been found. Matcher: [(decl type==namespace_t) and (name==std)]
```

edwinpha, 2024-05-16:
Same issue here...although I have tried both in ubuntu20 and 22 (docker images) with ARM64

johnnynunez, 2025-01-31:
I updated castxml etc with arm runners, we should go fast now with this problem

johnnynunez, 2025-03-04:
it is working https://github.com/johnnynunez/ompl/actions/runs/13643981441
close :)

johnnynunez, 2025-03-06:
close https://github.com/ompl/ompl/releases/tag/prerelease

closed 2025-10-01 by saahu27