ompl/ompl #1118: How can I save the solved path as a eigen matrix

BolunDai0216, 2023-11-27:
Hi,

I was using the RRTstar and RRTConnect planners in OMPL. However, I noticed that the solution is only printed to the console and I was not able to find a way to save the path to something like an Eigen Matrix or Vector.  Does anyone have any experience in this?

Thanks in advance!

zkingston, 2023-11-27:
Probably the simplest way that is space-agnostic is to copy to reals and then feed that vector to the Eigen type of your choice:
```c++
std::vector<double> reals;
for (const auto &state : path.getStates())
{
    space->copyToReals(reals, state);

    // Or copy out however you want
    const auto &vector = Eigen::Map<Eigen::VectorXd>(reals.data(), dimension);
}
```

closed 2023-11-27 by zkingston

BolunDai0216, 2023-11-27:
Hi,

@zkingston Thanks for the response! What type do you assume your `path` is?

If I am correct, the [`ompl::base::Path`](https://github.com/ompl/ompl/blob/main/src/ompl/base/Path.h#L119) class does not have a `getStates()` function.

zkingston, 2023-11-27:
Change: 
```c++
ob::PathPtr path = pdef->getSolutionPath();
```
to:
```c++
auto path = pdef->getSolutionPath();
```
It will be a `og::PathGeometric` - you can access the states from there.

BolunDai0216, 2023-11-27:
@zkingston I have updated my code to

```cpp
if (solved)
{
  // get the goal representation from the problem definition (not the same as the goal state)
  // and inquire about the found path
  auto path = pdef->getSolutionPath();
  std::cout << "Found solution:" << std::endl;

  std::vector<double> reals;
  for (const auto &state : path.getStates())
  {
    space->copyToReals(reals, state);
    // Or copy out however you want
    const auto &vector = Eigen::Map<Eigen::VectorXd>(reals.data(), 3);
  }
}
```

However, I still get the error

```bash
error: 'class std::shared_ptr<ompl::base::Path>' has no member named 'getStates'
82 |     for (const auto &state : path.getStates())
```

And when I look at the [getSolutionPath()](https://ompl.kavrakilab.org/classompl_1_1base_1_1ProblemDefinition.html#aac56933db4a17e1394b5775eefd63f19) it only outputs [`ompl::base::PathPtr`](https://ompl.kavrakilab.org/classompl_1_1base_1_1PathPtr.html).

Would it be possible to help me understand what might be the issue? Thanks in advance!

mamoll, 2023-11-27:
ProblemDefinition::getSolutionPath returns a PathPtr, not a GeometricPath Ptr. Change `auto path = pdef->getSolutionPath();` to `auto path = pdef->getSolutionPath()->as<ompl::base::geometric::PathGeometric>();` and `for (const auto &state : path.getStates())` to `for (const auto &state : path->getStates())`

BolunDai0216, 2023-11-27:
@mamoll Thanks! It works. 

For future reference, the code I am using looks like this:

```cpp
if (solved)
{
  // get the goal representation from the problem definition (not the same as the goal state)
  // and inquire about the found path
  auto path = pdef->getSolutionPath()->as<ompl::geometric::PathGeometric>();
  std::cout << "Found solution:" << std::endl;

  std::vector<double> reals;
  for (const auto &state : path->getStates())
  {
      space->copyToReals(reals, state);
      std::cout << "[" << reals[0] << ", " << reals[1] << ", " << reals[2] << "]" << std::endl;
  }
}
else
  std::cout << "No solution found" << std::endl;
```