ompl/ompl #1121: Segmentation Fault in python using Development Build Wheels

joao-pm-santos96, 2023-12-05:
Hello developers,
first things first: thank you for the amazing work with OMPL and thank you for start publishing the python wheels without the need to compile all the source code. I am truly grateful.

I know that the wheels are still in early development, but I've tested them and think I should report the issue that I am encountering. Both using python 3.10 and 3.11, following the instructions in https://github.com/ompl/ompl/pull/1103, I always receive a segmentation fault when running https://ompl.kavrakilab.org/RigidBodyPlanning_8py_source.html and https://ompl.kavrakilab.org/StateSampling_8py_source.html .

Let me know if I can help with something on the python side!

joao-pm-santos96, 2023-12-05:
Running on ubuntu 22.04.3 inside WSL2.

wbthomason, 2023-12-05:
Thanks for the report! 

Unfortunately, we don't really have access to a WSL environment to test with, and I've been unable to reproduce the issue with either script under native Linux (with Python 3.11; haven't tested 3.10 yet, but you said both had the same behavior). Could you please provide some more information about where the script segfaults? Both a Python line number and a core dump would be useful, but at least the Python line number will give us a starting point.

kylc, 2023-12-06:
Hi @joao-pm-santos96,

In a fresh Ubuntu 22.04 Docker container running under a Linux host, I'm not able to reproduce a crash. Pasted below is the minimal setup I use. If you are able, it would be helpful for you to try the equivalent under WSL and see if your issue persists.

```
# on host
docker run -it --rm ubuntu:22.04

# in container
apt update && apt install -y python3-pip git
pip install https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl
git clone https://github.com/ompl/ompl --depth 1
python3 ompl/demos/RigidBodyPlanning.py
```

joao-pm-santos96, 2023-12-06:
> Thanks for the report!
>
> Unfortunately, we don't really have access to a WSL environment to test with, and I've been unable to reproduce the issue with either script under native Linux (with Python [...]

Hi!
ompl/demos/RigidBodyPlanning.py has two functions, `planWithSimpleSetup()` and `planTheHardWay()`. 

On `planWithSimpleSetup()`  it fails here https://github.com/ompl/ompl/blob/df058b76fa305eaa075696d37ec20b803091497e/demos/RigidBodyPlanning.py#L87

On `planTheHardWay()` if fails here https://github.com/ompl/ompl/blob/df058b76fa305eaa075696d37ec20b803091497e/demos/RigidBodyPlanning.py#L123

I don't have a core dump, but if you guide me I can get it for you.

joao-pm-santos96, 2023-12-06:
> Hi @joao-pm-santos96,
> 
> In a fresh Ubuntu 22.04 Docker container running under a Linux host, I'm not able to reproduce a crash. Pasted below is the minimal setup I use. If you are able, it would be helpful for you to try the equivalent under WSL and see if your issue persists.
> 
> ```
> # on host
> docker run -it --rm ubuntu:22.04
> 
> # in container
> apt update && apt install -y python3-pip git
> python3 -m pip install https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl
> git clone https://github.com/ompl/ompl --depth 1
> python3 ompl/demos/RigidBodyPlanning.py
> ```

Using this steps, under WSL2 everything works as expected, no segmentation fault!

joao-pm-santos96, 2023-12-06:
> Thanks for the report!
>
> Unfortunately, we don't really have access to a WSL environment to test with, and I've been unable to reproduce the issue with either script under native Linux (with Python [...]

Is your distro Ubuntu? A colleague of mine has tested it under native Ubuntu and found the same issue...

kylc, 2023-12-06:
It looks like pip-installed OMPL can clash with other versions of OMPL that are installed on your system. In a `ros:humble` container, I see a segmentation fault running the demo after `apt install ros-humble-ompl`.

Probably some incompatible shared libraries getting crossed up. In the meantime I would recommend ensuring that you don't have OMPL installed via any other sources.

kylc, 2023-12-06:
Yeah, I still see the segfault in a clean virtualenv. I think the log below explains the issue, but I'm not really sure how we can work around it at build time. Maybe we should name the shared libs differently when built for Python wheels, e.g. `libompl.python.so.1.6.0`?

```
# echo $LD_LIBRARY_PATH
/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
# python3 /ompl/demos/RigidBodyPlanning.py
Segmentation fault (core dumped)

# unset LD_LIBRARY_PATH
# python3 /ompl/demos/RigidBodyPlanning.py
... runs fine

# ls /opt/ros/humble/lib/x86_64-linux-gnu
libompl.so  libompl.so.1.6.0  libompl.so.17

# ls /root/venv/lib/python3.10/site-packages/
libompl.so  libompl.so.1.6.0  libompl.so.17 ...
```

wbthomason, 2023-12-06:
Maybe related to this? https://github.com/ompl/ompl/pull/1103#issuecomment-1780379337

YukiCR, 2024-01-10:
Hi! 
I encountered the same `segmentation fault` problem as Moveit! installs `ros-neotic-ompl`  automatically for me before I install the wheels. Yet I still want to use ompl with python, will there be any solutions for it?

zkingston, 2024-01-15:
@YukiCR unfortunately, no solution yet. This is tricky since using the ROS binaries always throws a wrench into any local installation of OMPL.

YuhengZhi, 2025-01-02:
I seem to have found a workaround by adding these lines to the `dll_loader(lib, path)` of `~/miniconda3/envs/envname/lib/python3.9/site-packages/ompl/__init__.py` in the python package directory of OMPL, which was installed via wheel:
```
def dll_loader(lib, path):
    from platform import system
    from os.path import isfile
    import ctypes
    from ctypes.util import find_library

    sys = system()
    if sys == 'Windows':
        ext = '.dll'
    elif sys == 'Darwin':
        ext = '.dylib'
    else: # Linux, other UNIX systems
        ext = '.so'
    fname = path + '/lib' + lib + ext
    print(f'before: {fname}')
    if not isfile(fname): # Added
        fname = path + '/../../lib' + lib + ext # Added
        print(f'after: {fname}') # Added
    if not isfile(fname):
        fname = find_library(lib)
        print(f'after: {fname}')
    ctypes.CDLL(fname, ctypes.RTLD_GLOBAL)
```

`~/miniconda3/envs/envname/lib/python3.9/site-packages/ompl/util/__init__.py` calls `dll_loader` and expects `~/miniconda3/envs/envname/lib/python3.9/site-packages/ompl/util/libompl.so` exists. Otherwise, it starts looking into system-level libraries paths for any other ompl installations. This is true for the binding installed via the script. However, wheel installation puts `libompl.so` 2 levels higher at `~/miniconda3/envs/envname/lib/python3.9/site-packages/libompl.so`. So `dll_loader` may find the system-level library if you have installed OMPL before, no matter if it is the correct version.

These additional lines make sure you check for the `libompl.so` that comes with the wheel, before attempting to retrieve other versions installed in the system.

*My environment*:
1. I have 2 conda environments, one in 3.8, and one in 3.9.
2. Ubuntu 20.04
3. Installed OMPL through both ROS and source code.

duburcqa, 2025-03-24:
@YuhengZhi  I can confirm this. Thank you for clearing this up. It saved me a lot of trouble.

duburcqa, 2025-03-24:
Created a PR to fix the issue: https://github.com/ompl/ompl/pull/1251

closed 2025-10-01 by saahu27