ompl/ompl #1123: Motion Planning for dual-arm mobile manipulation using OMPL

im-renpei, 2023-12-07:
Hi everyone,

I would like to seek advice on utilizing OMPL for the implementation of cooperative motion planning for a dual-arm robot. The robot is equipped with two 7-DOF arms and a mobile chassis, with the body capable of rotation and lifting. In this scenario, how should I use OMPL for motion planning? I am aware that for a single robotic arm, we can directly construct a 7-dimensional joint space and use OMPL for planning within this space. However, in the case of dual-arm mobile manipulation, how can we prevent the two robotic arms from colliding with each other during the planning process?

Thank you in advance for any insights you can provide.

ChamzasKonstantinos, 2023-12-10:
Hi @im-renpei  you could use the CompoundStateSpace from OMPL to create an as complicated statespace as you need.
https://ompl.kavrakilab.org/classompl_1_1base_1_1CompoundStateSpace.html

WangYong-999, 2024-01-19:
I have the same problem. Is there any similar python example on how to use ompl to create a dual-arm mobile robot? @ChamzasKonstantinos

ChamzasKonstantinos, 2024-01-20:
Not aware of an existing python example. I have never done a dual arm robot before, but if you are able to do 1 arm just follow the same process to create a composite robot with 2 arms.

johnnynunez, 2025-03-06:
close https://github.com/ompl/ompl/releases/tag/prerelease