ompl/ompl #1128: Update_bindings not working on apple m2 macOS 13.2.1

bakuljangley, 2023-12-18:
Hello, I'm trying to use OMPL for a course project.

I've tried to install ompl on my MacBook on macOS and on an Ubuntu VM. I keep getting stuck. The main error is that it does not find a target when I run `make update_bindings`.

This is what I get while running `cmake ../..` inside build/Release (on the MacBook):
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.58") found components: serialization filesystem system program_options
warning: target 'Boost::system' is not absolute...
warning: target 'Boost::system' does not exist...
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Found PY_pyplusplus: /Users/bakuljangley/Library/Python/3.9/lib/python/site-packages/pyplusplus
-- Found PY_pygccxml: /Users/bakuljangley/Library/Python/3.9/lib/python/site-packages/pygccxml
-- Found castxml: /opt/homebrew/bin/castxml
-- Could NOT find yaml-cpp (missing: yaml-cpp_DIR)
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * flann (required version >= 1.9.2), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * boost_unit_test_framework (required version == 1.83.0)

-- The following REQUIRED packages have been found:

 * boost_serialization (required version == 1.83.0)
 * boost_atomic (required version == 1.83.0)
 * boost_filesystem (required version == 1.83.0)
 * boost_system (required version == 1.83.0)
 * boost_program_options (required version == 1.83.0)
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * boost_headers (required version == 1.83.0)
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).
 * Python, <https://www.python.org>
   Used for python bindings.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following Python modules were found:

 * pyplusplus pygccxml

-- Configuring done (0.5s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/bakuljangley/Q2/PDM/ompl/build/Release

This is what I get on running it on the VM:
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'ode'
--   Found ode, version 0.16
-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Found PY_pyplusplus: /usr/local/lib/python3.8/dist-packages/pyplusplus  
-- Found PY_pygccxml: /usr/local/lib/python3.8/dist-packages/pygccxml  
-- Found castxml: /usr/bin/castxml  
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.6.2
-- Found yaml-cpp: yaml-cpp  
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).
 * Python, <https://www.python.org>
   Used for python bindings.

-- The following Python modules were found:

 * pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/bakul/Downloads/ompl-1.6.0/build/Release

When I try to update the bindings, I get _make: *** No rule to make target 'update_bindings'.  Stop_.

zkingston, 2023-12-18:
If you do not need the C++ library, could you try first to `pip install` one of the pre-built Python wheels? https://github.com/ompl/ompl/releases/tag/prerelease

bakuljangley, 2023-12-18:
This worked thank you so much for the prompt reply !

SPDonaghy, 2024-08-14:
> As you suggested, I used a pre-built wheel.
> 
> When I try to run one of the ompl demos (RigidBodyPlanningWithControls.py), it gives me the error: TypeError: No to_python (by-value) converter found for C++ type: std::__1::shared_ptrompl::control::SpaceInformation
> 
> I need to use control for my project so this doesn't work

Were you able to get it working?