ompl/ompl #1137: RRT* often does not find a path with more time than RRT

LuukvandenBent, 2024-01-30:
So I am trying to compare some planners, and I noticed that for my toy problem, RRT finds (exact) solutions 100% of the time (n=100), also pretty fast: Minimum time: 24 ms Maximum time: 395 ms Average time: 78.26ms. However when i try to use RRTstar with the basic settings aswell, given a max planning time of 1.0 seconds, it will only find exact solutions about 94% of the time. 

I am not sure if this is intended behaviour or can be fixed with changing some settings. I was under the impression RRT* should behave like RRT (with a little overhead) untill an initial solution is found. 

Is this a bug?

scchow, 2024-02-08:
I would expect that RRT* to perform significantly slower than RRT.

RRT* has a very computationally expensive rewiring procedure that is done on every node expansion to ensure it's optimality guarantees.

zkingston, 2024-04-23:
This is expected behavior, RRT* can be very slow (it is actually quite a bit of overhead).

closed 2024-04-23 by zkingston