ompl/ompl #1138: ompl.util._util error

DavoodSZ1993, 2024-01-30:
Hello,
When I try to install OMPL on my Ubuntu 20.04 machine, I get the following error (I read and used all the previous related issues, but the problem persists!)

[ 79%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/Syclop.pypp.cpp.o
[ 79%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/SyclopEST.pypp.cpp.o
[ 79%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/SyclopRRT.pypp.cpp.o
[ 79%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/_control.main.cpp.o
[ 79%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/_control_enumerations.pypp.cpp.o
[ 80%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/_control_free_functions.pypp.cpp.o
[ 80%] Building CXX object py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/vectorControlPtr.pypp.cpp.o
[ 80%] Linking CXX shared module _control.so
Copying python module control into place
[ 80%] Built target py_ompl_control
Scanning dependencies of target test_ptc
[ 80%] Building CXX object tests/CMakeFiles/test_ptc.dir/base/ptc.cpp.o
[ 80%] Linking CXX executable test_ptc
[ 80%] Built target test_ptc
Scanning dependencies of target test_planner_data_control
[ 81%] Building CXX object tests/CMakeFiles/test_planner_data_control.dir/control/planner_data.cpp.o
[ 81%] Linking CXX executable test_planner_data_control
[ 81%] Built target test_planner_data_control
Scanning dependencies of target test_halton_sampling
[ 81%] Building CXX object tests/CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o
[ 81%] Linking CXX executable test_halton_sampling
/usr/bin/ld: CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o: in function `Halton_10D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x1cd): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o: in function `Halton_2D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x29c9): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o: in function `Halton_1D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x51b9): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o: in function `Halton_5D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x79a9): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/test_halton_sampling.dir/build.make:91: tests/test_halton_sampling] Error 1
make[1]: *** [CMakeFiles/Makefile2:1006: tests/CMakeFiles/test_halton_sampling.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

Any help would be greatly appreciated.

Best
David

zkingston, 2024-01-30:
Seems like there is an issue linking to boost. Have you installed `libboost-all-dev`?

DavoodSZ1993, 2024-01-31:
Thanks.

Yes, I have installed `libboost-all-dev`

sudo apt install libboost-all-dev
[sudo] password for davood: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libboost-all-dev is already the newest version (1.71.0.0ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

However, the problem still persists. Can this be a boost version inconsistency?

DavoodSZ1993, 2024-01-31:
Thanks

I have installed boost. However, I still get this error:

79%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/Syclop.pypp.cpp.o
[ 79%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/SyclopEST.pypp.cpp.o
[ 79%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/SyclopRRT.pypp.cpp.o
[ 79%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/_control.main.cpp.o
[ 79%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/_control_enumerations.pypp.cpp.o
[ 80%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/_control_free_functions.pypp.cpp.o
[ 80%] Building CXX object
py-bindings/CMakeFiles/py_ompl_control.dir/bindings/control/vectorControlPtr.pypp.cpp.o
[ 80%] Linking CXX shared module _control.so
Copying python module control into place
[ 80%] Built target py_ompl_control
Scanning dependencies of target test_ptc
[ 80%] Building CXX object tests/CMakeFiles/test_ptc.dir/base/ptc.cpp.o
[ 80%] Linking CXX executable test_ptc
[ 80%] Built target test_ptc
Scanning dependencies of target test_planner_data_control
[ 81%] Building CXX object
tests/CMakeFiles/test_planner_data_control.dir/control/planner_data.cpp.o
[ 81%] Linking CXX executable test_planner_data_control
[ 81%] Built target test_planner_data_control
Scanning dependencies of target test_halton_sampling
[ 81%] Building CXX object
tests/CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o
[ 81%] Linking CXX executable test_halton_sampling
/usr/bin/ld:
CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o:
in function Halton_10D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x1cd): undefined reference
to boost::filesystem::path::operator/=(boost::filesystem::path
const&)'
/usr/bin/ld:
CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o:
in function Halton_2D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x29c9): undefined reference
to boost::filesystem::path::operator/=(boost::filesystem::path
const&)'
/usr/bin/ld:
CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o:
in function Halton_1D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x51b9): undefined reference
to boost::filesystem::path::operator/=(boost::filesystem::path
const&)'
/usr/bin/ld:
CMakeFiles/test_halton_sampling.dir/base/halton_deterministic_sampling.cpp.o:
in function Halton_5D::test_method()':
halton_deterministic_sampling.cpp:(.text+0x79a9): undefined reference
to boost::filesystem::path::operator/=(boost::filesystem::path
const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/test_halton_sampling.dir/build.make:91:
tests/test_halton_sampling] Error 1
make[1]: *** [CMakeFiles/Makefile2:1006:
tests/CMakeFiles/test_halton_sampling.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

Thanks

yubinwang11, 2024-06-24:
Facing the same error when installing ompl on Ubuntu20.04. Is the problem solved?

iLUVE69, 2024-06-27:
> Facing the same error when installing ompl on Ubuntu20.04. Is the problem solved?

Hey man!, any updates on this?? Im trying to install OMPL on Ubuntu 24.04, pain in the ass